coolsnowwolf/lede #4881: newifi d1 开bbr之后，frps转发降速

tubaxiaosiji, 2020-06-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue
没有
## 2. 详细叙述
A：使用PPPOE拨号，获取到电信公网IP地址，开启frps服务，
使用默认cubic算法，frps转发速度大概在900 KB/S左右
使用bbr算法，转发速度降低至90～60KB/S。
已经过多次实测。
### (2) 路由器型号和固件版本

A： 

主机型号 | Newifi-D1
架构 | MediaTek MT7621 ver:1 eco:3 x 4
固件版本 | OpenWrt R20.5.20 / LuCI Master (git-20.141.34051-f750b5f)
内核版本 | 4.14.180

### (3) 详细日志

A：

系统日志：
Thu Jun 25 23:27:11 2020 kern.notice kernel: [    0.000000] Linux version 4.14.180 (xxxx@ubuntu) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r2567-340f1bff)) #0 SMP Thu Jun 4 08:38:42 2020
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] MIPS: machine is Newifi-D1
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Determined physical RAM map:
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000]  memory: 10000000 @ 00000000 (usable)
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Thu Jun 25 23:27:11 2020 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Thu Jun 25 23:27:11 2020 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Thu Jun 25 23:27:11 2020 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Zone ranges:
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000]   HighMem  empty
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Movable zone start for each node
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Early memory node ranges
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Thu Jun 25 23:27:11 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Thu Jun 25 23:27:11 2020 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 8061e080, node_mem_map 81003000
Thu Jun 25 23:27:11 2020 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Thu Jun 25 23:27:11 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Thu Jun 25 23:27:11 2020 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Thu Jun 25 23:27:11 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x9c/0x4e0 with crng_init=0
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26608 r8192 d22544 u57344
Thu Jun 25 23:27:11 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s26608 r8192 d22544 u57344 alloc=14*4096
Thu Jun 25 23:27:11 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Thu Jun 25 23:27:11 2020 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Writing ErrCtl register=000004a0
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Readback ErrCtl register=000004a0
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Memory: 251500K/262144K available (4943K kernel code, 252K rwdata, 1068K rodata, 1276K init, 256K bss, 10644K reserved, 0K cma-reserved, 0K highmem)
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] Hierarchical RCU implementation.
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] NR_IRQS: 256
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Thu Jun 25 23:27:11 2020 kern.info kernel: [    0.007819] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[... 643 lines cut ...]
Thu Jun 25 23:33:21 2020 daemon.info netdata[8735]: Initializing file /var/cache/netdata/cpu.cpu1/user.db.
Thu Jun 25 23:33:21 2020 daemon.info netdata[8735]: Initializing file /var/cache/netdata/cpu.cpu1/system.db.
Thu Jun 25 23:33:21 2020 daemon.info netdata[8735]: Initializing file /var/cache/netdata/cpu.cpu1/nice.db.
Thu Jun 25 23:33:21 2020 daemon.info netdata[8735]: Initializing file /var/cache/netdata/cpu.cpu1/iowait.db.
Thu Jun 25 23:33:21 2020 daemon.info netdata[8735]: Initializing file /var/cache/netdata/cpu.cpu1/idle.db.
Thu Jun 25 23:33:21 2020 daemon.info netdata[8735]: Initializing file /var/cache/netdata/cpu.cpu2/main.db.
Thu Jun 25 23:33:21 2020 daemon.info netdata[8735]: Initializing file /var/cache/netdata/cpu.cpu2/guest_nice.db.
Thu Jun 25 23:33:21 2020 daemon.info netdata[8735]: Initializing file /var/cache/netdata/disk_inodes._dev/used.db.
Thu Jun 25 23:33:21 2020 daemon.info netdata[8735]: Initializing file /var/cache/netdata/disk_inodes._dev/reserved_for_root.db.
Thu Jun 25 23:33:21 2020 daemon.info netdata[8735]: Initializing file /var/cache/netdata/netdata.plugin_diskspace/main.db.
Thu Jun 25 23:33:21 2020 daemon.info netdata[8735]: Initializing file /var/cache/netdata/netdata.plugin_diskspace/user.db.
Thu Jun 25 23:33:21 2020 daemon.info netdata[8735]: Initializing file /var/cache/netdata/netdata.plugin_diskspace/system.db.
Thu Jun 25 23:33:21 2020 daemon.info urandom_seed[8986]: Seed saved (/etc/urandom.seed)
Thu Jun 25 23:33:45 2020 daemon.info pppd[2349]: System time change detected.
Thu Jun 25 23:38:07 2020 uucp.err /usr/sbin/noddos[7023]: HostCache: ip6tables existing with code 32512
Thu Jun 25 23:38:07 2020 uucp.err /usr/sbin/noddos[7023]: HostCache: ip6tables-restore existing with code 32512
Thu Jun 25 23:43:09 2020 uucp.err /usr/sbin/noddos[7023]: HostCache: ip6tables existing with code 32512
Thu Jun 25 23:43:09 2020 uucp.err /usr/sbin/noddos[7023]: HostCache: ip6tables-restore existing with code 32512
Thu Jun 25 23:48:10 2020 uucp.err /usr/sbin/noddos[7023]: HostCache: ip6tables existing with code 32512
Thu Jun 25 23:48:10 2020 uucp.err /usr/sbin/noddos[7023]: HostCache: ip6tables-restore existing with code 32512
Thu Jun 25 23:53:11 2020 uucp.err /usr/sbin/noddos[7023]: HostCache: ip6tables existing with code 32512
Thu Jun 25 23:53:11 2020 uucp.err /usr/sbin/noddos[7023]: HostCache: ip6tables-restore existing with code 32512
Thu Jun 25 23:56:21 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Jun 25 23:56:21 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Jun 25 23:56:22 2020 user.notice nft-qos-monitor: ACTION=update, MACADDR=b8:f8:83:f8:c5:ab

tubaxiaosiji, 2020-06-25:
还有，补充一点，使用端口映射一台Debian的主机开bbr加frps服务，可以跑满上行带宽，但openwrt 上开bbr就降速的厉害。

coolsnowwolf, 2020-06-29:
玄学

closed 2020-07-02 by coolsnowwolf