coolsnowwolf/lede #4886: 可以建议大雕加入通用可设置参数的弱信号剔除功能吗？还是比较实用的。

mcucore, 2020-06-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
可以建议大雕加入通用可设置参数的弱信号剔除功能吗？还是比较实用的。

多个AP时，如果不能全部统一开启KVR，弱信号剔除是最实用最高效的选择。

所以强烈建议加入通用版本的弱信号剔除功能，谢谢！

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

coolsnowwolf, 2020-06-28:
有建议值吗

jxph1123, 2020-06-28:
建议可设置，然后自己根据情况自己实验适合自己的值。

mcucore, 2020-06-28:
> 有建议值吗

建议弄成可设置的，实际使用还是需要调节的。

按大雕回复，莫非在源代码里有地方直接改剔除阈值？hostapd、mac80211里？能否告知具体位置先应急用用，谢谢！

jw10126121, 2020-06-29:
强烈想要这个功能，但是，我已经用了脚本去实现自动剔除弱信号了，目前先用着了。。

chenhao6938, 2020-06-29:
强烈需要这个功能。
目前只有一个剔除弱信号选项，勾选了也没有什么特别的效果。最好是有个设定阈值，低于了就自动剔除。
现在用脚本+三方ipk实现的，毕竟没有原生的香。感谢大神！

lexin8, 2020-06-30:
不建议，本身驱动就有各种问题，就算加入该功能效果也不理想。专业的东西专业来做，上正儿八经的AP不好么

ElectronicElephant, 2020-06-30:
I'd suggest enabling KVR option in all APs.

jw10126121, 2020-07-01:
> 强烈需要这个功能。
> 目前只有一个剔除弱信号选项，勾选了也没有什么特别的效果。最好是有个设定阈值，低于了就自动剔除。
> 现在用脚本+三方ipk实现的，毕竟没有原生的香。感谢大神！

请问你是如何做的呢？ 有没有教程？？

closed 2020-07-04 by coolsnowwolf

jxph1123, 2020-08-12:
https://github.com/smallerxie/wifi-kickout  目前暂时用的这个，设置了5秒循环执行一次，效果还行。。
还是期待原生支持阀值设置。