coolsnowwolf/lede #4893: docker网络不通问题

lexin8, 2020-06-29:
X86环境固件：
最新源码多个月的测试，也编译其他作者的固件都是这个问题
macvlan自定义网络无法PING通外网，路由IP都PING不通。liunx系统下安装docker测试是可以访问外网正常的。

简单来说想实现docker里运行的容器可以获取到路由同网段局域网的IP上网。以下是不通的情况。
[image]
[image]

============================================
以下是使用其他liunx系统测试正常以及使用一位叫F大的N1固件测试也正常

[image]
[image]

hyy-666, 2020-07-01:
可以试一下在防火墙里勾选转发

hxlls, 2020-07-01:
更新把，最新的这些问题都解决了，包括之前docker引起的的端口环回不生效问题，刚才测试已经接近，docker 可以正式实用状态了

closed 2020-07-02 by coolsnowwolf

lexin8, 2020-07-02:
> 更新把，最新的这些问题都解决了，包括之前docker引起的的端口环回不生效问题，刚才测试已经接近，docker 可以正式实用状态了

刚刚重新编译最新版本还是一样的问题，macvlan网络通你上个图看看