coolsnowwolf/lede #4894: ddns 无法自定义

shzxm, 2020-06-29:
ddns 
服务提供商 选择 自定义 后
保存并应用 
会还原成默认第一项
[image]

[image]

脚本的权限已给
[image]

shzxm, 2020-06-29:
Openwrt Koolshare mod V2.35 r14297-5efd32232b

dolphinpaopao, 2020-06-29:
> Openwrt Koolshare mod V2.35 r14297-5efd32232b

..........
koolshare≠coolsnowwolf

shzxm, 2020-06-29:
emmm……都是同源嘛，所以问下哪位大佬知道问题在哪，或者有没有解决办法

dolphinpaopao, 2020-06-29:
这个是开源项目，就算是同源也不一定能解决问题，我刚试了这里的ddns是可以自定义的。举例koolshare的ddns是0.1的版本有问题，这里的ddns是0.2版本没有问题，那也要去找koolshare那边阿。这边无人能解答。

shzxm, 2020-06-29:
这不就好了嘛，我去卸载0.1装个0.2

closed 2020-06-29 by shzxm