coolsnowwolf/lede #4896: 网易云音乐解锁灰色音乐无法播放

xiexiaosheng, 2020-06-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：网易云音乐解锁灰色音乐。只加载不播放。

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：{"Id":"27937892","Size":18574778,"Br":999000,"Url":"http://other.player.ri03.sycdn.kuwo.cn/ec704aa307c2b5ed3a0064e34a4e9419/5ef9ec00/resource/s2/36/7/336359786.flac","Md5":"","Name":"寸断","Artist":"本兮","MatchScore":1}
--
{"Id":"36025563","Size":18574778,"Br":999000,"Url":"http://other.player.ri03.sycdn.kuwo.cn/369ab033f38575844e6ff8be4cec16b1/5ef9ec3a/resource/s2/36/7/336359786.flac","Md5":"","Name":"寸断","Artist":"本兮","MatchScore":1}

xiexiaosheng, 2020-06-29:
光猫拨号可以，路由器拨号不行。

xsm1997, 2020-06-30:
https://github.com/cnsilvan/UnblockNeteaseMusic/pull/23

已在原项目提交pr，等待合并后，再在本项目提交pr。

xsm1997, 2020-06-30:
也可自行修改/etc/init.d/unblockmusic，取消UnblockNeteaseMusic运行时的-e flag，应该可以暂时解决windows客户端的问题。

xiexiaosheng, 2020-06-30:
> [cnsilvan/UnblockNeteaseMusic#23](https://github.com/cnsilvan/UnblockNeteaseMusic/pull/23)
> 
> [pr，]，[pr]（窗口））

但是，我在光猫拨号可以，软路由拨号就不行。

hxlls, 2020-06-30:
全新重新新编译，全新不保留配置安装就行了，我遇到过

closed 2020-07-01 by coolsnowwolf

WROIATE, 2020-07-05:
> 也可自行修改/etc/init.d/unblockmusic，取消UnblockNeteaseMusic运行时的-e flag，应该可以暂时解决windows客户端的问题。

-e应该是为了解决ios的https，但是windows使用-e会出现问题，这个在原项目那可以查到