coolsnowwolf/lede #4899: 刷入新固件以后，cpu满载

ITdesk01, 2020-06-30:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：20200628编译的固件，刷入以后CPU开始满载，htop检查，发现有两个很吃cpu，一个trojan，一个docker

### (2) 路由器型号和固件版本

A：
X86路由
内核版本 | 4.19.122
OpenWrt R20.6.18 / LuCI Master (git-20.141.34051-f750b5f)

### (3) 详细日志

A：
docker系统日志
Tue Jun 30 20:45:43 2020 daemon.notice pppd[3155]: Connect: pppoe-wan <--> eth1
Tue Jun 30 20:45:43 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:43.892005120+08:00" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.btrfs" error="path /mnt/sdb3/docker/containerd/daemon/io.containerd.snapshotter.v1.btrfs must be a btrfs filesystem to be used with the btrfs snapshotter"
Tue Jun 30 20:45:43 2020 daemon.err modprobe: failed to find a module named aufs
Tue Jun 30 20:45:43 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:43.895985117+08:00" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.aufs" error="modprobe aufs failed: "": exit status 255"
Tue Jun 30 20:45:43 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:43.896553077+08:00" level=warning msg="could not use snapshotter zfs in metadata plugin" error="path /mnt/sdb3/docker/containerd/daemon/io.containerd.snapshotter.v1.zfs must be a zfs filesystem to be used with the zfs snapshotter: skip plugin"
Tue Jun 30 20:45:43 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:43.896566713+08:00" level=warning msg="could not use snapshotter btrfs in metadata plugin" error="path /mnt/sdb3/docker/containerd/daemon/io.containerd.snapshotter.v1.btrfs must be a btrfs filesystem to be used with the btrfs snapshotter"
Tue Jun 30 20:45:43 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:43.896581922+08:00" level=warning msg="could not use snapshotter aufs in metadata plugin" error="modprobe aufs failed: "": exit status 255"
Tue Jun 30 20:45:44 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:44.076207641+08:00" level=warning msg="Your kernel does not support swap memory limit"
Tue Jun 30 20:45:44 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:44.076294560+08:00" level=warning msg="Your kernel does not support cgroup cfs period"
Tue Jun 30 20:45:44 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:44.076306067+08:00" level=warning msg="Your kernel does not support cgroup cfs quotas"
Tue Jun 30 20:45:44 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:44.076315108+08:00" level=warning msg="Your kernel does not support cgroup blkio weight"
Tue Jun 30 20:45:44 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:44.076331210+08:00" level=warning msg="Your kernel does not support cgroup blkio weight_device"
Tue Jun 30 20:45:44 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:44.076340276+08:00" level=warning msg="Your kernel does not support cgroup blkio throttle.read_bps_device"
Tue Jun 30 20:45:44 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:44.076349292+08:00" level=warning msg="Your kernel does not support cgroup blkio throttle.write_bps_device"
Tue Jun 30 20:45:44 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:44.076363450+08:00" level=warning msg="Your kernel does not support cgroup blkio throttle.read_iops_device"
Tue Jun 30 20:45:44 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:44.076377565+08:00" level=warning msg="Your kernel does not support cgroup blkio throttle.write_iops_device"
Tue Jun 30 20:45:44 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:44.234272333+08:00" level=warning msg="Could not load necessary modules for IPSEC rules: protocol not supported"
Tue Jun 30 20:45:44 2020 daemon.info modprobe: Usage: modprobe [-q] filename
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.585696] IPv6: ADDRCONF(NETDEV_UP): docker0: link is not ready
Tue Jun 30 20:45:44 2020 daemon.err dockerd[3125]: time="2020-06-30T20:45:44.399310990+08:00" level=warning msg="Error (Unable to complete atomic operation, key modified) deleting object [endpoint 3a0737f954ac488f27dc279022a4b0b13f118415c5639b9fba741c8892e652df ca093969b2e55787d002be3cfce6d031cf82b3375863ac1cbbaa8aa0d4d39e35], retrying...."
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.854126] docker0: port 1(vethcf7cf3e) entered blocking state
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.854224] docker0: port 1(vethcf7cf3e) entered disabled state
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.854380] device vethcf7cf3e entered promiscuous mode
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.854639] IPv6: ADDRCONF(NETDEV_UP): vethcf7cf3e: link is not ready
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.854734] docker0: port 1(vethcf7cf3e) entered blocking state
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.854815] docker0: port 1(vethcf7cf3e) entered forwarding state
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.854930] docker0: port 1(vethcf7cf3e) entered disabled state
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.862556] docker0: port 2(veth0b5adc5) entered blocking state
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.862646] docker0: port 2(veth0b5adc5) entered disabled state
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.862775] device veth0b5adc5 entered promiscuous mode
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.862900] IPv6: ADDRCONF(NETDEV_UP): veth0b5adc5: link is not ready
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.863002] docker0: port 2(veth0b5adc5) entered blocking state
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.863089] docker0: port 2(veth0b5adc5) entered forwarding state
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.863274] IPv6: ADDRCONF(NETDEV_CHANGE): docker0: link becomes ready
Tue Jun 30 20:45:44 2020 kern.info kernel: [ 15.863472] docker0: port 2(veth0b5adc5) entered disabled state
Tue Jun 30 20:45:45 2020 kern.warn kernel: [ 16.471239] cgroup: runc (3488) created nested cgroup for controller "memory" which has incomplete hierarchy support. Nested cgroups may change behavior in the future.
Tue Jun 30 20:45:45 2020 kern.warn kernel: [ 16.471451] cgroup: "memory" requires setting use_hierarchy to 1 on the root
Tue Jun 30 20:45:45 2020 kern.info kernel: [ 16.664969] eth0: renamed from veth3bef81b
Tue Jun 30 20:45:45 2020 kern.info kernel: [ 16.764947] eth0: renamed from vethf102e9c
Tue Jun 30 20:45:45 2020 kern.info kernel: [ 16.804992] IPv6: ADDRCONF(NETDEV_CHANGE): vethcf7cf3e: link becomes ready
Tue Jun 30 20:45:45 2020 kern.info kernel: [ 16.805209] docker0: port 1(vethcf7cf3e) entered blocking state
Tue Jun 30 20:45:45 2020 kern.info kernel: [ 16.805345] docker0: port 1(vethcf7cf3e) entered forwarding state
Tue Jun 30 20:45:45 2020 kern.info kernel: [ 16.805594] IPv6: ADDRCONF(NETDEV_CHANGE): veth0b5adc5: link becomes ready
Tue Jun 30 20:45:45 2020 kern.info kernel: [ 16.805787] docker0: port 2(veth0b5adc5) entered blocking state
Tue Jun 30 20:45:45 2020 kern.info kernel: [ 16.805925] docker0: port 2(veth0b5adc5) entered forwarding state
[内核日志.txt](https://github.com/coolsnowwolf/lede/files/4852056/default.txt)
[系统日志.txt](https://github.com/coolsnowwolf/lede/files/4852058/default.txt)

hcym, 2020-07-01:
trojan不会吃力吧，128进程也就是瞬间满载，蜗牛是开64进程瞬间满载，很快就会恢复健康，docker是持续有点，我很少用，基本不编译或者关闭

coolsnowwolf, 2020-07-01:
环路了

ITdesk01, 2020-07-02:
> trojan不会吃力吧，128进程也就是瞬间满载，蜗牛是开64进程瞬间满载，很快就会恢复健康，docker是持续有点，我很少用，基本不编译或者关闭

最近才有这个问题。应该是修改了什么导致的

ITdesk01, 2020-07-02:
> 环路了

大雕你那边正常嘛

closed 2020-07-02 by coolsnowwolf