coolsnowwolf/lede #4904: openwrt如何釋放公網IP，獲取新的公網IP (WAN協議: DHCP客戶端)

hxhwing0522, 2020-07-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：HK的WAN協議不是PPOE撥號的，是依靠"DHCP客戶端"獲得新的公網IP。
以前用老毛子時，每次點擊"斷開網絡"或重啟路由，都會獲得新的公網IP。
不過刷了openwrt後，用了斷線﹑關掉WAN口再打開﹑重啟路由等各種方式，都是沒法獲得新的公網IP，一直都是同一個公網IP。所以希望大佬指點我應該如何才可以獲得釋放公網IP，獲取新的公網IP。

### (2) 路由器型号和固件版本

A：新路由3 D2

### (3) 详细日志

A：沒有日志
WAN口 ipv4怎弄都是同一個IP。

zerofo, 2020-07-01:
固定公网ip  好羡慕

hxhwing0522, 2020-07-01:
> 固定公网ip 好羡慕

不固定公網更好，隨時換IP。

zerofo, 2020-07-01:
> > 固定公网ip 好羡慕
> 
> 不固定公網更好，隨時換IP。

得閒無事，做乜要成日換

hxhwing0522, 2020-07-01:
> > > 固定公网ip 好羡慕
> > 
> > 
> > 不固定公網更好，隨時換IP。
> 
> 得閒無事，做乜要成日換

比如建節點 : )

zerofo, 2020-07-01:
> > > > 固定公网ip 好羡慕
> > > 
> > > 
> > > 不固定公網更好，隨時換IP。
> > 
> > 
> > 得閒無事，做乜要成日換
> 
> 比如建節點 : )

(>_<) MAC?

hxlls, 2020-07-01:
自己改MAC 啊，这个的DHCP SERVER 根据MAC 来分配的，你MAC变化后，IP自然变化了，如果不改MAC dhcp server 的缓存生效期间，你是没法获取新的IP地址的

hxhwing0522, 2020-07-01:
> 自己改MAC 啊，这个的DHCP SERVER 根据MAC 来分配的，你MAC变化后，IP自然变化了，如果不改MAC dhcp server 的缓存生效期间，你是没法获取新的IP地址的

接口 - WAN - 高級設置 - 重设 MAC 地址  這里嗎?
我改過來 好似不怎生效的樣子

zerofo, 2020-07-01:
> > 自己改MAC 啊，这个的DHCP SERVER 根据MAC 来分配的，你MAC变化后，IP自然变化了，如果不改MAC dhcp server 的缓存生效期间，你是没法获取新的IP地址的
> 
> 接口 - WAN - 高級設置 - 重设 MAC 地址  這里嗎?
> 我改過來 MAC也是沒有變化

隨機一個MAC, 自己改/etc/config/network

hxhwing0522, 2020-07-01:
> > > 自己改MAC 啊，这个的DHCP SERVER 根据MAC 来分配的，你MAC变化后，IP自然变化了，如果不改MAC dhcp server 的缓存生效期间，你是没法获取新的IP地址的
> > 
> > 
> > 接口 - WAN - 高級設置 - 重设 MAC 地址  這里嗎?
> > 我改過來 MAC也是沒有變化
> 
> 隨機一個MAC, 自己改/etc/config/network

嗯 
不過想看看大佬有沒有更簡單的方法 或將來可以一鍵釋放IP

acooler15, 2020-07-03:
写个shell脚本？从网上扒个随机MAC生成，然后用uci 命令设置wan_dev的MAC地址并保存，再然后用ifup重启WAN口🤣

closed 2020-07-04 by coolsnowwolf