coolsnowwolf/lede #4913: Intel 9260AC 无线网卡固件过新 驱动无法正常加载固件

Mandi-Sa, 2020-07-02:
反馈bug/问题模板，提建议请删除

Q：是否搜索了issue (使用 "x" 选择)
[x] 没有类似的issue

### (1) 具体问题

A：Intel 9260AC无线网卡无法被正常的加载固件，内核日志中显示无法加载固件。kmod-iwlwifi和iwlwifi-firmware-iwl9260已经被编译至固件中。观察/lib/firmware中，已经有iwlwifi-9260-th-b0-jf-b0-46.ucode，即v46版本的固件，但4.19内核最高支持v38版本的固件，故无法正常使用。

### (2) 路由器型号和固件版本

A：X86_64 Hyper-V虚拟机，无线网卡VT-d直通。

### (3) 详细日志

A：
root@OpenWrt:~# dmesg|grep iwl
[    6.366508] iwlwifi aa6b:00:00.0: can't derive routing for PCI INT A
[    6.366700] iwlwifi aa6b:00:00.0: PCI INT A: no GSI
[    6.458232] iwlwifi aa6b:00:00.0: Direct firmware load for iwlwifi-9260-th-b0-jf-b0-38.ucode failed with error -2
[    6.458484] iwlwifi aa6b:00:00.0: Falling back to syfs fallback for: iwlwifi-9260-th-b0-jf-b0-38.ucode
[    6.654138] firmware iwlwifi-9260-th-b0-jf-b0-38.ucode: firmware_loading_store: map pages failed
[    6.654516] iwlwifi aa6b:00:00.0: Direct firmware load for iwlwifi-9260-th-b0-jf-b0-37.ucode failed with error -2
[    6.654773] iwlwifi aa6b:00:00.0: Falling back to syfs fallback for: iwlwifi-9260-th-b0-jf-b0-37.ucode
[    6.657172] firmware iwlwifi-9260-th-b0-jf-b0-37.ucode: firmware_loading_store: map pages failed
[    6.657478] iwlwifi aa6b:00:00.0: Direct firmware load for iwlwifi-9260-th-b0-jf-b0-36.ucode failed with error -2
[    6.657751] iwlwifi aa6b:00:00.0: Falling back to syfs fallback for: iwlwifi-9260-th-b0-jf-b0-36.ucode
[    6.660056] firmware iwlwifi-9260-th-b0-jf-b0-36.ucode: firmware_loading_store: map pages failed
[    6.660390] iwlwifi aa6b:00:00.0: Direct firmware load for iwlwifi-9260-th-b0-jf-b0-35.ucode failed with error -2
[    6.660632] iwlwifi aa6b:00:00.0: Falling back to syfs fallback for: iwlwifi-9260-th-b0-jf-b0-35.ucode
[    6.662893] firmware iwlwifi-9260-th-b0-jf-b0-35.ucode: firmware_loading_store: map pages failed
[    6.663225] iwlwifi aa6b:00:00.0: Direct firmware load for iwlwifi-9260-th-b0-jf-b0-34.ucode failed with error -2
[    6.663467] iwlwifi aa6b:00:00.0: Falling back to syfs fallback for: iwlwifi-9260-th-b0-jf-b0-34.ucode
[    6.665755] firmware iwlwifi-9260-th-b0-jf-b0-34.ucode: firmware_loading_store: map pages failed
[    6.666076] iwlwifi aa6b:00:00.0: Direct firmware load for iwlwifi-9260-th-b0-jf-b0-33.ucode failed with error -2
[    6.666319] iwlwifi aa6b:00:00.0: Falling back to syfs fallback for: iwlwifi-9260-th-b0-jf-b0-33.ucode
[    6.668854] firmware iwlwifi-9260-th-b0-jf-b0-33.ucode: firmware_loading_store: map pages failed
[    6.669274] iwlwifi aa6b:00:00.0: Direct firmware load for iwlwifi-9260-th-b0-jf-b0-32.ucode failed with error -2
[    6.669523] iwlwifi aa6b:00:00.0: Falling back to syfs fallback for: iwlwifi-9260-th-b0-jf-b0-32.ucode
[    6.672018] firmware iwlwifi-9260-th-b0-jf-b0-32.ucode: firmware_loading_store: map pages failed
[    6.672391] iwlwifi aa6b:00:00.0: Direct firmware load for iwlwifi-9260-th-b0-jf-b0-31.ucode failed with error -2
[    6.672638] iwlwifi aa6b:00:00.0: Falling back to syfs fallback for: iwlwifi-9260-th-b0-jf-b0-31.ucode
[    6.674852] firmware iwlwifi-9260-th-b0-jf-b0-31.ucode: firmware_loading_store: map pages failed
[    6.675191] iwlwifi aa6b:00:00.0: Direct firmware load for iwlwifi-9260-th-b0-jf-b0-30.ucode failed with error -2
[    6.675435] iwlwifi aa6b:00:00.0: Falling back to syfs fallback for: iwlwifi-9260-th-b0-jf-b0-30.ucode
[    6.677585] firmware iwlwifi-9260-th-b0-jf-b0-30.ucode: firmware_loading_store: map pages failed
[    6.678002] iwlwifi aa6b:00:00.0: no suitable firmware found!
[    6.678142] iwlwifi aa6b:00:00.0: minimum version required: iwlwifi-9260-th-b0-jf-b0-30
[    6.678331] iwlwifi aa6b:00:00.0: maximum version supported: iwlwifi-9260-th-b0-jf-b0-38
[    6.678521] iwlwifi aa6b:00:00.0: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
root@OpenWrt:~#

Mandi-Sa, 2020-07-02:
通过下载openwrt.org官方的iwlwifi-firmware-iwl9260软件包，替换预编译的固件版本后可正常使用，但希望还是能解决这个问题，毕竟多一个手动的步骤后期更新系统比较麻烦。

附上正常情况的内核日志：
root@OpenWrt:~# dmesg|grep iwl
[    6.303934] iwlwifi aa6b:00:00.0: can't derive routing for PCI INT A
[    6.304094] iwlwifi aa6b:00:00.0: PCI INT A: no GSI
[    6.459533] iwlwifi aa6b:00:00.0: loaded firmware version 38.755cfdd8.0 op_mode iwlmvm
[    6.483107] iwlwifi aa6b:00:00.0: Detected Killer (R) Wireless-AC 1550 Wireless Network Adapter (9260NGW), REV=0x324
[    6.535928] iwlwifi aa6b:00:00.0: base HW address: 94:b8:6d:bd:31:39
[    6.604659] ieee80211 phy0: Selected rate control algorithm 'iwl-mvm-rs'
[    6.604915] iwlwifi aa6b:00:00.0 wlan0: no netdev found for vf serial:30515
[    8.800712] iwlwifi aa6b:00:00.0 wlan0: no netdev found for vf serial:30515
root@OpenWrt:~#

coolsnowwolf, 2020-07-02:
可以写个 Patch 发个 PR

Mandi-Sa, 2020-07-02:
已发PR

closed 2020-07-02 by coolsnowwolf