coolsnowwolf/lede #4914: 关于禁止解析 IPv6 DNS (AAAA) 记录的白名单问题之临时解决方案及不足

KumaTea, 2020-07-02:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 有类似的issue
  * #4749 

## 2. 详细叙述

### (1) 具体问题

A：希望设置解析 IPv6 DNS (AAAA) 记录的白名单，放行诸如PT站、路由器远程管理的动态IP等等。

---

原先希望通过添加 `server=/domain/1.2.3.4` 到 `额外的 SERVERS 文件` 这种方式解决，后来发现不成功，可能过滤 AAAA 发生在查询之后。

目前想出了一种临时解决方式。创建一个脚本，例如 `/usr/sbin/update-aaaa.sh`

```shell
host_file="/tmp/hosts-custom"
domains="npupt.com 4.npupt.com 6.npupt.com checkipv6.dyndns.com checkipv6.dyndns.com checkipv6.dyndns.com checkipv6.dyndns.com"
dns="223.5.5.5"
for domain in $domains; do
    ns_result=$(nslookup $domain $dns | tail -n 1)
    if [ -z "${ns_result##*Address*}" ];
    then
        result=$(echo $ns_result | cut -b 12-)
        echo "Add AAAA: $result for $domain to hosts file..."
        echo "$result $domain" >> $host_file
    else
        echo "ERROR: $domain cannot be found, ignored."
    fi
done
echo "Restarting dnsmasq..."
/etc/init.d/dnsmasq restart
echo "Done."
```

**[完整脚本](https://gist.github.com/KumaTea/03e948c1589cba7ec7235014272cca25)**

然后把 `/tmp/hosts-custom` 添加到 `额外的 HOSTS 文件` 里即可。

脚本设置 crontab 定时运行。

### 不足

* 无法应用于DDNS情景
* 无法动态读取更新文件，更新后必须重启 dnsmasq

---

有更好的解决方案吗？

hcym, 2020-07-02:
把它关闭取消吧，我试过，根本不影响代理，无论路由模式还是列表模式，感觉大雕对ipv6有成见，或者我用的插件少，只有ssr和haproxy，广告也是ssr内置的

KumaTea, 2020-07-02:
> **hcym**: 把它关闭取消吧，我试过，根本不影响代理，无论路由模式还是列表模式，感觉大雕对ipv6有成见，或者我用的插件少，只有ssr和haproxy，广告也是ssr内置的

这个感觉跟不同人的实际使用情况有关。

我上百科的时候总是优先解析到v6然后连不上，要刷新几次才连接到v4，关闭AAAA解析后有很大改善。

设置这个按钮相信也是根据大多数用户的使用环境 :thinking:

MeIsReallyBa, 2020-07-02:
v6和运营商有关系了。有的运营商v6 cdn不行用着体验就差

coolsnowwolf, 2020-07-02:
设置这个按钮相信也是根据大多数用户的使用环境 决定的

hcym, 2020-07-02:
一直都是把114换成移动分配的dns，我的使用也单纯，就是谷歌油管，国内就是淘宝微信，很少遇到稀奇古怪的问题😂

dongjuanyong, 2020-07-02:
自己动手丰衣足食，把过滤v6解析的活儿放在dnsmasq的上游，gfwlist模式就可以完美使用了，白名单模式还是容易误伤
于是上游的pdnsd必须换成能过滤v6解析的，用什么见仁见智（估计有人要吹sm***dns），我是用的dnscrypt-proxy

mjyhj, 2020-07-02:
感谢分享，暂时可以自己自定义添加域名使用了

coolsnowwolf, 2020-07-03:
新版本的 DNSMASQ 支持 检测 script 了，可以用新的方法

closed 2020-07-04 by coolsnowwolf

imkaihao, 2021-06-22:
因为每次路由器重启时会删除/tmp/hosts-custom，建议除了crontab定时运行脚本，开机时也执行脚本。

feifan1113, 2025-02-27:
感谢分享，根据我自己情况，搞了两个脚本，由于重启后前缀和解析的没及时更新，所以加了个本地提取ipv6，本机根据DUID提取地址，外网其他机用网络解析提取，在额外host那里填/tmp/hosts-custom-local和/tmp/hosts-custom-net
脚本放在/etc/config执行，
开机自启
sleep 300s && /etc/config/update-aaaa-local.sh
sleep 300s && /etc/config/update-aaaa-net.sh
计划任务
0 4 */3 * * /etc/config/update-aaaa-net.sh
0 5 */10 * * /etc/config/update-aaaa-local.sh
脚本有pw相关命令根据自己情况修改或删除
我用23.05版本的 op以及pw和.lucky做动态域名解析 关闭了slaac仅用dhcpv6分配 v6 的dns下发也关闭了
uci set dhcp.lan.ra_slaac=0
uci set dhcp.lan.dns_service="0"
uci set dhcp.lan.ra_dns="0"
uci commit dhcp
/etc/init.d/odhcpd restart
脚本后缀去掉txt
[update-aaaa-local.sh.txt](https://github.com/user-attachments/files/19001043/update-aaaa-local.sh.txt)

[update-aaaa-net.sh.txt](https://github.com/user-attachments/files/19001055/update-aaaa-net.sh.txt)