coolsnowwolf/lede #4923: 5.7内核的ubuntu20 编译报错，Command 'bin' not found

feifan1113, 2020-07-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
5.7内核的ubuntu20 编译报错，Command 'bin' not found

### (1) 具体问题

5.7内核的ubuntu20 编译报错，Command 'bin' not found

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：
make[1]: *** [package/Makefile:107: /home/zhs/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhs/lede'
make: *** [/home/zhs/lede/include/toplevel.mk:224: world] Error 2
zhs@zhs:~/lede$  *** [package/Makefile:107: /home/zhs/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2

Command 'bin' not found, did you mean:

  command 'bino' from deb bino (1.6.6-3build3)
  command 'din' from deb din (5.2.1-6build1)
  command 'bip' from deb bip (0.9.0~rc3-1)
  command 'bing' from deb bing (1.3.5-3)
  command 'ben' from deb ben (0.9.0ubuntu2)
  command 'tin' from deb tin (1:2.4.4-1build2)
  command 'win' from deb wily (0.13.41-8build1)
  command 'dbin' from deb dbmix (0.9.8-6.3ubuntu2)

Try: sudo apt install <deb name>

zhs@zhs:~/lede$ make[1]: Leaving directory '/home/zhs/lede'
make[1]:: command not found

dl250, 2020-07-04:
我也是Ubuntu20编译并没有报错

hcym, 2020-07-04:
自己升级内核的吗？我的2004和1804都正常，没在意内核版本，5.4应该是，这个和系统内核没什么事吧？

closed 2020-07-04 by coolsnowwolf