coolsnowwolf/lede #4924: nginx编译报错

bin20088, 2020-07-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：nginx编译报错

### (2) 路由器型号和固件版本

A：x64  n1
[image]

### (3) 详细日志

A：

menphy1989, 2020-07-03:
x86，一样情况

jinjin327, 2020-07-03:
x86，一样情况

EnnawYang, 2020-07-03:
x86 详细日志：
```
make[3]: Entering directory '/home/xw/openwrt/lede-autobuild/feeds/packages/net/nginx-util'
rm -f /home/xw/openwrt/lede-autobuild/build_dir/target-x86_64_musl/nginx-util-1.4/.built
touch /home/xw/openwrt/lede-autobuild/build_dir/target-x86_64_musl/nginx-util-1.4/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/xw/openwrt/lede-autobuild/build_dir/target-x86_64_musl/nginx-util-1.4=nginx-util-1.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/xw/openwrt/lede-autobuild/staging_dir/target-x86_64_musl/usr/include -I/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/xw/openwrt/lede-autobuild/build_dir/target-x86_64_musl/nginx-util-1.4=nginx-util-1.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/xw/openwrt/lede-autobuild/staging_dir/target-x86_64_musl/usr/include -I/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/xw/openwrt/lede-autobuild/staging_dir/target-x86_64_musl/usr/lib -L/home/xw/openwrt/lede-autobuild/staging_dir/target-x86_64_musl/lib -L/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro " make -j1 -C /home/xw/openwrt/lede-autobuild/build_dir/target-x86_64_musl/nginx-util-1.4/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/xw/openwrt/lede-autobuild/build_dir/target-x86_64_musl/nginx-util-1.4=nginx-util-1.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" CMAKE_COMMAND='/home/xw/openwrt/lede-autobuild/staging_dir/host/bin/cmake' CMAKE_DISABLE_cmake_check_build_system=1 ;
make[4]: Entering directory '/home/xw/openwrt/lede-autobuild/build_dir/target-x86_64_musl/nginx-util-1.4'
make[5]: Entering directory '/home/xw/openwrt/lede-autobuild/build_dir/target-x86_64_musl/nginx-util-1.4'
make[6]: Entering directory '/home/xw/openwrt/lede-autobuild/build_dir/target-x86_64_musl/nginx-util-1.4'
make[6]: Leaving directory '/home/xw/openwrt/lede-autobuild/build_dir/target-x86_64_musl/nginx-util-1.4'
make[6]: Entering directory '/home/xw/openwrt/lede-autobuild/build_dir/target-x86_64_musl/nginx-util-1.4'
[ 16%] Linking CXX executable nginx-ssl-util-nopcre
/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: CMakeFiles/nginx-ssl-util-nopcre.dir/nginx-util.cpp.o: in function `ubus::lock_shared_resources::get_context()::{lambda(ubus_context*)#1}::operator()(ubus_context*) const':
nginx-util-1.4/nginx-util-1.4/ubus-cxx.hpp:209: undefined reference to `ubus_free(ubus_context*)'
/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: CMakeFiles/nginx-ssl-util-nopcre.dir/nginx-util.cpp.o: in function `ubus::lock_shared_resources::get_blob_buf()::{lambda(blob_buf*)#1}::operator()(blob_buf*) const':
nginx-util-1.4/nginx-util-1.4/ubus-cxx.hpp:231: undefined reference to `blob_buf_free(blob_buf*)'
/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: CMakeFiles/nginx-ssl-util-nopcre.dir/nginx-util.cpp.o: in function `ubus::lock_shared_resources::get_context()':
nginx-util-1.4/nginx-util-1.4/ubus-cxx.hpp:211: undefined reference to `ubus_connect(char const*)'
/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: nginx-util-1.4/nginx-util-1.4/ubus-cxx.hpp:215: undefined reference to `ubus_connect(char const*)'
/home/xw/openwrt/lede-autobuild/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: CMakeFiles/nginx-ssl-util-nopcre.dir/nginx-util.cpp.o: in function `ubus::lock_shared_resources::get_context()::{lambda(ubus_context*)#1}::operator()(ubus_context*) const':
[... cut ...]
make[3]: Leaving directory '/home/xw/openwrt/lede-autobuild/feeds/packages/net/nginx-util'
time: package/feeds/packages/nginx-util/compile#0.17#0.05#0.22
make[2]: *** [package/Makefile:113: package/feeds/packages/nginx-util/compile] Error 2
make[2]: Leaving directory '/home/xw/openwrt/lede-autobuild'
make[1]: *** [package/Makefile:107: /home/xw/openwrt/lede-autobuild/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xw/openwrt/lede-autobuild'
make: *** [/home/xw/openwrt/lede-autobuild/include/toplevel.mk:224：world] 错误 2

```

closed 2020-07-04 by coolsnowwolf