coolsnowwolf/lede #4930: LUCI的未提交更改预览的返回界面的返回按钮链接是错的

ihipop, 2020-07-03:
LUCI的未提交更改预览的返回界面的返回按钮链接是错的

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在LUCI上做一些设置，提交但是不更改，点击右上角未提交的更改，底部左侧的返回界面的链接是错的。

以 `静态路由` 为例 路径为 `http://192.168.1.1/cgi-bin/luci/admin/network/routes`

从 `静态路由` 页面添加页面添加一个路由, 只点击 `保存` 按钮，这时候点击右上角未保存的配置的更改会显示数字

[image]

点进去，页面链接变成  ```http://192.168.1.1/cgi-bin/luci/admin/uci/changes?redir=admin%2fnetwork%2froutes``` 底下有个返回按钮 

[image]

按理说应该回到 `静态路由` 页面 `http://192.168.1.1/cgi-bin/luci/admin/network/routes`

点击返回按钮 实际跳转到了  `http://192.168.1.1/cgi-bin/luci/admin/uci/admin/network/routes`

这里建议 如果拼参数麻烦，干脆用`history.go(-1)`算了

### (2) 路由器型号和固件版本

A：小娱C5 （应该是型号无关的）R20.7.1

coolsnowwolf, 2020-07-03:
已修复

closed 2020-07-03 by coolsnowwolf

ihipop, 2020-07-03:
https://github.com/coolsnowwolf/luci/commit/4093ead5df1a4888888856abe6c4b4ef9a519951

ihipop, 2020-07-05:
> 已修复

有一个小小的强迫症，`<%=controller%>` 本身是带`/`的，然后后面还带了一个`/` 点返回后就有两个`/`,要不我提个PR改下？