coolsnowwolf/lede #4939: pppoe拨号成功但是无法上网

freesilent, 2020-07-03:
反馈bug/问题模板，提建议请删除

## 1.pppoe拨号成功，wan获取到了正常的ip地址，但是无法打开网页，网络诊断也是失败

找到了一个类似的 https://github.com/coolsnowwolf/lede/issues/4554
但是这个问题是接入的设备可以上网，op自身无法上网，而且我找不到clash这个软件包，启动项里也没有clash

## 2. 详细叙述

### (1) 具体问题
[image]
前几天编译好的，稳定用了几天了。前天关机，昨天开机就这种情况了。因为是虚拟机，用编译好的原始镜像替换用过的镜像，打开依然这种情况。某次尝试打开dns加速，曾经可以正常上网，但是上传了备份的设置后再次无法上网，并且再次替换原始镜像，无法复现这种情况。
[image]
接入的设备ping路由无问题，ping拨号得到的ip也无问题，ping其他失败，提示无法连到端口。关闭防火墙依然无法上网。

### (2) 路由器型号和固件版本
硬件型号：Ryzen 3 2200G @3500Mhz 1 Core 1 Thread 
固件型号：OpenWrt SNAPSHOT， r2602-ba8a697b
内核版本：4.19.122
处理器架构：x86_zen

### (3) 详细日志
`Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.285434] ata1: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe100 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.285562] ata2: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe180 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.285688] ata3: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe200 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.285814] ata4: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe280 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.285941] ata5: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe300 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.286067] ata6: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe380 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.286194] ata7: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe400 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.286319] ata8: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe480 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.286447] ata9: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe500 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.286574] ata10: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe580 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.286707] ata11: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe600 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.286840] ata12: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe680 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.286972] ata13: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe700 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.287106] ata14: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe780 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.287239] ata15: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe800 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.287370] ata16: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe880 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.287503] ata17: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe900 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.287636] ata18: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fe980 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.287769] ata19: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fea00 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.287902] ata20: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fea80 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.288035] ata21: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5feb00 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.288167] ata22: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5feb80 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.288300] ata23: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fec00 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.288433] ata24: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fec80 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.288566] ata25: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fed00 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.288699] ata26: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fed80 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.288833] ata27: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fee00 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.288965] ata28: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fee80 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.289098] ata29: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fef00 irq 27
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.289230] ata30: SATA max UDMA/133 abar m4096@0xfd5fe000 port 0xfd5fef80 irq 27
Fri Jul  3 20:57:10 2020 kern.debug kernel: [    0.289390] ata_piix 0000:00:07.1: version 2.13
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.289924] scsi host31: ata_piix
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.290084] scsi host32: ata_piix
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.290166] ata31: PATA max UDMA/33 cmd 0x1f0 ctl 0x3f6 bmdma 0x1060 irq 14
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.290287] ata32: PATA max UDMA/33 cmd 0x170 ctl 0x376 bmdma 0x1068 irq 15
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.290487] VMware vmxnet3 virtual NIC driver - version 1.4.16.0-k-NAPI
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.290615] vmxnet3 0000:0b:00.0: # of Tx queues : 1, # of Rx queues : 1
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.291422] vmxnet3 0000:0b:00.0 eth0: NIC Link is Up 10000 Mbps
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.291543] vmxnet3 0000:13:00.0: # of Tx queues : 1, # of Rx queues : 1
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.292229] vmxnet3 0000:13:00.0 eth1: NIC Link is Up 10000 Mbps
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.292349] vmxnet3 0000:1b:00.0: # of Tx queues : 1, # of Rx queues : 1
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.293047] vmxnet3 0000:1b:00.0 eth2: NIC Link is Up 10000 Mbps
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.293160] hv_vmbus: registering driver hv_netvsc
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.293244] Fusion MPT base driver 3.04.20
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.293319] Copyright (c) 1999-2008 LSI Corporation
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.293409] Fusion MPT SPI Host driver 3.04.20
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.293498] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
Fri Jul  3 20:57:10 2020 kern.info kernel: [    0.293610] ehci-pci: EHCI PCI platform driver
[... 581 lines cut ...]
Fri Jul  3 20:57:16 2020 daemon.info netdata[4028]: Initializing file /var/cache/netdata/disk_inodes._boot/reserved_for_root.db.
Fri Jul  3 20:57:16 2020 daemon.info netdata[4028]: Initializing file /var/cache/netdata/disk_space._dev/main.db.
Fri Jul  3 20:57:16 2020 daemon.info netdata[4028]: Initializing file /var/cache/netdata/disk_space._dev/avail.db.
Fri Jul  3 20:57:16 2020 daemon.info netdata[4028]: Initializing file /var/cache/netdata/disk_space._dev/used.db.
Fri Jul  3 20:57:16 2020 daemon.info netdata[4028]: Initializing file /var/cache/netdata/disk_space._dev/reserved_for_root.db.
Fri Jul  3 20:57:16 2020 daemon.info netdata[4028]: Initializing file /var/cache/netdata/disk_inodes._dev/main.db.
Fri Jul  3 20:57:16 2020 daemon.info netdata[4028]: Initializing file /var/cache/netdata/disk_inodes._dev/avail.db.
Fri Jul  3 20:57:16 2020 daemon.info netdata[4028]: Initializing file /var/cache/netdata/disk_inodes._dev/used.db.
Fri Jul  3 20:57:16 2020 daemon.info netdata[4028]: Initializing file /var/cache/netdata/disk_inodes._dev/reserved_for_root.db.
Fri Jul  3 20:57:16 2020 daemon.info netdata[4028]: Initializing file /var/cache/netdata/netdata.web_thread1_cpu/main.db.
Fri Jul  3 20:57:19 2020 daemon.warn netifd: Failed to assign requested subprefix of size 60 for lan2, trying other
Fri Jul  3 20:57:19 2020 daemon.warn netifd: Failed to assign subprefix of size 65 for lan2
Fri Jul  3 20:57:19 2020 daemon.info avahi-daemon[3768]: Leaving mDNS multicast group on interface br-lan.IPv6 with address fe80::20c:29ff:fee8:7459.
Fri Jul  3 20:57:19 2020 daemon.info avahi-daemon[3768]: Joining mDNS multicast group on interface br-lan.IPv6 with address 2408:8207:24af:6440::1.
Fri Jul  3 20:57:19 2020 daemon.info avahi-daemon[3768]: Registering new address record for 2408:8207:24af:6440::1 on br-lan.*.
Fri Jul  3 20:57:19 2020 daemon.info avahi-daemon[3768]: Withdrawing address record for fe80::20c:29ff:fee8:7459 on br-lan.
Fri Jul  3 20:57:19 2020 daemon.info : 16[KNL] 2408:8206:24a1:9d8c:b432:d4e0:d6df:f42f appeared on pppoe-wan
Fri Jul  3 20:57:19 2020 daemon.info : 06[KNL] 2408:8207:24af:6440::1 appeared on br-lan
[... cut ...]
Fri Jul  3 20:59:42 2020 authpriv.info dropbear[16049]: Child connection from 192.168.0.101:62939
Fri Jul  3 20:59:46 2020 authpriv.notice dropbear[16049]: Password auth succeeded for 'root' from 192.168.0.101:62939
Fri Jul  3 21:02:05 2020 authpriv.info dropbear[25976]: Child connection from 192.168.0.101:63198
Fri Jul  3 21:02:06 2020 authpriv.warn dropbear[25976]: Bad password attempt for 'root' from 192.168.0.101:63198
Fri Jul  3 21:02:09 2020 authpriv.notice dropbear[25976]: Password auth succeeded for 'root' from 192.168.0.101:63198
`

coolsnowwolf, 2020-07-03:
两个 端口 IP 重复了

Erope, 2020-07-03:
LAN两个端口重复
另外检查光猫和路由IP是不是也冲突了

freesilent, 2020-07-03:
> 
> 
> 两个 端口 IP 重复了

两个lan是桥接的，ip应该设为一样吧。一直怎么设置。

@Erope 光猫已设为桥接。

freesilent, 2020-07-03:
截图的是最开始出问题的镜像，现在的虚拟机已经不是这样了。
贴上现在的网络设置
[image]
这是编译成功后保存的原镜像，接入虚拟机之后只修改过网络设置，其他未改动。依然是wan口正常，但是无法上网，网络测试失败。

9l, 2020-07-04:
你 eth1 eth2 都桥接了，你 lan2 有何用，删掉

closed 2020-07-04 by coolsnowwolf