coolsnowwolf/lede #4941: 更新后x86编译报错

Margular, 2020-07-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* 有类似的issue但是不是同一个问题

## 2. 详细叙述

### (1) 具体问题

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for base-files:
 * 	libubox20170601
 * opkg_install_cmd: Cannot install package base-files.
 * check_data_file_clashes: Package libustream-openssl20200215 wants to install file /home/parallels/sources/lede/build_dir/target-x86_64_musl/root-x86/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls20200215
 * opkg_install_cmd: Cannot install package libustream-openssl20200215.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for urngd:
 * 	libubox20170601
 * opkg_install_cmd: Cannot install package urngd.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/parallels/sources/lede'
package/Makefile:108: recipe for target '/home/parallels/sources/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/parallels/sources/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/parallels/sources/lede'
/home/parallels/sources/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

### (2) 路由器型号和固件版本

x86_64 generic

liwenjie119, 2020-07-03:
更新并make clean再编译试试，我更新到最新源码后makeclean后没问题了

kl9800, 2020-07-04:
请问最终解决了吗？我也是这个问题

asushugo, 2020-07-04:
实测make clean后再编译就通过了

Margular, 2020-07-04:
最后我没编译这个了，可能是ssr和vssr冲突了

closed 2020-07-04 by Margular