coolsnowwolf/lede #4944: wan口Obtain IPv6-Address设置为自动时，系统日志报错

hyy-666, 2020-07-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题

A：在WAN口进行PPPoe拨号同时获取IPv6地址的话若将Obtain IPv6-Address设置为自动，系统日志输出反复提示wan 6_4掉线重连（wan 6_4在网络配置中不可见，查询资料可知时用于在PPPoe链路上获取IPv6地址的端口）

### (2) 路由器型号和固件版本

A：OpenWrt x86_64 R20.7.1

### (3) 详细日志

A：
Sat Jul  4 01:56:46 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:56:46 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:56:49 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:56:49 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:56:52 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:56:52 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:56:55 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:56:55 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:56:58 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:56:58 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:57:01 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:57:01 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:57:04 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:57:04 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:57:07 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:57:07 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:57:11 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:57:11 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:57:14 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:57:14 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:57:17 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:57:17 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:57:20 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:57:20 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:57:23 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:57:23 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:57:26 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:57:26 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Sat Jul  4 01:57:29 2020 daemon.notice netifd: Interface 'wan_6_4' is now down
Sat Jul  4 01:57:29 2020 daemon.notice netifd: Interface 'wan_6_4' is setting up now
下为截图
[image]
希望大佬有空能修一下这个Bug，万分感激

closed 2020-07-04 by coolsnowwolf

XuCpeng, 2021-12-26:
同样的问题。配置ipv6中继时也出现了 `Interface 'wan_6_4' is now down` `Interface 'wan_6_4' is setting up now`系统日志刷屏的现象，每三秒一次。目前无法解决或绕过

Cineray, 2023-05-06:
同样的问题，至今也没有解决。。。

sunyink, 2023-06-02:
我也遇到，我重新编译把6 to 4什么的全取消了 只留下被依赖的6in4，刷进去这会儿不报了。

hyy-666, 2023-06-02:
> 

emmm，我很久之前就没有这个问题了，没想到这么多跟帖，之前大雕在群里给我说的，但是好久了。现在忘了怎么回事了。

Cineray, 2023-06-09:
> 同样的问题，至今也没有解决。。。

已经解决了，基本就是协议包的问题，具体解决方法我po到我的blog了：[解决 OpenWrt 中 “Interface wan_6_4” 不断重启的问题](https://blog.cineray.net/fix-openwrt-interface-wan_6_4-reboot-constantly.html)