coolsnowwolf/lede #4946: 5.4版的config下没有默认的network、system等配置，

cokyrain, 2020-07-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
5.4版的config下没有默认的network配置
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：一、最新5.4版直接在pe里刷好后，启动大概9秒的时候，加载了网卡后，所有网口全部失效(网口灯也不亮)。vi /etc/config/network 内容为空，需要手动写入lan口配置后重启，网口才正常。
。二、进入web后台,/cgi-bin/luci/admin/system/system界面，系统属性一项丢失，内容不显示，主机名显示(none)，以前备份的\etc\config\system文件用WinSCP上传后，才正常显示

### (2) 路由器型号和固件版本

A：X86,0703

### (3) 详细日志

A：

cokyrain, 2020-07-04:
附配置：
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_GRUB_TIMEOUT="2"
CONFIG_PACKAGE_6in4=y
# CONFIG_PACKAGE_amdgpu-firmware is not set
# CONFIG_PACKAGE_boost is not set
CONFIG_PACKAGE_dnsmasq_full_auth=y
CONFIG_PACKAGE_dnsmasq_full_conntrack=y
CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_dnsmasq_full_dnssec=y
CONFIG_PACKAGE_dnsmasq_full_noid=y
# CONFIG_PACKAGE_etherwake is not set
# CONFIG_PACKAGE_ipt2socks is not set
# CONFIG_PACKAGE_iptables-mod-ipsec is not set
CONFIG_PACKAGE_ipv6helper=y
# CONFIG_PACKAGE_kmod-crypto-cbc is not set
# CONFIG_PACKAGE_kmod-crypto-deflate is not set
# CONFIG_PACKAGE_kmod-crypto-des is not set
# CONFIG_PACKAGE_kmod-crypto-echainiv is not set
# CONFIG_PACKAGE_kmod-crypto-hmac is not set
# CONFIG_PACKAGE_kmod-crypto-md5 is not set
# CONFIG_PACKAGE_kmod-crypto-sha1 is not set
# CONFIG_PACKAGE_kmod-drm-amdgpu is not set
# CONFIG_PACKAGE_kmod-drm-kms-helper is not set
# CONFIG_PACKAGE_kmod-drm-ttm is not set
# CONFIG_PACKAGE_kmod-fb is not set
# CONFIG_PACKAGE_kmod-fb-cfb-copyarea is not set
# CONFIG_PACKAGE_kmod-fb-cfb-fillrect is not set
# CONFIG_PACKAGE_kmod-fb-cfb-imgblt is not set
# CONFIG_PACKAGE_kmod-fb-sys-fops is not set
# CONFIG_PACKAGE_kmod-fb-sys-ram is not set
# CONFIG_PACKAGE_kmod-ipsec is not set
# CONFIG_PACKAGE_kmod-ipt-ipsec is not set
CONFIG_PACKAGE_kmod-ipt-nat6=y
CONFIG_PACKAGE_kmod-iptunnel=y
# CONFIG_PACKAGE_kmod-iptunnel6 is not set
CONFIG_PACKAGE_kmod-nf-nat6=y
CONFIG_PACKAGE_kmod-phy-realtek=y
CONFIG_PACKAGE_kmod-r8169=y
CONFIG_PACKAGE_kmod-sit=y
# CONFIG_PACKAGE_kmod-tun is not set
# CONFIG_PACKAGE_libcap-ng is not set
# CONFIG_PACKAGE_libevent2 is not set
# CONFIG_PACKAGE_libminiupnpc is not set
# CONFIG_PACKAGE_libnatpmp is not set
CONFIG_PACKAGE_libnetfilter-conntrack=y
CONFIG_PACKAGE_libnettle=y
CONFIG_PACKAGE_libnfnetlink=y
# CONFIG_PACKAGE_luci-app-accesscontrol is not set
# CONFIG_PACKAGE_luci-app-arpbind is not set
# CONFIG_PACKAGE_luci-app-ipsec-vpnd is not set
CONFIG_PACKAGE_luci-app-netdata=y
# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Redsocks2 is not set
# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server is not set
# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Trojan is not set
CONFIG_PACKAGE_luci-app-udpxy=y
# CONFIG_PACKAGE_luci-app-upnp is not set
# CONFIG_PACKAGE_luci-app-vlmcsd is not set
# CONFIG_PACKAGE_luci-app-webadmin is not set
# CONFIG_PACKAGE_luci-app-wol is not set
# CONFIG_PACKAGE_luci-app-xlnetacc is not set
# CONFIG_PACKAGE_luci-app-zerotier is not set
CONFIG_PACKAGE_luci-i18n-netdata-zh-cn=y
CONFIG_PACKAGE_luci-proto-ipv6=y
# CONFIG_PACKAGE_miniupnpd is not set
CONFIG_PACKAGE_netdata=y
CONFIG_PACKAGE_odhcp6c=y
CONFIG_PACKAGE_odhcp6c_ext_cer_id=0
CONFIG_PACKAGE_odhcpd-ipv6only=y
CONFIG_PACKAGE_odhcpd_ipv6only_ext_cer_id=0
CONFIG_PACKAGE_r8169-firmware=y
# CONFIG_PACKAGE_redsocks2 is not set
# CONFIG_PACKAGE_shadowsocksr-libev-server is not set
# CONFIG_PACKAGE_strongswan is not set
# CONFIG_PACKAGE_trojan is not set
CONFIG_PACKAGE_udpxy=y
# CONFIG_PACKAGE_vlmcsd is not set
# CONFIG_PACKAGE_wol is not set
# CONFIG_PACKAGE_zerotier is not set
CONFIG_TARGET_IMAGES_GZIP=y
# CONFIG_VMDK_IMAGES is not set

OleStore, 2020-07-04:
确实有这问题

martin0607, 2020-07-04:
先关注了，刚编译好还没刷。能不能直接预置好network文件到源码临时解决一下？

OleStore, 2020-07-04:
刷回4.19内核了，等稳定了再编译

closed 2020-07-04 by coolsnowwolf