coolsnowwolf/lede #4955: 一个小BUG

wilson-chan2019, 2020-07-04:
路由器每次重启后，在电脑那里查看活动的网络，名字都会变一次（不是SSID变化），比如我的wifi名称是OpenWRT_5G，第一次重启后变成OpenWRT_5G 2，第二次就变成OpenWRT_5G 3，第三次就变成OpenWRT_5G 4，以此类推。有线也是这样。。而刷回旧版本，不会有这样的问题。

[image]
[image]
[image]
[image]

sxx1314, 2020-07-04:
这是因为mac地址在不断变化   你可以注意下每次变化时的mac地址  尝试在config下固化mac  估计是驱动的问题  应外别把模板删除  详细的软硬件参数和log请贴全

wilson-chan2019, 2020-07-04:
补充一下：
主机型号： ASUS RT-ACRH17

固件版本：OpenWrt R20.7.1 / LuCI Master (git-20.185.57036-e5309d2)

wilson-chan2019, 2020-07-04:
> 这是因为mac地址在不断变化 你可以注意下每次变化时的mac地址 尝试在config下固化mac 估计是驱动的问题 另外别把模板删除 详细的软硬件参数和log请贴全

确实是每次重启LAN的MAC地址变化了。

wilson-chan2019, 2020-07-04:
系统日志：

Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Sat Jul  4 19:07:39 2020 kern.notice kernel: [    0.000000] Linux version 5.4.48 (wilson@VMUB) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2746-90e48a34b)) #0 SMP Fri Jul 3 16:13:28 2020
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] OF: fdt: Machine model: ASUS RT-ACRH17
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Sat Jul  4 19:07:39 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 65024
Sat Jul  4 19:07:39 2020 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Sat Jul  4 19:07:39 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sat Jul  4 19:07:39 2020 kern.debug kernel: [    0.000000]   Normal zone: 65024 pages, LIFO batch:15
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s30028 r8192 d23220 u61440
Sat Jul  4 19:07:39 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s30028 r8192 d23220 u61440 alloc=15*4096
Sat Jul  4 19:07:39 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64448
Sat Jul  4 19:07:39 2020 kern.notice kernel: [    0.000000] Kernel command line: root_rfs=0x00000000 flash_type=nand  ubi.mtd=UBI_DEV
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] Memory: 248636K/260096K available (5538K kernel code, 183K rwdata, 1456K rodata, 1024K init, 231K bss, 11460K reserved, 0K cma-reserved, 0K highmem)
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Sat Jul  4 19:07:39 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x2f0/0x494 with crng_init=0
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000021] Switching to timer-based delay loop, resolution 20ns
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000217] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000232] pid_max: default: 32768 minimum: 301
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000419] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.000436] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.001412] CPU: Testing write buffer coherency: ok
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.002405] Setting up static identity map for 0x80300000 - 0x80300060
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.002538] rcu: Hierarchical SRCU implementation.
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.002970] smp: Bringing up secondary CPUs ...
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.005977] smp: Brought up 1 node, 4 CPUs
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.005997] SMP: Total of 4 processors activated (384.00 BogoMIPS).
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.006005] CPU: All CPU(s) started in SVC mode.
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.010115] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.010266] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.010290] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.010551] pinctrl core: initialized pinctrl subsystem
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.011584] NET: Registered protocol family 16
Sat Jul  4 19:07:39 2020 kern.info kernel: [    0.011899] DMA: preallocated 256 KiB pool for atomic coherent allocations
[... cut ...]
Sat Jul  4 19:11:40 2020 daemon.info hostapd: wlan0: STA d8:c7:71:8c:ed:51 RADIUS: starting accounting session 355E64DD59DAEDCF
Sat Jul  4 19:11:40 2020 daemon.info hostapd: wlan0: STA d8:c7:71:8c:ed:51 WPA: pairwise key handshake completed (RSN)
Sat Jul  4 19:11:41 2020 daemon.warn odhcpd[1424]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sat Jul  4 19:11:44 2020 kern.warn kernel: [   80.025893] ath10k_pci 0000:01:00.0: Invalid VHT mcs 15 peer stats

wilson-chan2019, 2020-07-04:
内核日志：

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.48 (wilson@VMUB) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2746-90e48a34b)) #0 SMP Fri Jul 3 16:13:28 2020
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: ASUS RT-ACRH17
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 65024
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65024 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30028 r8192 d23220 u61440
[    0.000000] pcpu-alloc: s30028 r8192 d23220 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64448
[    0.000000] Kernel command line: root_rfs=0x00000000 flash_type=nand  ubi.mtd=UBI_DEV
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 248636K/260096K available (5538K kernel code, 183K rwdata, 1456K rodata, 1024K init, 231K bss, 11460K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2f0/0x494 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000021] Switching to timer-based delay loop, resolution 20ns
[    0.000217] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000232] pid_max: default: 32768 minimum: 301
[    0.000419] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000436] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001412] CPU: Testing write buffer coherency: ok
[    0.002405] Setting up static identity map for 0x80300000 - 0x80300060
[    0.002538] rcu: Hierarchical SRCU implementation.
[    0.002970] smp: Bringing up secondary CPUs ...
[    0.005977] smp: Brought up 1 node, 4 CPUs
[    0.005997] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.006005] CPU: All CPU(s) started in SVC mode.
[    0.010115] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.010266] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.010290] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.010551] pinctrl core: initialized pinctrl subsystem
[    0.011584] NET: Registered protocol family 16
[    0.011899] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.013064] cpuidle: using governor ladder
[    0.013109] cpuidle: using governor menu
[    0.031889] cryptd: max_cpu_qlen set to 1000
[    0.035225] usbcore: registered new interface driver usbfs
[    0.035292] usbcore: registered new interface driver hub
[    0.035381] usbcore: registered new device driver usb
[    0.035430] pps_core: LinuxPPS API ver. 1 registered
[    0.035439] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.035465] PTP clock support registered
[    0.037008] clocksource: Switched to clocksource arch_sys_counter
[    0.037820] thermal_sys: Registered thermal governor 'step_wise'
[    0.038104] NET: Registered protocol family 2
[    0.038876] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.038918] TCP established hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.038956] TCP bind hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.039007] TCP: Hash tables configured (established 2048 bind 2048)
[    0.039125] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.039162] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.039388] NET: Registered protocol family 1
[    0.039429] PCI: CLS 0 bytes, default 64
[    0.041333] workingset: timestamp_bits=14 max_order=16 bucket_order=2
[    0.046703] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[... 249 lines cut ...]
[   42.342724] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   42.343955] ath10k_pci 0000:01:00.0: wmi print 'free: 84856 iram: 13140 sram: 11224'
[   42.728474] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   42.744814] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   42.751583] br-lan: port 2(wlan0) entered blocking state
[   42.751628] br-lan: port 2(wlan0) entered disabled state
[   42.756493] device wlan0 entered promiscuous mode
[... cut ...]
[   45.248105] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   45.249976] br-lan: port 3(wlan1) entered blocking state
[   45.253681] br-lan: port 3(wlan1) entered forwarding state
[   45.640024] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   45.642320] br-lan: port 2(wlan0) entered blocking state
[   45.645713] br-lan: port 2(wlan0) entered forwarding state
[   48.089101] pppoe-wan: renamed from ppp0
[   80.025893] ath10k_pci 0000:01:00.0: Invalid VHT mcs 15 peer stats

wilson-chan2019, 2020-07-04:
系统日志：

Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Sat Jul  4 19:18:22 2020 kern.notice kernel: [    0.000000] Linux version 5.4.48 (wilson@VMUB) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2746-90e48a34b)) #0 SMP Fri Jul 3 16:13:28 2020
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] OF: fdt: Machine model: ASUS RT-ACRH17
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Sat Jul  4 19:18:22 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 65024
Sat Jul  4 19:18:22 2020 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Sat Jul  4 19:18:22 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sat Jul  4 19:18:22 2020 kern.debug kernel: [    0.000000]   Normal zone: 65024 pages, LIFO batch:15
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s30028 r8192 d23220 u61440
Sat Jul  4 19:18:22 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s30028 r8192 d23220 u61440 alloc=15*4096
Sat Jul  4 19:18:22 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64448
Sat Jul  4 19:18:22 2020 kern.notice kernel: [    0.000000] Kernel command line: root_rfs=0x00000000 flash_type=nand  ubi.mtd=UBI_DEV
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] Memory: 248636K/260096K available (5538K kernel code, 183K rwdata, 1456K rodata, 1024K init, 231K bss, 11460K reserved, 0K cma-reserved, 0K highmem)
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Sat Jul  4 19:18:22 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x2f0/0x494 with crng_init=0
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000021] Switching to timer-based delay loop, resolution 20ns
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000216] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000232] pid_max: default: 32768 minimum: 301
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000421] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.000438] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.001414] CPU: Testing write buffer coherency: ok
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.002412] Setting up static identity map for 0x80300000 - 0x80300060
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.002545] rcu: Hierarchical SRCU implementation.
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.002972] smp: Bringing up secondary CPUs ...
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.005973] smp: Brought up 1 node, 4 CPUs
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.005992] SMP: Total of 4 processors activated (384.00 BogoMIPS).
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.006000] CPU: All CPU(s) started in SVC mode.
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.010112] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.010263] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.010288] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.010545] pinctrl core: initialized pinctrl subsystem
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.011573] NET: Registered protocol family 16
Sat Jul  4 19:18:22 2020 kern.info kernel: [    0.011886] DMA: preallocated 256 KiB pool for atomic coherent allocations
[... cut ...]
Sat Jul  4 19:18:43 2020 daemon.warn netifd: You have delegated IPv6-prefixes but haven't assigned them to any interface. Did you forget to set option ip6assign on your lan-interfaces?
Sat Jul  4 19:18:43 2020 daemon.notice netifd: Interface 'wan_6' is now up
Sat Jul  4 19:18:43 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sat Jul  4 19:18:44 2020 daemon.warn odhcpd[1422]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sat Jul  4 19:20:10 2020 daemon.warn odhcpd[1422]: A default route is present but there is no public prefix on lan thus we don't announce a default route!

wilson-chan2019, 2020-07-04:
内核日志：
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.48 (wilson@VMUB) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2746-90e48a34b)) #0 SMP Fri Jul 3 16:13:28 2020
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: ASUS RT-ACRH17
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 65024
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65024 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30028 r8192 d23220 u61440
[    0.000000] pcpu-alloc: s30028 r8192 d23220 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64448
[    0.000000] Kernel command line: root_rfs=0x00000000 flash_type=nand  ubi.mtd=UBI_DEV
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 248636K/260096K available (5538K kernel code, 183K rwdata, 1456K rodata, 1024K init, 231K bss, 11460K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2f0/0x494 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000021] Switching to timer-based delay loop, resolution 20ns
[    0.000216] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000232] pid_max: default: 32768 minimum: 301
[    0.000421] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000438] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001414] CPU: Testing write buffer coherency: ok
[    0.002412] Setting up static identity map for 0x80300000 - 0x80300060
[    0.002545] rcu: Hierarchical SRCU implementation.
[    0.002972] smp: Bringing up secondary CPUs ...
[    0.005973] smp: Brought up 1 node, 4 CPUs
[    0.005992] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.006000] CPU: All CPU(s) started in SVC mode.
[    0.010112] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.010263] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.010288] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.010545] pinctrl core: initialized pinctrl subsystem
[    0.011573] NET: Registered protocol family 16
[    0.011886] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.013042] cpuidle: using governor ladder
[    0.013086] cpuidle: using governor menu
[    0.031872] cryptd: max_cpu_qlen set to 1000
[    0.035109] usbcore: registered new interface driver usbfs
[    0.035174] usbcore: registered new interface driver hub
[    0.035262] usbcore: registered new device driver usb
[    0.035313] pps_core: LinuxPPS API ver. 1 registered
[    0.035321] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.035348] PTP clock support registered
[    0.036897] clocksource: Switched to clocksource arch_sys_counter
[    0.037707] thermal_sys: Registered thermal governor 'step_wise'
[    0.037995] NET: Registered protocol family 2
[    0.038767] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.038810] TCP established hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.038847] TCP bind hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.038898] TCP: Hash tables configured (established 2048 bind 2048)
[    0.039019] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.039055] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.039279] NET: Registered protocol family 1
[    0.039322] PCI: CLS 0 bytes, default 64
[    0.041226] workingset: timestamp_bits=14 max_order=16 bucket_order=2
[    0.046599] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.046614] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[... 247 lines cut ...]
[   41.688295] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[   41.771850] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   41.773085] ath10k_pci 0000:01:00.0: wmi print 'free: 84856 iram: 13140 sram: 11224'
[   42.159122] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   42.169778] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[... cut ...]
[   45.068430] ath10k_ahb a000000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   45.082516] br-lan: port 3(wlan1) entered blocking state
[   45.082557] br-lan: port 3(wlan1) entered disabled state
[   45.087480] device wlan1 entered promiscuous mode
[   45.394889] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   45.395522] br-lan: port 3(wlan1) entered blocking state
[   45.400459] br-lan: port 3(wlan1) entered forwarding state
[   47.825957] pppoe-wan: renamed from ppp0

closed 2020-07-04 by coolsnowwolf