coolsnowwolf/lede #4956: R169AC编译不通过

Cathgao, 2020-07-04:
## 1.关于你要提交的问题

竞斗云2.0从7月2号开始云编译不通过

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

action配置没修改，但从7月2号开始编译一直失败。下载log后查看提示是image too big，root.squashfs大小为36MB，但竞斗云的flash有128MB大小，而且之前编译成功的固件也有接近40MB的大小，容量应该是足够的

### (2) 路由器型号和固件版本

竞斗云2.0（R169AC）

### (3) 详细日志

```
2020-07-03T23:06:40.1696387Z FIT description: OpenWrt factory image
2020-07-03T23:06:40.1697041Z Created:         Fri Jul  3 16:13:28 2020
2020-07-03T23:06:40.1697282Z  Image 0 (askey_kernel)
2020-07-03T23:06:40.1697504Z   Description:  askey_kernel
2020-07-03T23:06:40.1697980Z   Created:      Fri Jul  3 16:13:28 2020
2020-07-03T23:06:40.1698183Z   Type:         Firmware
2020-07-03T23:06:40.1698313Z   Compression:  uncompressed
2020-07-03T23:06:40.1698480Z   Data Size:    3729472 Bytes = 3642.06 KiB = 3.56 MiB
2020-07-03T23:06:40.1699597Z   Architecture: ARM
2020-07-03T23:06:40.1699823Z   OS:           Unknown OS
2020-07-03T23:06:40.1700037Z   Load Address: unavailable
2020-07-03T23:06:40.1700248Z   Hash algo:    crc32
2020-07-03T23:06:40.1700458Z   Hash value:   9d3784ca
2020-07-03T23:06:40.1700650Z  Image 1 (askey_fs)
2020-07-03T23:06:40.1700858Z   Description:  askey_fs
2020-07-03T23:06:40.1701072Z   Created:      Fri Jul  3 16:13:28 2020
2020-07-03T23:06:40.1701289Z   Type:         Firmware
2020-07-03T23:06:40.1701495Z   Compression:  uncompressed
2020-07-03T23:06:40.1701710Z   Data Size:    35620636 Bytes = 34785.78 KiB = 33.97 MiB
2020-07-03T23:06:40.1701928Z   Architecture: ARM
2020-07-03T23:06:40.1702133Z   OS:           Unknown OS
2020-07-03T23:06:40.1702351Z   Load Address: unavailable
2020-07-03T23:06:40.1702553Z   Hash algo:    crc32
2020-07-03T23:06:40.1702744Z   Hash value:   5027f194
2020-07-03T23:06:40.1702942Z  Image 2 (ubifs)
2020-07-03T23:06:40.1703150Z   Description:  ubifs
2020-07-03T23:06:40.1703365Z   Created:      Fri Jul  3 16:13:28 2020
2020-07-03T23:06:40.1703577Z   Type:         Firmware
2020-07-03T23:06:40.1703789Z   Compression:  uncompressed
2020-07-03T23:06:40.1704009Z   Data Size:    37093376 Bytes = 36224.00 KiB = 35.38 MiB
2020-07-03T23:06:40.1704224Z   Architecture: ARM
2020-07-03T23:06:40.1704432Z   OS:           Unknown OS
2020-07-03T23:06:40.1704643Z   Load Address: unavailable
2020-07-03T23:06:40.1704851Z   Hash algo:    crc32
2020-07-03T23:06:40.1705051Z   Hash value:   1997f40f
2020-07-03T23:06:40.1807898Z cp /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-compex_wpj428-initramfs-fit-uImage.itb /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/bin/targets/ipq40xx/generic/openwrt-ipq40xx-generic-compex_wpj428-initramfs-fit-uImage.itb
2020-07-03T23:06:40.2234303Z [ -f /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/compex_wpj428-fit-uImage.itb -a -f /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/root.squashfs ]
2020-07-03T23:06:40.2251642Z dd if=/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/compex_wpj428-fit-uImage.itb >> /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-compex_wpj428-squashfs-sysupgrade.bin
2020-07-03T23:06:40.2448456Z 7277+1 records in
2020-07-03T23:06:40.2448805Z 7277+1 records out
2020-07-03T23:06:40.2449176Z 3726172 bytes (3.7 MB, 3.6 MiB) copied, 0.0172873 s, 216 MB/s
2020-07-03T23:06:40.2452891Z dd if=/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/root.squashfs >> /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-compex_wpj428-squashfs-sysupgrade.bin
2020-07-03T23:06:40.3184971Z [ -f /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/compex_wpj428-fit-uImage.itb -a -f /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/root.squashfs ]
2020-07-03T23:06:40.3207237Z dd if=/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/compex_wpj428-fit-uImage.itb >> /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-compex_wpj428-squashfs-cpximg-6a04.bin
2020-07-03T23:06:40.3427409Z 7277+1 records in
2020-07-03T23:06:40.3427734Z 7277+1 records out
2020-07-03T23:06:40.3428234Z 3726172 bytes (3.7 MB, 3.6 MiB) copied, 0.0195646 s, 190 MB/s
2020-07-03T23:06:40.3434234Z dd if=/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/root.squashfs >> /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-compex_wpj428-squashfs-cpximg-6a04.bin
2020-07-03T23:06:40.4529879Z 69571+1 records in
2020-07-03T23:06:40.4530135Z 69571+1 records out
2020-07-03T23:06:40.4530926Z 35620636 bytes (36 MB, 34 MiB) copied, 0.205422 s, 173 MB/s
2020-07-03T23:06:40.4535914Z /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/staging_dir/host/bin/padjffs2 /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-compex_wpj428-squashfs-sysupgrade.bin  64
[... 32 lines cut ...]
2020-07-03T23:06:42.5829483Z WARNING: Image file /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-dlink_dap-2610-squashfs-sysupgrade.bin is too big
2020-07-03T23:06:42.5928319Z echo '{  "metadata_version": "1.0", "supported_devices":["dlink,dap-2610"], "version": { "dist": "OpenWrt", "version": "SNAPSHOT", "revision": "r2746-90e48a34b", "target": "ipq40xx/generic", "board": "dlink_dap-2610" } }' | fwtool -I - /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-dlink_dap-2610-squashfs-sysupgrade.bin
2020-07-03T23:06:42.5955070Z Failed to open firmware file
2020-07-03T23:06:42.5959337Z Makefile:729: recipe for target '/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-dlink_dap-2610-squashfs-sysupgrade.bin' failed
2020-07-03T23:06:42.5961949Z [ ! -s "/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/key-build" -o ! -s "/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/key-build.ucert" -o ! -s "/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-dlink_dap-2610-squashfs-sysupgrade.bin" ] || { cp "/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/key-build.ucert" "/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-dlink_dap-2610-squashfs-sysupgrade.bin.ucert" ; usign -S -m "/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-dlink_dap-2610-squashfs-sysupgrade.bin" -s "/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/key-build" -x "/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-dlink_dap-2610-squashfs-sysupgrade.bin.sig" ; ucert -A -c "/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-dlink_dap-2610-squashfs-sysupgrade.bin.ucert" -x "/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-dlink_dap-2610-squashfs-sysupgrade.bin.sig" ; fwtool -S "/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-dlink_dap-2610-squashfs-sysupgrade.bin.ucert" "/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx- [... cut ...]
2020-07-03T23:06:42.6045083Z target/Makefile:19: recipe for target '/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install' failed
2020-07-03T23:06:42.6098460Z make[1]: Leaving directory '/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede'
2020-07-03T23:06:42.6102409Z Build failed - please re-run with -j1 to see the real error message
2020-07-03T23:06:42.7530051Z make: *** [world] Error 1
2020-07-03T23:06:42.7531343Z /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/include/toplevel.mk:222: recipe for target 'world' failed
2020-07-03T23:06:43.5558792Z ##[error]Process completed with exit code 2.
```

sxx1314, 2020-07-04:
7月2号左右更新了5.4 config和19.07.3不同  请重新在本地生成

Cathgao, 2020-07-04:
> 7月2号左右更新了5.4 config和19.07.3不同 请重新在本地生成

意思是需要重写config文件再push到云端吗？

sxx1314, 2020-07-04:
> > 7月2号左右更新了5.4 config和19.07.3不同 请重新在本地生成
> 
> 意思是需要重写config文件再push到云端吗？

重新 git pull 最新的源码 生产 config之后 git push 到自己的项目下 就可以编译成功了

Cathgao, 2020-07-04:
> > > 7月2号左右更新了5.4 config和19.07.3不同 请重新在本地生成
> > 
> > 
> > 意思是需要重写config文件再push到云端吗？
> 
> 重新 git pull 最新的源码 生成 config之后 git push 到自己的项目下 就可以编译成功了

好的谢谢，我有机会的时候会尝试

closed 2020-07-04 by coolsnowwolf

leopardciaw, 2020-08-14:
> > > > 7月2号左右更新了5.4 config和19.07.3不同 请重新在本地生成
> > > 
> > > 
> > > 意思是需要重写config文件再push到云端吗？
> > 
> > 
> > 重新 git pull 最新的源码 生成 config之后 git push 到自己的项目下 就可以编译成功了
> 
> 好的谢谢，我有机会的时候会尝试

> ## 1.关于你要提交的问题
> 竞斗云2.0从7月2号开始云编译不通过
>
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [x]  没有类似的issue
>
> ## 2. 详细叙述
> action配置没修改，但从7月2号开始编译一直失败。下载log后查看提示是image too [...]

> > 7月2号左右更新了5.4 config和19.07.3不同 请重新在本地生成
> 
> 意思是需要重写config文件再push到云端吗？

想问下题主，解决了没，我的也一直出现问题，但我是采用云编译的。出现以下错误：
[mkmylofw] Error: block at 0x00180000 is too big