coolsnowwolf/lede #4957: 新3上修改wireless导致无信号

catsimple, 2020-07-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：默认刚刚刷入时是可以搜到OpenWrt的ssid，修改无线设置以后，无法搜索到ssid

### (2) 路由器型号和固件版本

A：newifi3 d2 最新源码

### (3) 详细日志

A：Sat Jul  4 17:47:20 2020 daemon.err hostapd: 2 errors found in configuration file '/var/run/hostapd-phy1.conf'
Sat Jul  4 17:47:20 2020 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy1.conf
Sat Jul  4 17:47:20 2020 daemon.notice netifd: radio1 (7641): Command failed: Invalid argument
Sat Jul  4 17:47:20 2020 daemon.err odhcpd[2088]: Failed to send to ff02::1%lan@br-lan (Address not available)
Sat Jul  4 17:47:20 2020 daemon.notice netifd: radio1 (7641): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Sat Jul  4 17:47:20 2020 daemon.notice netifd: radio1 (7641): Command failed: Invalid argument
Sat Jul  4 17:47:20 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Sat Jul  4 17:47:20 2020 daemon.err hostapd: Line 6: Invalid country_code '00'
Sat Jul  4 17:47:20 2020 daemon.err hostapd: Cannot enable IEEE 802.11d without setting the country_code
Sat Jul  4 17:47:20 2020 daemon.err hostapd: 2 errors found in configuration file '/var/run/hostapd-phy0.conf'
Sat Jul  4 17:47:20 2020 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy0.conf
Sat Jul  4 17:47:20 2020 daemon.notice netifd: radio0 (7640): Command failed: Invalid argument
Sat Jul  4 17:47:20 2020 daemon.notice netifd: radio0 (7640): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Sat Jul  4 17:47:20 2020 daemon.notice netifd: radio0 (7640): Command failed: Invalid argument
Sat Jul  4 17:47:21 2020 daemon.info dnsmasq[8297]: Connected to system UBus
Sat Jul  4 17:47:23 2020 kern.info kernel: [  269.143260] mt7530 mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control off
Sat Jul  4 17:47:23 2020 kern.info kernel: [  269.150627] br-lan: port 2(lan2) entered blocking state
Sat Jul  4 17:47:23 2020 kern.info kernel: [  269.155863] br-lan: port 2(lan2) entered forwarding state
Sat Jul  4 17:47:23 2020 daemon.notice netifd: Network device 'lan2' link is up
Sat Jul  4 17:47:23 2020 kern.info kernel: [  269.162203] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Sat Jul  4 17:47:23 2020 daemon.notice netifd: bridge 'br-lan' link is up
Sat Jul  4 17:47:23 2020 daemon.notice netifd: Interface 'lan' has link connectivity
Sat Jul  4 17:47:24 2020 daemon.err odhcpd[2088]: Failed to send to ff02::1%lan@br-lan (Address not available)
Sat Jul  4 17:47:27 2020 kern.info kernel: [  273.367348] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
Sat Jul  4 17:47:27 2020 kern.info kernel: [  273.374856] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
Sat Jul  4 17:47:27 2020 daemon.notice netifd: Network device 'wan' link is up
Sat Jul  4 17:47:27 2020 daemon.notice netifd: Interface 'wan' has link connectivity
Sat Jul  4 17:47:27 2020 daemon.notice netifd: Interface 'wan' is setting up now
Sat Jul  4 17:47:27 2020 daemon.notice netifd: Interface 'wan6' has link connectivity
Sat Jul  4 17:47:27 2020 daemon.notice netifd: Interface 'wan6' is setting up now
Sat Jul  4 17:47:27 2020 daemon.notice netifd: wan (8491): udhcpc: started, v1.31.1
Sat Jul  4 17:47:27 2020 daemon.err odhcp6c[8494]: Failed to send RS (Address not available)
Sat Jul  4 17:47:27 2020 daemon.notice netifd: wan (8491): udhcpc: sending discover
Sat Jul  4 17:47:27 2020 daemon.notice netifd: wan (8491): udhcpc: sending select for 10.14.36.21
Sat Jul  4 17:47:27 2020 daemon.notice netifd: wan (8491): udhcpc: lease of 10.14.36.21 obtained, lease time 43200
Sat Jul  4 17:47:27 2020 daemon.err odhcp6c[8494]: Failed to send DHCPV6 message to ff02::1:2 (Address not available)
Sat Jul  4 17:47:27 2020 daemon.notice netifd: Interface 'wan' is now up
Sat Jul  4 17:47:29 2020 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Sat Jul  4 17:47:31 2020 daemon.info privoxy[6972]: Restart request due to 'ifup' of interface 'lan'
Sat Jul  4 17:47:32 2020 daemon.notice privoxy[6972]: Shutdown successfully
Sat Jul  4 17:47:34 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1) --> new PHY
Sat Jul  4 17:47:34 2020 daemon.err hostapd: Line 6: Invalid country_code '00'
Sat Jul  4 17:47:34 2020 daemon.err hostapd: Cannot enable IEEE 802.11d without setting the country_code
Sat Jul  4 17:47:34 2020 daemon.err hostapd: 2 errors found in configuration file '/var/run/hostapd-phy1.conf'
Sat Jul  4 17:47:34 2020 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy1.conf
Sat Jul  4 17:47:34 2020 daemon.notice netifd: radio1 (8753): Command failed: Invalid argument
Sat Jul  4 17:47:34 2020 daemon.notice netifd: radio1 (8753): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Sat Jul  4 17:47:34 2020 daemon.notice netifd: radio1 (8753): Command failed: Invalid argument
Sat Jul  4 17:47:34 2020 daemon.notice privoxy[9050]: Started successfully
Sat Jul  4 17:47:34 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Sat Jul  4 17:47:34 2020 daemon.err hostapd: Line 6: Invalid country_code '00'
[... 29 lines cut ...]
Sat Jul  4 17:52:00 2020 daemon.err hostapd: Line 6: Invalid country_code '00'
Sat Jul  4 17:52:00 2020 daemon.err hostapd: Cannot enable IEEE 802.11d without setting the country_code
Sat Jul  4 17:52:00 2020 daemon.err hostapd: 2 errors found in configuration file '/var/run/hostapd-phy0.conf'
Sat Jul  4 17:52:00 2020 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy0.conf
Sat Jul  4 17:52:00 2020 daemon.notice netifd: radio0 (11400): Command failed: Invalid argument
Sat Jul  4 17:52:00 2020 daemon.notice netifd: radio0 (11400): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Sat Jul  4 17:52:00 2020 daemon.notice netifd: radio0 (11400): Command failed: Invalid argument
Sat Jul  4 17:52:02 2020 kern.info kernel: [  547.766610] br-lan: port 6(wlan0) entered blocking state
Sat Jul  4 17:52:02 2020 kern.info kernel: [  547.772028] br-lan: port 6(wlan0) entered disabled state
Sat Jul  4 17:52:02 2020 kern.info kernel: [  547.778218] device wlan0 entered promiscuous mode
Sat Jul  4 17:52:16 2020 daemon.info dnsmasq[11835]: Connected to system UBus
Sat Jul  4 17:52:29 2020 daemon.notice netifd: radio0 (12112): WARNING: Variable 'data' does not exist or is not an array/object
Sat Jul  4 17:52:29 2020 kern.info kernel: [  575.526499] device wlan0 left promiscuous mode
Sat Jul  4 17:52:29 2020 kern.info kernel: [  575.531387] br-lan: port 6(wlan0) entered disabled state
Sat Jul  4 17:52:31 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Sat Jul  4 17:52:31 2020 daemon.err hostapd: Line 6: Invalid country_code '00'
Sat Jul  4 17:52:31 2020 daemon.err hostapd: Cannot enable IEEE 802.11d without setting the country_code
Sat Jul  4 17:52:31 2020 daemon.err hostapd: 2 errors found in configuration file '/var/run/hostapd-phy0.conf'
Sat Jul  4 17:52:31 2020 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy0.conf
Sat Jul  4 17:52:31 2020 daemon.notice netifd: radio0 (12133): Command failed: Invalid argument
Sat Jul  4 17:52:31 2020 daemon.notice netifd: radio0 (12133): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Sat Jul  4 17:52:31 2020 daemon.notice netifd: radio0 (12133): Command failed: Invalid argument
Sat Jul  4 17:52:32 2020 kern.info kernel: [  578.546368] br-lan: port 6(wlan0) entered blocking state
Sat Jul  4 17:52:32 2020 kern.info kernel: [  578.551797] br-lan: port 6(wlan0) entered disabled state
Sat Jul  4 17:52:32 2020 kern.info kernel: [  578.557769] device wlan0 entered promiscuous mode

acooler15, 2020-07-04:
Cannot enable IEEE 802.11d without setting the country_code，可能是wifi区域码没有设置

catsimple, 2020-07-04:
> Cannot enable IEEE 802.11d without setting the country_code，可能是wifi区域码没有设置

设置了CN或者其他国家也不行，好奇怪

acooler15, 2020-07-04:
Lean大可能修复了这个BUG， 你更新下源码，重新编译下试试。

catsimple, 2020-07-04:
> Lean大可能修复了这个BUG， 你更新下源码，重新编译下试试。

重新pull了一下，果然解决了
https://github.com/coolsnowwolf/lede/commit/104ed945a6935b16d7195d89121caa272bdc8097

closed 2020-07-04 by catsimple