coolsnowwolf/lede #4958: 无法停止ShadowSocksR Plus+的服务

ipeterlu, 2020-07-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ShadowSocksR Plus+的服务无法停止，按了复位之后也没用，

### (2) 路由器型号和固件版本

A：软路由，安装在群晖的虚拟机中。Powered by LuCI Master (git-20.088.29068-cebe120) / OpenWrt R20.4.8

### (3) 详细日志

A：2020-07-04 20:19:45 Shadowsocks/ShadowsocksR 2 Threads Started!
2020-07-04 20:18:38 Shadowsocks/ShadowsocksR 2 Threads Started!
2020-07-04 20:18:36 Shadowsocks/ShadowsocksR 2 Threads Started!
2020-07-04 20:18:21 Shadowsocks/ShadowsocksR 2 Threads Started!
2020-07-04 20:18:20 订阅更新成功
2020-07-04 20:18:20 新增节点数量: 78 删除节点数量: 0
2020-07-04 20:18:20 维持当前主服务器节点。
2020-07-04 20:18:20 成功解析节点数量: 181
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 IPLC 04
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 IPLC 03
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 IPLC 02
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 IPLC 01
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 IEPL 04
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 IEPL 03
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 IEPL 02
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 IEPL 01
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 CN2
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 BGP 10
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 BGP 09
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 BGP 08
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 BGP 07
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 BGP 06
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 BGP 05
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 BGP 04
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 BGP 03
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 BGP 02
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 BGP 01
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 BGP - Auto
2020-07-04 20:18:20 成功解析: ss 节点, 香港高级 9929
2020-07-04 20:18:20 成功解析: ss 节点, 香港标准 IPLC 11
2020-07-04 20:18:20 成功解析: ss 节点, 香港标准 IPLC 10
2020-07-04 20:18:20 成功解析: ss 节点, 香港标准 IPLC 09
2020-07-04 20:18:20 成功解析: ss 节点, 香港标准 IPLC 08
2020-07-04 20:18:20 成功解析: ss 节点, 香港标准 IPLC 07
2020-07-04 20:18:20 成功解析: ss 节点, 香港标准 IPLC 06
2020-07-04 20:18:20 成功解析: ss 节点, 香港标准 IPLC 05
2020-07-04 20:18:20 成功解析: ss 节点, 香港标准 IPLC 04
2020-07-04 20:18:20 成功解析: ss 节点, 香港标准 IPLC 03
2020-07-04 20:18:20 成功解析: ss 节点, 香港标准 IPLC 02
2020-07-04 20:18:20 成功解析: ss 节点, 香港标准 IPLC 01
2020-07-04 20:18:20 成功解析: ss 节点, 香港标准 BGP - Auto
2020-07-04 20:18:19 成功解析: ss 节点, 香港 WTT
2020-07-04 20:18:19 成功解析: ss 节点, 香港 PCCW 05
2020-07-04 20:18:19 成功解析: ss 节点, 香港 PCCW 04
2020-07-04 20:18:19 成功解析: ss 节点, 香港 PCCW 03
2020-07-04 20:18:19 成功解析: ss 节点, 香港 PCCW 02
2020-07-04 20:18:19 成功解析: ss 节点, 香港 PCCW 01
2020-07-04 20:18:19 成功解析: ss 节点, 香港 NTT 02
2020-07-04 20:18:19 成功解析: ss 节点, 香港 NTT 01
2020-07-04 20:18:19 成功解析: ss 节点, 香港 Azure 02
2020-07-04 20:18:19 成功解析: ss 节点, 香港 Azure 01
[... 117 lines cut ...]
2020-07-04 20:18:19 成功解析: ss 节点, 台湾高级 BGP 08
2020-07-04 20:18:19 成功解析: ss 节点, 台湾高级 BGP 07
2020-07-04 20:18:19 成功解析: ss 节点, 台湾高级 BGP 06
2020-07-04 20:18:19 成功解析: ss 节点, 台湾高级 BGP 05
2020-07-04 20:18:19 成功解析: ss 节点, 台湾高级 BGP 04
2020-07-04 20:18:19 成功解析: ss 节点, 台湾高级 BGP 03
2020-07-04 20:18:19 成功解析: ss 节点, 台湾高级 BGP 02
2020-07-04 20:18:19 成功解析: ss 节点, 台湾高级 BGP 01
2020-07-04 20:18:19 成功解析: ss 节点, 台湾高级 BGP - Auto
2020-07-04 20:18:19 成功解析: ss 节点, 台湾高级 9929
2020-07-04 20:18:19 成功解析: ss 节点, 台湾标准 BGP - Auto
2020-07-04 20:18:19 成功解析: ss 节点, 台湾 TFN
2020-07-04 20:18:19 成功解析: ss 节点, 台湾 CHT 02
2020-07-04 20:18:19 成功解析: ss 节点, 台湾 CHT 01
2020-07-04 20:18:19 成功解析: ss 节点, 印度标准 BGP
2020-07-04 20:18:19 成功解析: ss 节点, 印度 Azure 01
2020-07-04 20:18:19 成功解析: ss 节点, 印度 AWS
2020-07-04 20:18:19 成功解析: ss 节点, 俄罗斯标准 BGP
2020-07-04 20:18:19 成功解析: ss 节点, 俄罗斯 RT
2020-07-04 20:18:19 成功解析: ss 节点, 俄罗斯 GIA 02
2020-07-04 20:18:19 成功解析: ss 节点, 俄罗斯 GIA 01
sed: can't create temp file '/etc/crontabs/rootXXXXXX': No space left on device
sed: can't create temp file '/etc/crontabs/rootXXXXXX': No space left on device
2020-07-04 20:18:17 服务正在暂停
2020-07-04 20:16:38 Shadowsocks/ShadowsocksR 2 Threads Started!

ricksuzade-maker, 2020-07-04:
好老的版本，试试最新的commit行不行？

coolsnowwolf, 2020-07-05:
空间满了

closed 2020-07-05 by coolsnowwolf