coolsnowwolf/lede #4960: 截至目前位置最新版存在的问题：1、防火墙端口转发和流量规则配置界面显示“源于 uci: Entry not found”错误。2、SS状态界面下，【谷歌】连通性检查一直失败，但是翻墙正常。

ricksuzade-maker, 2020-07-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

从今天的使用情况看，今天到目前为止的最新版存在以下问题：1、防火墙端口转发和流量规则配置界面显示“源于 uci: Entry not found”错误。2、SS状态界面下，【谷歌】连通性检查一直失败，但是翻墙正常。

### (1) 具体问题

A：总体使用都正常，就是luci界面下有一些奇奇怪怪的显示问题。

### (2) 路由器型号和固件版本

A：x86软路由，intel i5-7200u

### (3) 详细日志

A：

Fddh2012, 2020-07-04:
1：可能是主题兼容性问题，我用的是更新版本的Argon主题没有在这两个页面发现此显示错误。
2：检察一下SSRP+的DNS设置，当配置为“使用本机5225端口“时 ”【谷歌】连通性检查“会失败，应该是无法进行【谷歌】的域名解析导致的。

ruderal04o, 2020-07-05:
连接失败倒不会，源于 uci: Entry not found，确实会

helianling, 2020-07-05:
连接检测正常，防火墙显示确实有问题：

[image]

closed 2020-07-05 by coolsnowwolf

kktt007, 2020-10-15:
设置下dns为腾讯的就没问题,可能默认的有问题,另外我之前设置发现阿里的223.6.6.6谷歌也连不上.阿里不给力