coolsnowwolf/lede #4966: 使用5.4内核最新代码后作为nfs服务器只能列出共享的文件夹名称，点进去不能显示内容

monsterzzzz, 2020-07-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [* ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用5.4内核最新代码后作为nfs服务器只能列出如/mnt/sda3的共享的文件夹名称，点进去不能显示内容，4.19内核时候正常
在Windows 10和安卓上现象相同

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：内核日志：
[   50.940451] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   50.941213] NFSD: Using legacy client tracking operations.
[   50.941715] NFSD: starting 10-second grace period (net f000008f)
系统日志：
Sun Jul  5 08:37:02 2020 daemon.notice rpc.mountd[5408]: authenticated mount request from 192.168.222.198:913 for /mnt/internal (/mnt/internal)
Sun Jul  5 08:39:45 2020 daemon.notice rpc.mountd[5408]: authenticated unmount request from 192.168.222.198:996 for /mnt/internal (/mnt/internal)
Sun Jul  5 08:40:58 2020 daemon.notice rpc.mountd[5408]: authenticated mount request from 192.168.222.198:905 for /mnt/internal (/mnt/internal)

monsterzzzz, 2020-07-05:
在路由器上执行mount现象一样，表现为没有任何回应，卡死，只能ctrl-c

Aes64X, 2020-07-05:
两位，检查你们代码 packages 部分，指向：net/nfs-kernel-server  
我在将 nfs-kernel-server 更新到版本 2.5.1 时发生了与两位类似的问题，表现为可以看到共享目录，点击共享目录后卡死。
Android KODI播放器是这样，我尝试WIN10 NFS管理挂载，仍然卡死。
退回这个更新后正常挂载。
他没有任何错误报告，系统日志也反映挂载成功。
（对了，我在退回更新后还增加了NFSv3支持，不知是否影响。）

monsterzzzz, 2020-07-05:
原来的feeds里面源nfs-kernel-server版本是2.3.4，现在转到官方源，nfs-kernel-server版本是2.5.1，是这个原因吗？编译时候确实选了nfsV4支持

Aes64X, 2020-07-05:
> 原来的feeds里面源nfs-kernel-server版本是2.3.4，现在转到官方源，nfs-kernel-server版本是2.5.1，是这个原因吗？编译时候确实选了nfsV4支持

退回吧，2.5.1 不知道什么问题，我测试了加上NFSv3支持还是会卡死。关键是没日志。。。。。

monsterzzzz, 2020-07-05:
日志显示服务器没有响应，其实也有服务器认证成功的日志，不知道为什么没响应

brianzc0306, 2020-07-05:
不知NFS这样，包括samba也是这样而且hd-idle无法使硬盘休眠

monsterzzzz, 2020-07-06:
ksmbd确实不太正常，windows 10可以正常访问，android就不行了（之前是可以的），没测试samba4怎么样

monsterzzzz, 2020-07-07:
经试验，nfs-kernel-server退回到2.3.4功能正常，ksmbd windows 10可以正常访问，某些android下的客户端也可以正常访问，某些就不行。换用samba4后全部正常，就是固件大了不少

shu0802, 2020-07-08:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [* ] 没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A：使用5.4内核最新代码后作为nfs服务器只能列出如/mnt/sda3的共享的文件夹名称，点进去不能显示内容，4.19内核时候正常
> 在Windows 10和安卓上现象相同
>
> ### (2) [...]

windows上关闭服务功能，你看一下是不是与下面的一样
[image]
[image]

monsterzzzz, 2020-07-09:
> windows上关闭服务功能，你看一下是不是与下面的一样
>
> [image]
> [image]

关于这个ksmbd，Windows 10一直是正常可访问的，知识某些安卓系统访问不正常，感觉兼容性不及samba4，但是后者大得多

closed 2020-07-09 by coolsnowwolf

lingyunzhiss, 2020-08-11:
> 经试验，nfs-kernel-server退回到2.3.4功能正常，ksmbd windows 10可以正常访问，某些android下的客户端也可以正常访问，某些就不行。换用samba4后全部正常，就是固件大了不少

我 也出现跟你一样的问题，5.4内核 ，LEDE V2.36.目前SAMBA挂载正常win10和android都能访问。就是NFS不行。能看到挂载目录，进去说没权限。请问怎样退回nfs-kernel-server2.3.4?