coolsnowwolf/lede #4967: staging_dir/target-x86_64_musl/stamp/.package_compile' failed

nebulite, 2020-07-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：20200704 git pull 的code编译失败
package/Makefile:107: recipe for target '/home/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed

### (2) 路由器型号和固件版本
A：
x86_64

### (3) 详细日志

A：

make[3]: Entering directory '/home/lede/feeds/packages/sound/forked-daapd'
touch /home/lede/build_dir/target-x86_64_musl/forked-daapd-27.1/.prepared_975db62ff570379715557375fca6170f_6664517399ebbbc92a37c5bb081b5c53_check
. /home/lede/include/shell.sh; xzcat /home/lede/dl/forked-daapd-27.1.tar.xz | tar -C /home/lede/build_dir/target-x86_64_musl/forked-daapd-27.1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/lede/build_dir/target-x86_64_musl/forked-daapd-27.1

Applying ./patches/010-errno.patch using plaintext: 
patching file src/websocket.c
Hunk #1 FAILED at 25.
1 out of 1 hunk FAILED -- saving rejects to file src/websocket.c.rej
Patch failed!  Please fix ./patches/010-errno.patch!
Makefile:79: recipe for target '/home/lede/build_dir/target-x86_64_musl/forked-daapd-27.1/.prepared_975db62ff570379715557375fca6170f_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/lede/build_dir/target-x86_64_musl/forked-daapd-27.1/.prepared_975db62ff570379715557375fca6170f_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/lede/feeds/packages/sound/forked-daapd'
time: package/feeds/packages/forked-daapd/compile#0.43#0.19#0.54
package/Makefile:111: recipe for target 'package/feeds/packages/forked-daapd/compile' failed
make[2]: *** [package/feeds/packages/forked-daapd/compile] Error 2
make[2]: Leaving directory '/home/lede'
package/Makefile:107: recipe for target '/home/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2

dlj390, 2020-07-05:
same issue

coolsnowwolf, 2020-07-05:
官方的package 补丁出现了错误

ccbcfan, 2020-07-05:
难怪从昨天到今天编译了不下五次都是错误，有办法解决吗？

loydcheung, 2020-07-05:
same issue here.

closed 2020-07-06 by coolsnowwolf

aaaol, 2022-08-15:
怎么办 我现在也是这种错误
Applying ./patches-pip/001-pep517-pyc-fix.patch using plaintext: 
patching file pip/_vendor/pep517/in_process/__init__.py
touch /home/moe/lede/staging_dir/hostpkg/lib/python3.9/site-packages/.pip_installed_22.0.4-1
mkdir -p /home/moe/lede/staging_dir/hostpkg/stamp
touch /home/moe/lede/build_dir/hostpkg/Python-3.9.12/.built
touch /home/moe/lede/staging_dir/hostpkg/stamp/.python3_installed
make[3]: Leaving directory '/home/moe/lede/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#179.05#19.45#1007.19
make[2]: Leaving directory '/home/moe/lede'
make[1]: *** [package/Makefile:110: /home/moe/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/moe/lede'
make: *** [/home/moe/lede/include/toplevel.mk:230：world] 错误 2

guo553, 2024-07-11:
same issue