coolsnowwolf/lede #4972: menuconfig选中了transmission，编译出来界面没有

tjlive, 2020-07-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
好几位网友都遇到相关的问题。

## 2. 详细叙述

### (1) 具体问题

A：最新源码，menuconfig选中了transmission，编译出来界面没有 

### (2) 路由器型号和固件版本

A：Newifi-D2

### (3) 详细日志

A：

nb12nb34, 2020-07-05:
同样问题。。x86-64 一样没有菜单。。有遇到同样问题的吗

EnnawYang, 2020-07-05:
[image]

tjlive, 2020-07-05:
TO EnnawYang
你这个应不是这两天获取的最新源码编译的吧。

shu0802, 2020-07-05:
是不是编译固件的时候固件大小不够大的原因

nb12nb34, 2020-07-05:
应该不是吧，已经设置了内核16m，分区300m,所以说应该不是固件大小的问题。。

coolsnowwolf, 2020-07-05:
需要手动选软件包了，官方 pacakge 就这样

tjlive, 2020-07-05:
coolsnowwolf 大，请问手工选软件包是什么意思？怎样操作？
从固件上看，它已编译到固件中了，但我怎样拿出来使用？

shu0802, 2020-07-05:
> 应该不是吧，已经设置了内核16m，分区300m,所以说应该不是固件大小的问题。。
是不是内核小了呢

tjlive, 2020-07-05:
原来需要在network --> bit ttorrent里面，选择daemon和Web control即可。
多谢L大和各位网友的指点。

closed 2020-07-05 by tjlive