coolsnowwolf/lede #4973: 编译报错

kk3375, 2020-07-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：NEWIFI D2

### (3) 详细日志
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[6]: Entering directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/libiberty/testsuite'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/libiberty/testsuite'
make[5]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/build-x86_64-pc-linux-gnu/libiberty'
make[5]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/libiberty'
make[2]: Leaving directory '/home/my/lede'
make[5]: Entering directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/libbacktrace'
make  all-am
make[6]: Entering directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/libbacktrace'
make[5]: Entering directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/libdecnumber'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/libdecnumber'
true  DO=all multi-do # make
make[6]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/libbacktrace'
make[5]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/libbacktrace'
make[5]: Entering directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/build-x86_64-pc-linux-gnu/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[5]: Entering directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/build-x86_64-pc-linux-gnu/fixincludes'
make[5]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/build-x86_64-pc-linux-gnu/libcpp'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/build-x86_64-pc-linux-gnu/fixincludes'
make[5]: Entering directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/fixincludes'
make[5]: Entering directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/fixincludes'
make[5]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/libcpp'
make[5]: Entering directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/lto-plugin'
make  all-am
make[6]: Entering directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/lto-plugin'
make[6]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/lto-plugin'
make[5]: Leaving directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/lto-plugin'
make[5]: Entering directory '/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/gcc'
g++ -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o cc1 c/c-lang.o c-family/stub-objc.o attribs.o c/c-errors.o c/c-decl.o c/c-typeck.o c/c-convert.o c/c-aux-info.o c/c-objc-common.o c/c-parser.o c/c-fold.o c/gimple-parser.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o c-family/c-format.o c-family/c-gimplify.o c-family/c-indentation.o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c-family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec.o c-family/c-ubsan.o c-family/known-headers.o c-family/c-attribs.o c-family/c-warn.o c-family/c-spellcheck.o glibc-c.o \
  cc1-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a   -L/home/my/lede/staging_dir/host/lib -L/home/my/lede/staging_dir/host/lib -L/home/my/lede/staging_dir/host/lib -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz
g++ -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o lto1 \
	lto/lto-lang.o lto/lto.o lto/lto-object.o attribs.o lto/lto-partition.o lto/lto-symtab.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a  -L/home/my/lede/staging_dir/host/lib -L/home/my/lede/staging_dir/host/lib -L/home/my/lede/staging_dir/host/lib -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a 
gimple-fold.o: In function `gimple_fold_stmt_to_constant_1(gimple*, tree_node* (*)(tree_node*), tree_node* (*)(tree_node*))':
/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/gimple-fold.c:6107: undefined reference to `gimple_simplify(gimple*, code_helper*, tree_node**, gimple**, tree_node* (*)(tree_node*), tree_node* (*)(tree_node*))'
/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/gimple-fold.c:6112: undefined reference to `mprts_hook'
gimple-fold.o: In function `gimple_build(gimple**, unsigned int, tree_code, tree_node*, tree_node*)':
/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/gimple-fold.c:7133: undefined reference to `gimple_simplify(tree_code, tree_node*, tree_node*, gimple**, tree_node* (*)(tree_node*))'
gimple-fold.o: In function `gimple_build(gimple**, unsigned int, tree_code, tree_node*, tree_node*, tree_node*)':
/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/gimple-fold.c:7159: undefined reference to `gimple_simplify(tree_code, tree_node*, tree_node*, tree_node*, gimple**, tree_node* (*)(tree_node*))'
gimple-fold.o: In function `gimple_build(gimple**, unsigned int, tree_code, tree_node*, tree_node*, tree_node*, tree_node*)':
/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/gimple-fold.c:7180: undefined reference to `gimple_simplify(tree_code, tree_node*, tree_node*, tree_node*, tree_node*, gimple**, tree_node* (*)(tree_node*))'
gimple-fold.o: In function `gimple_build(gimple**, unsigned int, built_in_function, tree_node*, tree_node*)':
/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/gimple-fold.c:7206: undefined reference to `gimple_simplify(built_in_function, tree_node*, tree_node*, gimple**, tree_node* (*)(tree_node*))'
gimple-fold.o: In function `gimple_build(gimple**, unsigned int, built_in_function, tree_node*, tree_node*, tree_node*)':
/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/gimple-fold.c:7232: undefined reference to `gimple_simplify(built_in_function, tree_node*, tree_node*, tree_node*, gimple**, tree_node* (*)(tree_node*))'
[... 57 lines cut ...]
/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/tree-ssa-sccvn.c:3995: undefined reference to `mprts_hook'
/home/my/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/tree-ssa-sccvn.c:3997: undefined reference to `mprts_hook'
tree-ssa-sccvn.o: In function `sccvn_dom_walker::before_dom_children(basic_block_def*)':
[... cut ...]
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory '/home/my/lede'
toolchain/Makefile:96: recipe for target '/home/my/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/my/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/my/lede'
Build failed - please re-run with -j1 to see the real error message
/home/my/lede/include/toplevel.mk:222: recipe for target 'world' failed

kk3375, 2020-07-05:
这个是什么报错啊，大佬给解答下，编译不了

closed 2020-07-05 by kk3375

rohanjuneja, 2022-02-15:
May I know if this issue is resolved? And what was the error here?