coolsnowwolf/lede #4977: 新三编译报错

xiaoguoyi, 2020-07-05:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最新源码，删除tmp 和.config后编译，仍然出错

### (2) 路由器型号和固件版本

A：NEWIFI3D2

### (3) 详细日志

A：Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for base-files:
 * 	libuci20130104
 * opkg_install_cmd: Cannot install package base-files.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-cryptodev:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-cryptodev.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-scsi-core:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-scsi-core.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-crypto-crc32c:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-crypto-crc32c.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-crypto-user:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-crypto-user.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-exfat:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-fs-exfat.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-ext4:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-fs-ext4.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-vfat:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-fs-vfat.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fuse:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-fuse.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-leds-gpio:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-leds-gpio.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-lib-crc-ccitt:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-lib-crc-ccitt.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-lib-textsearch:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-lib-textsearch.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-asn1-decoder:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-asn1-decoder.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-nf-reject:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-nf-reject.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-nf-reject6:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-nf-reject6.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-nf-ipt:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-nf-ipt.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-nf-ipt6:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-nf-ipt6.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-ipt-core:
[... 78 lines cut ...]
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-usb-storage-extras.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb3:
 * 	kernel (= 5.4.48-1-2bd334606daa38fa88fccbbf8c6e7c64)
 * opkg_install_cmd: Cannot install package kmod-usb3.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for block-mount:
 * 	libuci20130104
 * opkg_install_cmd: Cannot install package block-mount.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 * 	libuci20130104
 * opkg_install_cmd: Cannot install package default-settings.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ipv6helper:
 * 	libuci20130104
 * opkg_install_cmd: Cannot install package ipv6helper.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-vsftpd:
 * 	libuci20130104
 * opkg_install_cmd: Cannot install package luci-app-vsftpd.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/pt/lede'
package/Makefile:108: recipe for target '/home/pt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/pt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/pt/lede'
/home/pt/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

wangxin6, 2020-07-05:
x86_64 遇见同一个错误
`* satisfy_dependencies_for: Cannot satisfy the following dependencies for base-files:`
`* libuci20130104`
`* opkg_install_cmd: Cannot install package base-files.`
`* satisfy_dependencies_for: Cannot satisfy the following dependencies for block-mount:`
` * 	libuci20130104`
` * opkg_install_cmd: Cannot install package block-mount.`
` * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:`
` * 	libuci20130104`
` * opkg_install_cmd: Cannot install package default-settings.`

EnnawYang, 2020-07-05:
make clean

liwenjie119, 2020-07-05:
目前make clean后编译完成，但是连不上了，breed清除config也有问题

EnnawYang, 2020-07-05:
> 目前make clean后编译完成，但是连不上了，breed清除config也有问题

升级不要保留配置，如果进不去的话进breed重刷一遍固件

liwenjie119, 2020-07-05:
选好插件编译完成，breed刷入并清除config，正常开机，但是设置好wifi就挂了，重启启动不起来。现在我在全清默认编译，试试看是否正常

EnnawYang, 2020-07-05:
> 选好插件编译完成，breed刷入并清除config，正常开机，但是设置好wifi就挂了，重启启动不起来。现在我在全清默认编译，试试看是否正常

[image]
[image]
参考设置，LAN里面的物理接口一定要绑定上无线，不然会导致无线无法使用，目前就只能这样，以后应该还会优化的

liwenjie119, 2020-07-05:
> > 选好插件编译完成，breed刷入并清除config，正常开机，但是设置好wifi就挂了，重启启动不起来。现在我在全清默认编译，试试看是否正常
> 
> [image]
> [image]
> 参考设置，LAN里面的物理接口一定要绑定上无线，不然会导致无线无法使用，闭源驱动目前就只能这样，以后应该还会优化的
lan1234也需要手动绑定吗？我作为二级路由关了dhcp并改到192.168.1.2，重启照样连不上

EnnawYang, 2020-07-05:
> > > 选好插件编译完成，breed刷入并清除config，正常开机，但是设置好wifi就挂了，重启启动不起来。现在我在全清默认编译，试试看是否正常
> > 
> > 
> > [image]
> > [image]
> > 参考设置，LAN里面的物理接口一定要绑定上无线，不然会导致无线无法使用，闭源驱动目前就只能这样，以后应该还会优化的
> > lan1234也需要手动绑定吗？我作为二级路由关了dhcp并改到192.168.1.2，重启照样连不上

我这个是默认设置 具体怎样看你自己的

xiaoguoyi, 2020-07-05:
> > 选好插件编译完成，breed刷入并清除config，正常开机，但是设置好wifi就挂了，重启启动不起来。现在我在全清默认编译，试试看是否正常
> 
> [image]
> [image]
> 参考设置，LAN里面的物理接口一定要绑定上无线，不然会导致无线无法使用，闭源驱动目前就只能这样，以后应该还会优化的

从breed全新刷入可用，也没有绑定你说的物理接口

closed 2020-07-06 by coolsnowwolf