coolsnowwolf/lede #4982: ios下IPsec可以连上，但是无法访问网络

ldy8182, 2020-07-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
ios下IPsec可以连上，但是无法访问网络
### (1) 具体问题

A：

### (2) 路由器型号和固件版本
X64  OpenWrt R20.7.1 / LuCI Master (git-20.187.18390-dee77a6)
A：

### (3) 详细日志
[系统日志.txt](https://github.com/coolsnowwolf/lede/files/4874750/default.txt)
[内核日志.txt](https://github.com/coolsnowwolf/lede/files/4874751/default.txt)

A：

happymukey, 2020-07-06:
android同样情况来报道。

keivenliao, 2020-07-07:
[image]

coolsnowwolf, 2020-07-07:
Sun Jul  5 18:17:44 2020 daemon.info : 08[ENC] parsed QUICK_MODE request 1349671876 [ HASH ]
Sun Jul  5 18:17:44 2020 daemon.info : 08[KNL] received netlink error: No such file or directory (2)
Sun Jul  5 18:17:44 2020 daemon.info : 08[KNL] unable to add SAD entry with SPI ce6f6e87 (FAILED)
Sun Jul  5 18:17:44 2020 daemon.info : 08[KNL] received netlink error: No such file or directory (2)
Sun Jul  5 18:17:44 2020 daemon.info : 08[KNL] unable to add SAD entry with SPI 053968c3 (FAILED)
Sun Jul  5 18:17:44 2020 daemon.info : 08[IKE] unable to install inbound and outbound IPsec SA (SAD) in kernel

你们的 feeds 有问题，需要彻底删除feeds 后重新编译

poweryb9802, 2020-07-08:
缺少一个插件 自己安装strongswan-mod-kernel-libipsec

closed 2020-07-08 by coolsnowwolf