coolsnowwolf/lede #4991: 启用SSR+后，重启虚拟环境导致高CPU和高内存占用，不知为何之前开的case被L大直接关了

guytogay, 2020-07-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
使用Proxmox搭建的虚拟环境，当重启整个虚拟环境的时候（里面包含ROS，OpenWrt，等其他虚拟机）会出现OpenWrt这个虚拟机的高CPU和高内存占用，并持续20分钟左右；而只单单重启OpenWrt这个虚拟机，它就不会出现高CPU和高内存占用问题。更奇怪的是，这个问题只出现在启用OpenWrt里面的SSR+之后，如果不启用SSR+不论怎么重启虚拟环境都不会出现高CPU和高内存占用问题。
不知道为什么这个case没有任何答案就被直接关掉了。
不过我自己测试过，解决方案就是加下面一条在本地启动脚本里面，就可以解决。还是想借此请各位大神看下是不是这个shadowsocksr有什么需要修改的地方。
/etc/init.d/shadowsocksr restart

### (2) 路由器型号和固件版本

A：

主机名 | OpenWrt

主机型号 | Common KVM processor : 2 Core 8 Thread
CPU 信息 | 1995.380 MHz
固件版本 | OpenWrt R20.7.1 GDQ v7.1[2020] Compiled by eSir / LuCI Master (git-20.180.30743-de9fdad)
内核版本 | 4.19.123
本地时间 | Sat Jul 4 10:21:28 2020
运行时间 | 0h 22m 21s
平均负载 | 0.05, 5.53, 9.63
CPU 使用率（%） | 0 %
  

### (3) 详细日志

A：
[  364.323053] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000000
[  364.323717] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  364.324417] Mem-Info:
[  364.324646] active_anon:449014 inactive_anon:1943 isolated_anon:0
[  364.324646]  active_file:276 inactive_file:1144 isolated_file:64
[  364.324646]  unevictable:0 dirty:0 writeback:0 unstable:0
[  364.324646]  slab_reclaimable:7719 slab_unreclaimable:31283
[  364.324646]  mapped:2103 shmem:2276 pagetables:1397 bounce:0
[  364.324646]  free:6166 free_pcp:714 free_cma:0
[  364.327747] Node 0 active_anon:1796056kB inactive_anon:7772kB active_file:1104kB inactive_file:4576kB unevictable:0kB isolated(anon):0kB isolated(file):256kB mapped:8412kB dirty:0kB writeback:0kB shmem:9104kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[  364.329926] DMA free:8024kB min:124kB low:152kB high:180kB active_anon:7200kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:0kB writepending:0kB present:15992kB managed:15908kB mlocked:0kB kernel_stack:0kB pagetables:20kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[  364.332332] lowmem_reserve[]: 0 1976 1976 1976
[  364.332763] DMA32 free:16640kB min:16256kB low:20320kB high:24384kB active_anon:1788856kB inactive_anon:7772kB active_file:900kB inactive_file:4544kB unevictable:0kB writepending:0kB present:2080632kB managed:2028500kB mlocked:0kB kernel_stack:3456kB pagetables:5568kB bounce:0kB free_pcp:2620kB local_pcp:1368kB free_cma:0kB
[  364.335449] lowmem_reserve[]: 0 0 0 0
[  364.335800] DMA: 6*4kB (U) 12*8kB (UM) 4*16kB (UME) 3*32kB (U) 3*64kB (UME) 1*128kB (M) 3*256kB (UME) 3*512kB (UME) 1*1024kB (E) 2*2048kB (UE) 0*4096kB = 8024kB
[  364.337173] DMA32: 196*4kB (UME) 199*8kB (UME) 190*16kB (UME) 66*32kB (UME) 45*64kB (UME) 20*128kB (UME) 11*256kB (UME) 2*512kB (U) 0*1024kB 0*2048kB 0*4096kB = 16808kB
[  364.338739] 3963 total pagecache pages
[  364.339113] 0 pages in swap cache
[  364.339431] Swap cache stats: add 0, delete 0, find 0/0
[  364.339922] Free swap  = 0kB
[  364.340216] Total swap = 0kB
[  364.340700] 524156 pages RAM
[  364.341565] 0 pages HighMem/MovableOnly
[  364.342696] 13054 pages reserved
[  364.343013] Tasks state (memory values in pages):
[  364.345165] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
[  364.346251] [   1413]     0  1413      217       10    36864        0             0 ubusd
[  364.347028] [   1415]     0  1415      180        7    36864        0             0 askfirst
[  364.348911] [   1426]     0  1426      194        9    36864        0             0 urngd
[  364.350765] [   2476]     0  2476      240       38    40960        0             0 logd
[  364.351624] [   2503]     0  2503      263       27    40960        0             0 rpcd
[  364.352414] [   2526]     0  2526      315        9    36864        0             0 mdadm
[  364.353211] [   2752] 65533  2752      279       46    36864        0             0 rpcbind
[  364.353991] [   2820]     0  2820      324       61    40960        0             0 netifd
[... 1446 lines cut ...]
[  904.672582] [   6514] 65534  6514      473       10    45056        0             0 pdnsd
[  904.673418] [   6741]     0  6741      278       20    36864        0             0 sh
[  904.674200] [   7155]     0  7155      229       24    40960        0             0 miniupnpd
[  904.675014] [   7275]     0  7275     6998      565    94208        0             0 minidlna
[  904.675827] [   7334] 65537  7334      357       81    36864        0             0 rpc.statd
[  904.676694] [   7335]     0  7335      418       77    40960        0             0 rpc.mountd
[  904.677520] [   7954]   453  7954      749      496    40960        0             0 dnsmasq
[  904.678322] [   7960]     0  7960      730      491    40960        0             0 dnsmasq
[  904.679119] [   8484]     0  8484     1343       89    45056        0             0 ttyd
[  904.679890] [   8554]     0  8554      270       12    40960        0             0 crond
[  904.680653] [   8844]     0  8844      187       18    40960        0             0 wsdd2
[  904.681596] [  15489]     0 15489      331       72    40960        0             0 sh
[  904.682355] [  15506]     0 15506      269       10    36864        0             0 sh
[  904.683581] [  15510]     0 15510      287       29    40960        0             0 sh
[  904.685620] [  15511]     0 15511      207       13    36864        0             0 uci
[  904.687445] [  15512]     0 15512      278       20    36864        0             0 sh
[  904.688220] [  15514]     0 15514      269       11    32768        0             0 ps
[  904.689000] [  15515]     0 15515      271       14    36864        0             0 grep
[  904.689775] [  15516]     0 15516      269        6    32768        0             0 grep
[  904.690555] [  15517]     0 15517      269        7    32768        0             0 wc
[  904.691334] [  15521]     0 15521      289       29    40960        0             0 sh
[  904.692111] [  15522]     0 15522      304       72    36864        0             0 procd
[  904.693037] Out of memory: Kill process 6409 (trojan) score 155 or sacrifice child
[  904.693753] Killed process 6409 (trojan) total-vm:337476kB, anon-rss:316864kB, file-rss:0kB, shmem-rss:0kB
[  904.719594] oom_reaper: reaped process 6409 (trojan), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB

dolphinpaopao, 2020-07-06:
大概率又会被关闭，这种第三方修改编译的固件应该去找编译固件的作者咨询，因为这是开源项目，不清楚是否修改替换了哪些组件。修改版的固件有问题无法解决可以尝试原版编译的。问题复现才有利于问题定位，看日志像是修改版的Trojan有问题。

coolsnowwolf, 2020-07-06:
请联系固件编译作者

closed 2020-07-06 by coolsnowwolf