coolsnowwolf/lede #4993: 最新版本编译的固件不可用

hehhuig, 2020-07-05:
反馈bug/问题模板，提建议请删除

## 1.编译成功，但固件不兼容我原来的路由器，反复编译多次仍是如此。

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：固件不兼容

### (2) 路由器型号和固件版本

A：linksys 3200acm rango

### (3) 详细日志

A：

piaoyizy, 2020-07-06:
Configuring kmod-usb-net.
Configuring kmod-usb-net-asix-ax88179.
Configuring libuclient20160123.
Configuring uclient-fetch.
Configuring opkg.
Configuring kmod-pps.
Configuring kmod-ptp.
Configuring kmod-mlx4-core.
Configuring adbyby.
Configuring kmod-ipt-ipset.
Configuring libmnl0.
Configuring libipset13.
Configuring ipset.
Configuring dnsmasq-full.
Configuring luci-app-adbyby-plus.
Configuring libpcre.
Configuring kmod-crypto-authenc.
Configuring kmod-cryptodev.
Configuring libopenssl1.1.
Configuring wget.
Configuring kmod-bonding.
Configuring kmod-usb-storage-uas.
Configuring ubusd.
Configuring ubus.
Configuring iw.
Configuring wireless-regdb.
Configuring kmod-cfg80211.
Configuring hostapd-common.
Configuring kmod-mac80211.
Configuring kmod-ath.
Configuring kmod-ath10k.
Configuring ddns-scripts.
Configuring ca-certificates.
Configuring ddns-scripts_dnspod.
Configuring kmod-mdio.
Configuring kmod-hwmon-core.
Configuring kmod-ixgbe.
Configuring rpcd.
Configuring busybox.
Configuring kmod-lib-zlib-deflate.
Configuring coremark.
Configuring libustream-openssl20200215.
Configuring luci-lib-ip.
Configuring libubus-lua.
Configuring kmod-nf-conntrack6.
Configuring kmod-nf-reject6.
Configuring kmod-nf-ipt6.
Configuring kmod-ip6tables.
Configuring kmod-ipt-tproxy.
Configuring amdgpu-firmware.
Configuring kmod-nf-nat.
Configuring kmod-ipt-fullconenat.
Configuring kmod-tun.
Configuring libiwinfo-lua.
Configuring v2ray.
Configuring kmod-r8125.
Configuring kmod-lib-zlib-inflate.
Configuring kmod-crypto-acompress.
Configuring kmod-crypto-deflate.
Configuring kmod-crypto-cbc.
Configuring kmod-crypto-des.
Configuring kmod-crypto-echainiv.
Configuring kmod-crypto-hmac.
Configuring kmod-crypto-md5.
Configuring kmod-crypto-sha1.
Configuring kmod-ipsec.
Configuring kmod-iptunnel4.
Configuring kmod-ipsec4.
Configuring kmod-iptunnel6.
Configuring kmod-ipsec6.
[... 250 lines cut ...]
Configuring urngd.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for base-files:
 *      libuci20130104
 * opkg_install_cmd: Cannot install package base-files.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for libiwinfo20200105:
 *      libuci20130104
 * opkg_install_cmd: Cannot install package libiwinfo20200105.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for block-mount:
 *      libuci20130104
 * opkg_install_cmd: Cannot install package block-mount.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 *      libuci20130104
 * opkg_install_cmd: Cannot install package default-settings.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-vsftpd:
 *      libuci20130104
 * opkg_install_cmd: Cannot install package luci-app-vsftpd.
make[2]: *** [package/Makefile:68: package/install] Error 255
make[2]: Leaving directory '/root/lede'
make[1]: *** [package/Makefile:108: /root/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/root/lede'
make: *** [/root/lede/include/toplevel.mk:224: world] Error 2

OleStore, 2020-07-06:
make clean再来一次

closed 2020-07-06 by coolsnowwolf