coolsnowwolf/lede #4994: 截至目前最新版(commit bc65647)x86固件的问题

ricksuzade-maker, 2020-07-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

1、配置SSR+后，在SSR“状态”页面，点击“【谷歌】连通性检查”，会出现连接错误。查看SSR+日志，会发现不断有服务器失效的报错，然后自动切换或者重启SSR+后台进程的记录。

2、在开启“Turbo ACC”加速所有选项的情况下，防火墙端口转发出现故障，具体表现是内外网隐射端口不一致的情况下，访问内网服务速度极慢，并且容易发生连接中断。

3、“状态-->系统日志”页面下出现大量报错信息，具体是提示cannot find sensor, make sure load all kmod....，具体记不清了。

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：intel i5-7200u 软路由。

### (3) 详细日志

A：

ricksuzade-maker, 2020-07-06:
经测试上述第3个问题，在今天的版本已经解决

但是第1个问题变得更严重了，不仅谷歌连通性失败，连百度也失败了，而且下面更新规则也统统失败。。。

CHINAshanonline, 2020-07-07:
上述第二个问题，我遇到的情况是这样的
内网中有一台主机，通过端口转发至外网，内外端口号不一致
a.在外网使用IPv4网络通过外部端口访问内网主机，可以正常访问
b.在外网使用带有IPv6的网络（电脑连接联通手机热点）访问，**访问出现严重卡顿**
c.直接使用手机访问，可以正常访问

另外内外端口号一致的主机，无论用什么方式访问都正常

个人感觉问题出在IPv6上，但是在系统日志里看不到错误信息

目前暂时采取的措施是关闭所有ACC加速

coolsnowwolf, 2020-07-07:
基本是因为你们喜欢的 IPv6 导致的

closed 2020-07-08 by coolsnowwolf