coolsnowwolf/lede #4997: 更新到5.4内核后系统日志刷屏报错：No sensors found!

minfengyk, 2020-07-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ *] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：7月5日全新编译的固件，是装在虚拟机ESXI中运行，装了open-vm-tools插件，系统日志刷屏报错：No sensors found!，应该是没集成传感器驱动，Lean大能否集成驱动或系统未发现传感器时停止反复检查传感器。在就是请教一个问题：open-vm-tools和open-vm-tools-fuse有什么区别，编译时是否需要都选上？

### (2) 路由器型号和固件版本

A：

主机型号 | Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz : 1 Core 2 Thread

固件版本 | OpenWrt R20.7.1 / LuCI Master (git-20.187.38662-0cb730a)

内核版本 | 5.4.48

### (3) 详细日志

A：
Mon Jul  6 09:42:02 2020 daemon.err uhttpd[3042]: Make sure you loaded all the kernel drivers you need.
Mon Jul  6 09:42:02 2020 daemon.err uhttpd[3042]: Try sensors-detect to find out which these are.
Mon Jul  6 09:42:03 2020 daemon.err uhttpd[3042]: No sensors found!
Mon Jul  6 09:42:03 2020 daemon.err uhttpd[3042]: Make sure you loaded all the kernel drivers you need.
Mon Jul  6 09:42:03 2020 daemon.err uhttpd[3042]: Try sensors-detect to find out which these are.
Mon Jul  6 09:42:04 2020 daemon.err uhttpd[3042]: No sensors found!
Mon Jul  6 09:42:04 2020 daemon.err uhttpd[3042]: Make sure you loaded all the kernel drivers you need.
Mon Jul  6 09:42:04 2020 daemon.err uhttpd[3042]: Try sensors-detect to find out which these are.
Mon Jul  6 09:42:05 2020 daemon.err uhttpd[3042]: No sensors found!
Mon Jul  6 09:42:05 2020 daemon.err uhttpd[3042]: Make sure you loaded all the kernel drivers you need.
Mon Jul  6 09:42:05 2020 daemon.err uhttpd[3042]: Try sensors-detect to find out which these are.
Mon Jul  6 09:42:09 2020 daemon.err uhttpd[3042]: No sensors found!
Mon Jul  6 09:42:09 2020 daemon.err uhttpd[3042]: Make sure you loaded all the kernel drivers you need.
Mon Jul  6 09:42:09 2020 daemon.err uhttpd[3042]: Try sensors-detect to find out which these are.
Mon Jul  6 09:42:10 2020 daemon.err uhttpd[3042]: No sensors found!
Mon Jul  6 09:42:10 2020 daemon.err uhttpd[3042]: Make sure you loaded all the kernel drivers you need.
Mon Jul  6 09:42:10 2020 daemon.err uhttpd[3042]: Try sensors-detect to find out which these are.
Mon Jul  6 09:42:11 2020 daemon.err uhttpd[3042]: No sensors found!
Mon Jul  6 09:42:11 2020 daemon.err uhttpd[3042]: Make sure you loaded all the kernel drivers you need.
Mon Jul  6 09:42:11 2020 daemon.err uhttpd[3042]: Try sensors-detect to find out which these are.
Mon Jul  6 09:42:12 2020 daemon.err uhttpd[3042]: No sensors found!
Mon Jul  6 09:42:12 2020 daemon.err uhttpd[3042]: Make sure you loaded all the kernel drivers you need.
Mon Jul  6 09:42:12 2020 daemon.err uhttpd[3042]: Try sensors-detect to find out which these are.
Mon Jul  6 09:42:13 2020 daemon.err uhttpd[3042]: No sensors found!
Mon Jul  6 09:42:13 2020 daemon.err uhttpd[3042]: Make sure you loaded all the kernel drivers you need.
Mon Jul  6 09:42:13 2020 daemon.err uhttpd[3042]: Try sensors-detect to find out which these are.

ricksuzade-maker, 2020-07-06:
我也遇到这个问题了，刷屏故障，还有mwan3 连接8.8.4.4一堆错。。。

coolsnowwolf, 2020-07-06:
这些是官方的包的问题，请向 openwrt 官方反映

已经根据大家的意见切换到官方的包了

ricksuzade-maker, 2020-07-06:
经测试，今天的版本已经解决这个问题。

closed 2020-07-06 by coolsnowwolf