coolsnowwolf/lede #5002: MTK 无线闭源驱动的测试反馈

coolsnowwolf, 2020-07-06:
目前在源代码中已经集成了 MT7603， MT7612 ， MT7615 三种无线闭源驱动（来自 hanwckf 的老毛子）移植 5.4 内核

第一批初步默认集成了闭源驱动的机器型号有 新三 / K2P / 小米红米 AC2100/ 网件 R6850 这几款

大家测试中遇到什么问题，或者使用体验，麻烦统一在这里跟帖回复下

hcym, 2020-07-06:
又要开始编译mtk😂

Rsyan, 2020-07-06:
同样的线路，  潘多拉千兆op只有百兆

venpong, 2020-07-06:
期待加入极路由4

cattypapa, 2020-07-06:
ac2100的5g wifi简直就是...一位神秘人士...时有时无，大部分时间不能使用...速度和2.4g速度一样...300M宽带...下载速度每秒5-8M....

AmadeusGhost, 2020-07-06:
> 期待加入极路由4

自己取消掉开源驱动（kmod-mt7603/kmod-mt76x2/wpad/hostapd)，然后选上 kmod-mt7603e kmod-mt76x2e 和 luci-app-mtwifi，测试下 ok 我就改成默认用闭源了
其他人要用闭源驱动的同理，目前支持 mt7603+mt7612，mt7603+mt7615，mt7615dn， mt7615+mt7615

coolsnowwolf, 2020-07-06:
需要删除 .config 重新选择编译，你用老的配置还是开源的驱动

hyy-666, 2020-07-06:
双mt-7615要怎么设置闭源驱动？我是自己在搞MR2600的适配

bjzhou, 2020-07-06:
wr1200js也可以搞起来了

AmadeusGhost, 2020-07-06:
> 双mt-7615要怎么设置闭源驱动？我是自己在搞MR2600的适配

取消掉开源驱动，然后选择 kmod-mt7615d 和 luci-app-mtwifi

hyy-666, 2020-07-06:
> > 双mt-7615要怎么设置闭源驱动？我是自己在搞MR2600的适配
> 
> 取消掉开源驱动，然后选择 kmod-mt7615d 和 luci-app-mtwifi

谢谢大佬！！！

kkss2020, 2020-07-06:
有k2p成功的兄弟么

liwenjie119, 2020-07-06:
新三目前比较稳定，暂时没发现什么大问题，就是无线的概况不显示在状态-概况了，这应该是因为luci-app-mtwifi而改在网络-无线显示，不算bug

lxl6125, 2020-07-06:
k2p编译成功，但是。。。。不能无线桥接？

zmzhuai, 2020-07-06:
小米的R3P支持吗？

wangfeng35, 2020-07-06:
> ac2100的5g wifi简直就是...一位神秘人士...时有时无，大部分时间不能使用...速度和2.4g速度一样...300M宽带...下载速度每秒5-8M....

请问ac2100需要将哪几个设置替换成闭源驱动，谢谢~

AmadeusGhost, 2020-07-06:
> k2p编译成功，但是。。。。不能无线桥接？

mtwifi 支持桥接的，就是比较复杂，不知道有没有好一点的实现方法

AmadeusGhost, 2020-07-06:
> 小米的R3P支持吗？

支持的，r3p 好像是双 7615，取消掉开源驱动（kmod-7615e/mt7615-firmware/wpad/hostapd）然后选择 kmod-7615d 和 luci-app-mtwifi 就可以试了

AmadeusGhost, 2020-07-06:
> 请问ac2100需要将哪几个设置替换成闭源驱动，谢谢~

ac2100 默认闭源驱动

lxl6125, 2020-07-06:
> > k2p编译成功，但是。。。。不能无线桥接？
> 
> mtwifi 支持桥接的，就是比较复杂，不知道有没有好一点的实现方法

怎么弄？

AmadeusGhost, 2020-07-06:
> 怎么弄？

具体参考这位大佬的帖子 https://www.right.com.cn/forum/thread-4040044-1-1.html
[image]

zmzhuai, 2020-07-07:
> > 小米的R3P支持吗？
> 
> 支持的，r3p 好像是双 7615，取消掉开源驱动（kmod-7615e/mt7615-firmware/wpad/hostapd）然后选择 kmod-7615d 和 luci-app-mtwifi 就可以试了

编译了r3p的固件试了试。5G在mac下显示TX速率只能到866Mbps。还有编译的时候需要配置一下kmod-7615d，选择第二块7615网卡。不然编译出来只有2Gwifi可用

MeIsReallyBa, 2020-07-07:
> > > 小米的R3P支持吗？
> > 
> > 
> > 支持的，r3p 好像是双 7615，取消掉开源驱动（kmod-7615e/mt7615-firmware/wpad/hostapd）然后选择 kmod-7615d 和 luci-app-mtwifi 就可以试了
> 
> 编译了r3p的固件试了试。5G在mac下显示TX速率只能到866Mbps。还有编译的时候需要配置一下kmod-7615d，选择第二块7615网卡。不然编译出来只有2Gwifi可用

dat里的HT_RxStream和HT_TxStream改成4

liwenjie119, 2020-07-07:
新三iperf3测速结果：
5G wifi：上传到路由器157 Mbits/sec，从路由器下载188 Mbits/sec。（866Mbps接入）
2.4G wifi：上传到路由器25 Mbits/sec，从路由器下载113 Mbits/sec。（300Mbps接入）
不知道为啥2.4g上传这么慢，连续几次测速都是

AmadeusGhost, 2020-07-07:
> 新三iperf3测速结果：
> 5G wifi：上传到路由器157 Mbits/sec，从路由器下载188 Mbits/sec。（866Mbps接入）
> 2.4G wifi：上传到路由器25 Mbits/sec，从路由器下载113 Mbits/sec。（300Mbps接入）
> 不知道为啥2.4g上传这么慢，连续几次测速都是

别在路由器上跑 iperf，跑 wan or lan to 2.4G/5G

coolsnowwolf, 2020-07-07:
@AmadeusGhost 看来双 7615 的配置没加入

hyy-666, 2020-07-07:
我发现我昨天尝试的MR2600（双MT7615），我搞不动，试了一下适配，LED灯都有问题，现在的版本网口灯是不是直接由交换机芯片负责了，我没配置led 的情况下网口是没问题的，还有应该是我的问题，适配MT7615的闭源驱动，2.4G和5G不能同时开启。头大

hyy-666, 2020-07-07:
求大佬指一条明路

AmadeusGhost, 2020-07-07:
> @AmadeusGhost 看来双 7615 的配置没加入

晚一点在弄吧，我现在在外面😂

AmadeusGhost, 2020-07-07:
> 我发现我昨天尝试的MR2600（双MT7615），我搞不动，试了一下适配，LED灯都有问题，现在的版本网口灯是不是直接由交换机芯片负责了，我没配置led 的情况下网口灯是没问题的，还有应该是我的问题，适配MT7615的闭源驱动，2.4G和5G不能同时开启。头大

无线 led 的触发器应该是变成 ra0/rax0 了

hyy-666, 2020-07-07:
> > 我发现我昨天尝试的MR2600（双MT7615），我搞不动，试了一下适配，LED灯都有问题，现在的版本网口灯是不是直接由交换机芯片负责了，我没配置led 的情况下网口灯是没问题的，还有应该是我的问题，适配MT7615的闭源驱动，2.4G和5G不能同时开启。头大
> 
> 无线 led 的触发器应该是变成 ra0/rax0 了

大佬我贴一下我的配置文件，您有空的时候看一下吧。
`/dts-v1/;

#include "mt7621.dtsi"

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>

/ {
	compatible = "motorola,motorola-mr2600", "mediatek,mt7621-soc";
	model = "Motorola MR2600";

	aliases {
		led-boot = &led_power;
		led-failsafe = &led_power;
		label-mac-device = &gmac0;
	};

	chosen {
		bootargs = "console=ttyS0,115200";
	};

	leds {
		compatible = "gpio-leds";

		internet {
			label = "mr2600:green:internet";
			gpios = <&gpio 6 GPIO_ACTIVE_LOW>;
		};

		led_power: power {
			label = "mr2600:green:power";
			gpios = <&gpio 16 GPIO_ACTIVE_LOW>;
		};

		usb {
			label = "mr2600:blue:usb";
			gpios = <&gpio 14 GPIO_ACTIVE_LOW>;
			trigger-sources = <&xhci_ehci_port1>, <&ehci_port2>;
			linux,default-trigger = "usbport";
		};
	};

	keys {
		compatible = "gpio-keys";

		reset {
			label = "reset";
			gpios = <&gpio 15 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_RESTART>;
		};

		wps {
			label = "wps";
			gpios = <&gpio 18 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_WPS_BUTTON>;
		};

		wifi {
			label = "wifi";
			gpios = <&gpio 7 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_RFKILL>;
		};
	};
};

&spi0 {
	status = "okay";

	flash@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <10000000>;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			partition@0 {
				label = "u-boot";
				reg = <0x0 0x30000>;
				read-only;
			};

			partition@30000 {
				label = "u-boot-env";
				reg = <0x30000 0x10000>;
				read-only;
			};

			factory: partition@40000 {
				label = "factory";
				reg = <0x40000 0x10000>;
				read-only;
			};

			partition@50000 {
				compatible = "denx,uimage";
				label = "firmware";
				reg = <0x50000 0xfb0000>;
			};
		};
	};
};

&pcie {
	status = "okay";
};

&pcie0 {
	mt76@0,0 {
		compatible = "mediatek,mt76";
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x0000>;
	};
};

&pcie1 {
	mt76@0,0 {
		compatible = "mediatek,mt76";
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x8000>;
	};
};

&gmac0 {
	mtd-mac-address = <&factory 0xe000>;
};

&switch0 {
	ports {
		port@0 {
			status = "okay";
			label = "wan";
			mtd-mac-address = <&factory 0xe006>;
		};

		port@1 {
			status = "okay";
			label = "lan1";
		};

		port@2 {
			status = "okay";
			label = "lan2";
		};

		port@3 {
			status = "okay";
			label = "lan3";
		};

		port@4 {
			status = "okay";
			label = "lan4";
		};
	};
};

&state_default {
	gpio {
		groups = "i2c" , "wdt";
		function = "gpio";
	};
};`

nivalxer, 2020-07-07:
昨天使用Linksys EA7500v2（双MT7615），取消开源驱动，勾选mt7615d编译后。只有2.4G可用，5G可以设置，但无法启动。
因为放办公室用，没有及时查看日志后，就刷回开源驱动了。

coolsnowwolf, 2020-07-07:
双 7615 还需要等一等

AmadeusGhost, 2020-07-07:
> 我发现我昨天尝试的MR2600（双MT7615），我搞不动，试了一下适配，LED灯都有问题，现在的版本网口灯是不是直接由交换机芯片负责了，我没配置led 的情况下网口灯是没问题的，还有应该是我的问题，适配MT7615的闭源驱动，2.4G和5G不能同时开启。头大

话说是哪个 led 出问题来着... 如果是 wan/lan led 那我就不清楚了
无线 led 的话要把触发器改成 ra0/rai0，像新3这样：
```
ucidef_set_led_netdev "wlan2g" "WiFi 2.4GHz" "$boardname:blue:wlan2g" "ra0"
ucidef_set_led_netdev "wlan5g" "WiFi 5GHz" "$boardname:blue:wlan5g" "rai0"
```

hyy-666, 2020-07-07:
> ucidef_set_led_netdev "wlan2g" "WiFi 2.4GHz" "$boardname:blue:wlan2g" "ra0"
> ucidef_set_led_netdev "wlan5g" "WiFi 5GHz" "$boardname:blue:wlan5g" "rai0"

收到！

hyy-666, 2020-07-07:
> 话说是哪个 led 出问题来着... 如果是 wan/lan led 那我就不清楚了
> 无线 led 的话要把触发器改成 ra0/rai0，像新3这样：
>
> ```
> ucidef_set_led_netdev "wlan2g" "WiFi 2.4GHz" "$boardname:blue:wlan2g" "ra0"
> ucidef_set_led_netdev "wlan5g" "WiFi [...]

我放弃了，两天刷了8版固件。放弃了，目前知道的问题是两颗7615应该只驱动了一颗，表现是只可开单频，再者是LED的问题，目前没做调整交换机的没问题，另外由系统和7615管的两颗5颗LED，其中由7615管的俩貌似开不起来。
不投毒了，谢谢大佬的提示和lean对这一套源码的维护，贵安。

hyy-666, 2020-07-07:
看了这个兄弟的878的适配PR，我又行了
https://github.com/Lucky1openwrt/openwrt

AmadeusGhost, 2020-07-07:
> 我放弃了，两天刷了8版固件。放弃了，目前知道的问题是两颗7615应该只驱动了一颗，表现是只可开单频，再者是LED的问题，目前没做调整交换机的没问题，另外由系统和7615管的两颗5颗LED，其中由7615管的俩貌似开不起来。

[image]
make menuconfig 应该是这样显示的
> 要引用他的配置文件，要打声招呼吧？

不用，coomit 的话引用下原作者的 commit message 就行

hyy-666, 2020-07-07:
> > 我放弃了，两天刷了8版固件。放弃了，目前知道的问题是两颗7615应该只驱动了一颗，表现是只可开单频，再者是LED的问题，目前没做调整交换机的没问题，另外由系统和7615管的两颗5颗LED，其中由7615管的俩貌似开不起来。
> 
> [image]
> make menuconfig 应该是这样显示的
> 
> > 要引用他的配置文件，要打声招呼吧？
> 
> 不用，coomit 的话引用下原作者的 commit message 就行

[image]
我成了，现在做微调，看那位大佬的dts文件，mr2600的灯好像归mt7615管，我试试看能不能把他打开

hyy-666, 2020-07-07:
[image]
ojbk了，但是wifi的灯还是没法点亮，需要手动去network的配置。

hyy-666, 2020-07-07:
测试了一下闭源驱动在OP的表现，在大流量是会跳ping，或者没有响应，毛子可以跑满差不多跑满866的连接速度，OP只有500左右，跑满家宽300MCPU占用在10以内。
[image]
[image]
[image]

kk3375, 2020-07-08:
> > 怎么弄？
> 
> 具体参考这位大佬的帖子https://www.right.com.cn/forum/thread-4040044-1-1.html
> [image]

此链接新3已经实现中继扫描界面了，源码什么时候有啊

AmadeusGhost, 2020-07-08:
> 我成了，现在做微调，看那位大佬的dts文件，mr2600的灯好像归mt7615管，我试试看能不能把他打开

所以是哪个选项有问题来着

hyy-666, 2020-07-08:
> > 我成了，现在做微调，看那位大佬的dts文件，mr2600的灯好像归mt7615管，我试试看能不能把他打开
> 
> 所以是哪个选项有问题来着

选项没问题，led的驱动目前还在爬，现在有两种说法，一种是在mt7615，一种是7621的gpio，我再调整看看，对了貌似得手动加到lan里面，而且在luci界面不显示ap，每次改network的设置都会导致无线被踢出lan，还有一个是2.4G和5G改任意一个的配置，俩都会重启(算是bug吧)。

hyy-666, 2020-07-08:
> > 我成了，现在做微调，看那位大佬的dts文件，mr2600的灯好像归mt7615管，我试试看能不能把他打开
> 
> 所以是哪个选项有问题来着

找到了，确实在mt7615里，不过在config.in里被注释了。
[image]

hyy-666, 2020-07-08:
> > 我成了，现在做微调，看那位大佬的dts文件，mr2600的灯好像归mt7615管，我试试看能不能把他打开
> 
> 所以是哪个选项有问题来着

我发[PR](https://github.com/coolsnowwolf/lede/pull/5035)了大佬，在config.in里取消注释，就有2.4G和5G的灯了，现在基本上好了

coolsnowwolf, 2020-07-08:
目前还没做多核 NAT 和无线的内核优化，先测试 WIFI to LAN 的性能为准

lphgor, 2020-07-09:
在 K2P 上测试，修改 `接口 - LAN` 的设置以后，重启会自动禁用 5G 无线。
查看配置文件`/etc/config/network`，发现默认的`ifname`是`'lan1 lan2 lan3 lan4 ra0 rax0'`，但是在网页上保存配置以后，`ra0`丢失了，变成这样：
```
	option _orig_ifname 'lan1 lan2 lan3 lan4 ra0 rax0'
	option ifname 'lan1 lan2 lan3 lan4 rax0'
```
物理设置的接口列表里面也看不到`ra0`。
临时的解决方法是：每次修改 `接口 - LAN` 的设置，都需要手动添加`自定义接口: ra0`才行。

AmadeusGhost, 2020-07-09:
> 在 K2P 上测试，修改 `接口 - LAN` 的设置以后，重启会自动禁用 5G 无线。

在 4.14 上（或者说之前的固件）会这样嘛？

lphgor, 2020-07-09:
> 在 4.14 上（或者说之前的固件）会这样嘛？

好吧，原来一直都有这个问题，不是新内核的锅。

i5g5vip, 2020-07-10:
https://www.right.com.cn/forum/thread-4040044-1-1.html
贴子说明中是有中继界面的，
可是我编译出来的没有中继界面，
请问哪里设置不对？

shu0802, 2020-07-12:
> 新三目前比较稳定，暂时没发现什么大问题，就是无线的概况不显示在状态-概况了，这应该是因为luci-app-mtwifi而改在网络-无线显示，不算bug

我的新三2G网络不能开启，编译时如何选择的WIFI插件

chenxudong2020, 2020-07-12:
> 目前在源代码中已经集成了 MT7603， MT7612 ， MT7615 三种无线闭源驱动（来自 hanwckf 的老毛子 和 MeIsReallyB 的 github https://github.com/MeIsReallyBa/Redmi2100-WIFI-ProprietaryDriver-linux5.4 ）移植 5.4 内核并做了自动适应
> 
> 第一批初步默认集成了闭源驱动的机器型号有 新三 / K2P / 小米红米 AC2100/ 网件 R6850 这几款
> 
> Update：中继的支持已经加入了，详细使用方法可以看这里
> https://www.right.com.cn/forum/thread-4040044-1-1.html
> 
> 大家测试中遇到什么问题，或者使用体验，麻烦统一在这里跟帖回复下

老版本内核版本用的是开源还是闭源驱动？ 闭源驱动中继是个大问题 ；实测padavan最新7603鸡血驱动4.1.2.0版本在newifi3上表现好，但是在极路由4 HC5962上中继不行 完全没网速 所以不知道你移植的是不是这个驱动 如果openwrt移植过来 其他机器可能是灾难！！！！

takenusername, 2020-07-12:
红米 AC2100 默认设置，5G信号会频繁掉。2G根本不能用。

[image]

[image]

coolsnowwolf, 2020-07-13:
iperf3 如果测试 192.168.1.1 （路由器 IP） ，实际上测试的是路由器 7621 的 CPU 性能而不是带宽

lexin8, 2020-07-13:
ac2100重启后wifi是被禁用状态，这个没有修复吗？有没有命令设置重启路由后自动重启wifi

AmadeusGhost, 2020-07-13:
> ac2100重启后wifi是被禁用状态，这个没有修复吗？有没有命令设置重启路由后自动重启wifi

将 ra0 和 rai0 添加进 lan 口

liwenjie119, 2020-07-13:
> iperf3 如果测试 192.168.1.1 （路由器 IP） ，实际上测试的是路由器 7621 的 CPU 性能而不是带宽

如何正确测试带宽？

lexin8, 2020-07-14:
> > ac2100重启后wifi是被禁用状态，这个没有修复吗？有没有命令设置重启路由后自动重启wifi
> 
> 将 ra0 和 rai0 添加进 lan 口

问题得到解决，但是重启后有可能丢失 ra0 和 rai0配置的几率

lexin8, 2020-07-14:
发现一个bug，开启双WAN口，其中一个口为拨号，则无法拨号成功。两个都要静态或者DHCP模式获取IP才可以同时双wan口。

jw10126121, 2020-07-16:
R3G默认设置了闭源wifi吗？

[125 more replies]

closed 2021-02-04 by coolsnowwolf