coolsnowwolf/lede #5006: 4.19内核下 K3的wifi不正常

barton0426, 2020-07-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：wifi信道无法正确设置，设置36显示40，设置149也出问题

### (2) 路由器型号和固件版本

A：斐讯K3   R20.7.1

### (3) 详细日志

A：

信道: 1 (2.412 GHz) | 传输速率: ? Mbit/s
SSID: UNSAFE | 模式: Master
BSSID: 2C:15:E1:26:88:B7 | 加密: WPA2 PSK (CCMP)

信道: 40 (5.200 GHz) | 传输速率: ? Mbit/s
SSID: UNSAFE_5G | 模式: Client
BSSID: 2C:15:E1:26:88:B8 | 加密: -

barton0426, 2020-07-06:
个人手动更换了https://github.com/Hill-98/phicommk3-firmware中69027固件，有待测试

barton0426, 2020-07-06:
经测试 上述固件不可用 怀疑和4.19内核不兼容

hcym, 2020-07-06:
k3就不要折腾了，老物件，armv8的确实有提高，x86还没刷，估计蜗牛之类的也没必要

barton0426, 2020-07-06:
替换ac88u的firmware后解决
https://github.com/Hill-98/phicommk3-firmware

closed 2020-07-06 by barton0426

zjh1982, 2020-07-08:
我想问下更到最新源码屏幕怎么加进luci哪里，以前直接复制进lean就可以了，现在复制到那个文件夹都不行