coolsnowwolf/lede #5007: 端口转发异常问题

lexin8, 2020-07-06:
环境：X86固件，有公网IP，路由局域网地址192.168.1.1。 192.168.2.15是个网站，未做端口转发内网正常访问一切正常。

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option proto 'tcp udp'
	option dest_port '8888'
	option name 'BT'
	option src_dport '8001'
	option dest_ip '192.168.2.15'

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option proto 'tcp udp'
	option dest_port '80'
	option name 'iptv'
	option src_dport '8002'
	option dest_ip '192.168.2.15'

===============================
设置重启了后外网可以正常访问了。
出现问题http://公网IP:8002/无法在内网打开，只能在外网打开。
http://192.168.2.15:8002/一开始正常可以打开，过了几天小时出现访问就出现不能再内网访问，公网没有问题。

重置后第二天测试也是出现这样的情况，删掉该端口转发就内网就恢复正常了。是哪里设置有问题？

ruderal04o, 2020-07-06:
同样的情况。还有CPU信息显示不正常。暂时没有问题。CPU信息：

acpitz-acpi-0 Adapter: ACPI interface temp1: +27.8°C (crit = +119.0°C) temp2:
--

coolsnowwolf, 2020-07-07:
你可能继承了老版本配置导致的

closed 2020-07-08 by coolsnowwolf