coolsnowwolf/lede #5010: aliyun ddns不接受通配符*

jamesbean08, 2020-07-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

全新配置，使用*.mydomain.xyz作为解析域名时报错无法启动ddns，日志显示存在characters outside allowed subset；将通配符*替换为任意二级域名都可正常解析，故可判断其他设置项无问题。

使用通配符在4.19内核固件下正常。

官方git搜索了一下发现类似的问题，不过是出在cf-v4的脚本上面。
https://github.com/openwrt/packages/issues/11042

### (1) 具体问题

A：Aliddns填入通配符无法运行

### (2) 路由器型号和固件版本

A：Ubuntu Qemu X64/OpenWrt R20.7.1 / LuCI Master (git-20.187.38662-0cb730a)

### (3) 详细日志

A： 
211230  note : PID '3509' started at 2020-07-06 21:12
211231       : ddns version  : 2.7.8-21
211231       : uci configuration:
ddns.myddns_ipv4.domain='*.mydomain.xyz'
ddns.myddns_ipv4.enabled='1'
ddns.myddns_ipv4.interface='wan'
ddns.myddns_ipv4.ip_network='wan'
ddns.myddns_ipv4.ip_source='network'
ddns.myddns_ipv4.lookup_host='www.mydomain.xyz'
ddns.myddns_ipv4.password='********'
ddns.myddns_ipv4.service_name='aliyun.com'
ddns.myddns_ipv4.username='********'
ddns.myddns_ipv4=service
 211231       : verbose mode  : 0 - run normal, NO console output
 211231  CRIT : sanitize on domain found characters outside allowed subset - TERMINATE
 211231  WARN : PID '3509' exit WITH ERROR '1' at 2020-07-06 21:12

coolsnowwolf, 2020-07-07:
这个是官方脚本的功能，* 作为关键字被过滤掉了。请向官方op package 反映

jamesbean08, 2020-07-07:
> 这个是官方脚本的功能，* 作为关键字被过滤掉了。请向官方op package 反映

感谢答疑。自己没有办法解决目前版本的各种小问题，还是暂时退回6.18，至少确实稳定没有问题。

closed 2020-07-07 by jamesbean08

mm2231031, 2020-11-11:
修改
/usr/lib/ddns/dynamic_dns_functions.sh 
变量
DNS_CHARSET="[@a-zA-Z0-9._-]" 
为 
DNS_CHARSET="[@a-zA-Z0-9._-*]"
可以临时解决问题