coolsnowwolf/lede #5012: X86开机不能自动启动

shu0802, 2020-07-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A:X86开机不能自动启动。开机界面一直停留在下面的图片位置。

### (2) 路由器型号和固件版本

A：

主机型号 | Intel(R) Celeron(R) CPU 3865U @ 1.80GHz : 2 Core 2 Thread
      
固件版本 | OpenWrt R20.7.1 / LuCI Master (git-20.187.38662-0cb730a)

内核版本 | 5.4.50

### (3) 详细日志

A：
[image]

求大神指点，是哪里要怎么设置还是其它原因

lexin8, 2020-07-06:
这个一般跟固件没有关系，硬件启动引导的问题

OleStore, 2020-07-06:
我也卡着，动不了

shu0802, 2020-07-07:
> 这个一般跟固件没有关系，硬件启动引导的问题

没有修改硬件启动，换了原来的固件启动正常

主机型号 | Intel(R) Celeron(R) CPU 3865U @ 1.80GHz : 2 Core 2 Thread
-- | --
CPU信息 | 732.019 MHz +41.0°C
固件版本 | OpenWrt R20.5.20 / LuCI Master (git-20.117.60969-420c61a)
内核版本 | 4.9.223
同样用的openwrt-x86-64-generic-squashfs-combined.img这种格式写入的硬盘

hcym, 2020-07-07:
这个在54以前的最后版本就不行了，干脆就没管，一直用的armv8

closed 2020-07-07 by coolsnowwolf