coolsnowwolf/lede #5015: 主机型号和CPU信息好像有些错乱

ruderal04o, 2020-07-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

主机型号 | Intel(R) Core(TM) i5-7200U CPU @ 2.50GHz : 2 Core 4 Thread (CpuMark : 67405.316594 Scores)
-- | --
CPU 信息 | acpitz-acpi-0 Adapter: ACPI interface temp1: +27.8°C (crit = +119.0°C) temp2: +29.8°C (crit = +119.0°C) coretemp-isa-0000 Adapter: ISA adapter Package id 0: +43.0°C (high = +100.0°C, crit = +100.0°C) Core 0: +43.0°C (high = +100.0°C, crit = +100.0°C) Core 1: +43.0°C (high = +100.0°C, crit = +100.0°C) 637.706 MHz +43.0°C
固件版本 | OpenWrt R20.7.1 / LuCI Master (git-20.187.38662-0cb730a)

### (1) 具体问题
主机型号和CPU信息好像有些错乱
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

wangzhuok, 2020-07-07:
确实错乱

CPU 信息 | soc_dts0-virtual-0 Adapter: Virtual device temp1: +41.0°C acpitz-acpi-0 Adapter: ACPI interface temp1: +26.8°C (crit = +90.0°C) soc_dts1-virtual-0 Adapter: Virtual device temp1: +42.0°C coretemp-isa-0000 Adapter: ISA adapter Core 0: +43.0°C (high = +105.0°C, crit = +105.0°C) Core 1: +43.0°C (high = +105.0°C, crit = +105.0°C) Core 2: +44.0°C (high = +105.0°C, crit = +105.0°C) Core 3: +44.0°C (high = +105.0°C, crit = +105.0°C) 1413.813 MHz +43.0°C
-- | --

lxwq, 2020-07-07:
乱了

aglent, 2020-07-07:
信息量多了 4个核的温度都显示了 就是排版乱

coolsnowwolf, 2020-07-07:
新版 sensor 格式改了，晚点再看看

ruderal04o, 2020-07-07:
> 新版 sensor 格式改了，晚点再看看

真的辛苦了，这几天更新很猛，还请注意休息

closed 2020-07-08 by coolsnowwolf