coolsnowwolf/lede #5018: 最新编译的固件，Turbo ACC 网络中 FULLCONE NAT 加速显示未运行

tonyliangli, 2020-07-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：7月1日晚最新编译的固件，5.4 内核，防火墙中打开“启用FullCone-NAT”，但是在 Turbo ACC 网络中“FULLCONE NAT 加速”显示“未运行“，之前 4.19 内核没有问题。看到这个版本集成的是 “sfe” 而不再是“flowoffload”，请问和这个有关系吗？使用的是 target 中 .config-5.4 的配置，没有动这方面的配置。

### (2) 路由器型号和固件版本

A：软路由 x86_64

### (3) 详细日志

A：系统日志和内核日志中均无相关信息

closed 2020-07-07 by tonyliangli

jikkyfu, 2020-07-11:
请问你是怎么解决的啊
SFE Fast Path 我这个开了没用。FULLCONE NAT 开了也没用。

tonyliangli, 2020-07-12:
> 请问你是怎么解决的啊
> 
> SFE Fast Path 我这个开了没用。FULLCONE NAT 开了也没用。
> 
> 

SFE 无法启动不是很清楚。FULLCONE NAT 我是把防火墙里下的 WAN 区域的规则删除了，这样 FULLCONE NAT 就无法启动了，即使打上勾。

jikkyfu, 2020-07-19:
找到问题了，这个SFE 需要编译IPV6才能正常打开，这个FULLCONE NAT 需要 勾选后面两个选项。