coolsnowwolf/lede #5020: miniupnpd最新版 NAT-PMP 和 UPnP 都无法使用

msoayu56, 2020-07-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
使用最新的源码编译 miniupnpd NAT-PMP 和 UPnP 都无法使用
比特彗星提示 **UPnP NAT端口映射:	失败 [MiniUPnP (http://miniupnp.free.fr/)]**

原来的源码编译则没问题，是不是由于miniupnpd的版本更新导致的？
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：新路由3

### (3) 详细日志

A：
Tue Jul  7 11:00:11 2020 daemon.err miniupnpd[3466]: Failed to add NAT-PMP 47338 tcp->192.168.2.100:47338 'NAT-PMP 47338 tcp'
Tue Jul  7 11:00:11 2020 daemon.err miniupnpd[3466]: Failed to add NAT-PMP 47338 udp->192.168.2.100:47338 'NAT-PMP 47338 udp'
Tue Jul  7 11:00:11 2020 daemon.err miniupnpd[3466]: upnp_event_recv: recv(): Connection reset by peer
Tue Jul  7 11:00:11 2020 daemon.err miniupnpd[3466]: upnpevents_processfds: 0x4287f0, remove subscriber uuid:82a1f7c6-22c0-455b-bdec-29dda616a63d after an ERROR cb: http://192.168.2.100:2869/upnp/eventing/rnocpffhvb
Tue Jul  7 11:00:11 2020 daemon.err miniupnpd[3466]: upnp_event_recv: recv(): Connection reset by peer
Tue Jul  7 11:00:11 2020 daemon.err miniupnpd[3466]: upnpevents_processfds: 0x4287f0, remove subscriber uuid:5f85017c-a892-45af-ac00-69f62d4a61c8 after an ERROR cb: http://192.168.2.100:2869/upnp/eventing/famoiqmwaa

这个是bittorrent进行nat-pmp映射操作时的log，比特彗星和其他使用upnp映射的软件没有日志出来...

coolsnowwolf, 2020-07-07:
如图设置

[image]

lynnial, 2020-07-08:
同样问题，卸载了mwan3还是不行，miniupnpd的版本太新了

coolsnowwolf, 2020-07-08:
有一说一，我之前维护的那个 miniupnpd 版本确实更适合国内的应用。。。。

coolsnowwolf, 2020-07-08:
[image]

测试可用

msoayu56, 2020-07-08:
20170421确实好用。

coolsnowwolf, 2020-07-08:
[image]

[image]

新版本已做过调整，迅雷，BitComet 和 QBT 都已通过测试正常

EnnawYang, 2020-07-08:
u1s1，国内环境和国外不一样的

dolphinpaopao, 2020-07-08:
我测试过新版5.4x86，哪怕是打开v1，迅雷部分好用的版本包括绿色版，是无法成功打开upnp，bit彗星，微信这些正常。5.4内核上，ssr+还会出现运行不稳定，频繁切换节点的情况，并不是节点故障。只需要把内核切换回4.19和2017版本，恢复正常。所以新内核暂时观望。

xhqpp, 2020-07-09:
> [image]
> 
> [image]
> 
> 新版本已做过调整，迅雷，BitComet 和 QBT 都已通过测试正常

这是哪个版本号的upnp？

dolphinpaopao, 2020-07-09:
> > [image]
> > [image]
> > 新版本已做过调整，迅雷，BitComet 和 QBT 都已通过测试正常
> 
> 这是哪个版本号的upnp？

目前最新2.1稳定版。2020年。

helianling, 2020-07-09:
新版本用最新版迅雷是有问题的，列表是不显示的，我尝试换回20170421版，但编译报错。

迅雷开了会员，下载速度不怎么受影响。
BT可以自己在防火墙打开端口。

lynnial, 2020-07-10:
昨天试了下新版5.4.50，只接lan口的情况下客户端开着迅雷upnp正常，设置成旁路由以后（主路由dhcp）还是不行，同环境下切换回6月的老版本是可以的。还有个奇怪的问题，6月10号以后的版本不知道为什么speedtest会连不上，connection failed，之前的版本都没问题，日志也没看出有啥问题。另外还想问下，actions里面workflow的云编译版本是不是没饮料了。。。要怎么才可以单独安装哈，小白一个，望有大神明示，谢啦~

msoayu56, 2020-07-10:
实测最新版依旧不行

fonlan, 2020-07-10:
我还遇到过很奇怪的现象，同样作为主路由，IP地址设为192.168.123.3的话有些软件的UPNP就不工作，比如群晖。但是把路由器IP地址改成192.168.123.1的话UPNP就正常了

helianling, 2020-07-11:
只要开了IPV6，必然无法正常工作，关了就可以了。
以前老版本2017的即使开了ipv6,还是可以正常运行的。

当前版本，启动的时候会判断有没有ipv6，有的话优先使用ipv6，没有ipv6就会关掉ipv6,现在通过看系统日志可以分析出来。

目前广大移动用户没有公网ipv4,只公网ipv6，所以移动用户想用DDNS,也只能用ipv6的。

移动宽带数量非常庞大，移动免费送，不要白不要。

不确定新版upnp有没有配置文件直接关闭对IPV6的支持，只解析IPV4,像2017版一样。

caicaicai21, 2020-07-11:
确实是用不了了
[image]

alepennyq, 2020-07-12:
我的也用不了。
日志：
Sun Jul 12 18:31:53 2020 daemon.notice miniupnpd[15716]: HTTP listening on port 5000
Sun Jul 12 18:31:53 2020 daemon.notice miniupnpd[15716]: Listening for NAT-PMP/PCP traffic on port 5351
Sun Jul 12 18:32:38 2020 authpriv.info dropbear[16214]: Child connection from 192.168.1.102:65344
Sun Jul 12 18:32:38 2020 authpriv.notice dropbear[16214]: Password auth succeeded for 'root' from 192.168.1.102:65344
Sun Jul 12 18:32:45 2020 daemon.err miniupnpd[15716]: Failed to add NAT-PMP 20492 tcp->192.168.1.101:32400 'NAT-PMP 20492 tcp'

设置如图：
[image]

版本号：
[image]

toma77777, 2020-07-13:
7.11拉取的源码也用不了，版本是X64

`Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=474, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=474, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=490, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=486, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=474, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=472, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=419, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=458, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=419, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=478, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=419, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=482, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=410, dest=[ff0e::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=474, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=474, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=490, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=486, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=474, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=472, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=419, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=458, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=419, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=478, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=419, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=482, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=410, dest=[ff05::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=474, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=474, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=490, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=486, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=474, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=472, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=419, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=458, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=419, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=478, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=419, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=482, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=410, dest=[ff02::c]:1900): sendto: Permission denied
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=7, len=472, dest=239.255.255.250:1900): sendto: No such device
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=7, len=472, dest=239.255.255.250:1900): sendto: No such device
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=7, len=488, dest=239.255.255.250:1900): sendto: No such device
[... cut ...]
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=7, len=476, dest=239.255.255.250:1900): sendto: No such device
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=7, len=417, dest=239.255.255.250:1900): sendto: No such device
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=7, len=480, dest=239.255.255.250:1900): sendto: No such device
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=7, len=408, dest=239.255.255.250:1900): sendto: No such device
Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto failed to send 52 packets
`

lynnial, 2020-07-13:
> 只要开了IPV6，必然无法正常工作，关了就可以了。
> 以前老版本2017的即使开了ipv6,还是可以正常运行的。
> 
> 当前版本，启动的时候会判断有没有ipv6，有的话优先使用ipv6，没有ipv6就会关掉ipv6,现在通过看系统日志可以分析出来。
> 
> 目前广大移动用户没有公网ipv4,只公网ipv6，所以移动用户想用DDNS,也只能用ipv6的，主流BT客户端现在也支持IPV6了，虽然资源还是比较少的，但总比没有强。所以对于移动用户，IPV6是刚需。
> 
> 移动宽带数量非常庞大，移动免费送，不要白不要。
> 
> 最新版的upnp是可以直接关闭对IPV6的支持，只解析IPV4,像2017版一样。
> 
> 官方的changelog
> https://miniupnp.tuxfamily.org/files/changelog.php?file=miniupnpd-2.2.0-RC1.tar.gz
> 
> https://miniupnp.tuxfamily.org/files/changelog.php?file=miniupnpd-2.1.20200510.tar.gz
> 
> 2020/04/09:
> Option to disable IPv6 at runtime : -4 / ipv6_disable=yes
> 
> 找到解决办法了：
> 
> https://github.com/coolsnowwolf/packages/blob/master/net/miniupnpd/files/miniupnpd.init
> 
> [image]
> 
> IPV6下也正常了：
> 
> [image]

这个能通过SSH改么，能否指导一下，回去试试，谢啦

caicaicai21, 2020-07-13:
经过查找miniupnpd的源码，我找到了一个地方
如果 miniupnpd 发现外网的ip是内部地址（符合以下地址类型）
[image]
就会禁止添加转发。
[image]
但是根据提示如果开启 ext_perform_stun=yes ，还是可以打开转发功能，经过翻看启动文件啥的
在 /etc/config/upnpd 添加这两项，即可开启转发
[image]
option use_stun '1'
option stun_host 'stun.ideasip.com'
然后测试就正常了
[image]
如果有遇到此问题的，可以尝试以下。

82kg, 2020-07-13:
> 经过查找miniupnpd的源码，我找到了一个地方
> 如果 miniupnpd 发现外网的ip是内部地址（符合以下地址类型）
> [image]
> 就会禁止添加转发
> [image]
> 但是根据提示如果开启 ext_perform_stun=yes ，还是可以打开转发功能，经过翻看启动文件啥的
> 在 /etc/config/upnpd 添加这两项，即可开启转发
> [image]
> **option use_stun '1' option stun_host 'stun.ideasip.com'**
> 
> 然后测试就正常了
> [image]
> [image]
> 如果有遇到此问题的，可以尝试一下。

改完后，要重启upnp服务吗

82kg, 2020-07-13:
> > 经过查找miniupnpd的源码，我找到了一个地方
> > 如果 miniupnpd 发现外网的ip是内部地址（符合以下地址类型）
> > [image]
> > 就会禁止添加转发
> > [image]
> > 但是根据提示如果开启 ext_perform_stun=yes ，还是可以打开转发功能，经过翻看启动文件啥的
> > 在 /etc/config/upnpd 添加这两项，即可开启转发
> > [image]
> > **option use_stun '1' option stun_host 'stun.ideasip.com'**
> > 然后测试就正常了
> > [image]
> > [image]
> > 如果有遇到此问题的，可以尝试一下。
> 
> 改完后，要重启upnp服务吗,，重启upnp后还是没用。

caicaicai21, 2020-07-13:
> 经过查找miniupnpd的源码，我找到了一个地方
> 如果 miniupnpd 发现外网的ip是内部地址（符合以下地址类型）
> [image]
> 就会禁止添加转发
> [image]
> 但是根据提示如果开启 ext_perform_stun=yes ，还是可以打开转发功能，经过翻看启动文件啥的
> 在 /etc/config/upnpd 添加这两项，即可开启转发
> [image]
> **option [...]

发一下系统日志，我可以具体看一下是什么问题（upnp勾选 启用额外的日志）

alepennyq, 2020-07-13:
> 发一下系统日志，我可以具体看一下是什么问题（upnp勾选 启用额外的日志）

按你的方法之前：
Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: version 2.1.20200510 starting NAT-PMP/PCP UPnP-IGD ext if br-lan BOOTID=1594633439
Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: specific IPv6 ext if
Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: Reserved / private IP address 192.168.1.1 on ext interface br-lan: Port forwarding is impossible
Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: You are probably behind NAT, enable option ext_perform_stun=yes to detect public IP address
Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: Or use ext_ip= / -o option to declare public IP address
Mon Jul 13 17:43:59 2020 daemon.notice miniupnpd[13520]: HTTP listening on port 5000
Mon Jul 13 17:43:59 2020 daemon.notice miniupnpd[13520]: Listening for NAT-PMP/PCP traffic on port 5351
Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: HTTP REQUEST from 192.168.1.25:45399 : GET /rootDesc.xml (HTTP/1.1)
Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: accept-language HTTP header : 'en-us'
Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: Host: 192.168.1.1:5000
Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: HTTP REQUEST from 192.168.1.25:45400 : GET /rootDesc.xml (HTTP/1.1)
Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: accept-language HTTP header : 'en-us'
Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: Host: 192.168.1.1:5000
Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: HTTP REQUEST from 192.168.1.25:45401 : GET /rootDesc.xml (HTTP/1.1)
Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: accept-language HTTP header : 'en-us'
Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: Host: 192.168.1.1:5000
Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: HTTP REQUEST from 192.168.1.25:45402 : GET /rootDesc.xml (HTTP/1.1)
Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: accept-language HTTP header : 'en-us'
Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: Host: 192.168.1.1:5000
Mon Jul 13 17:44:00 2020 daemon.debug miniupnpd[13520]: level=0 type=8
Mon Jul 13 17:44:00 2020 daemon.debug miniupnpd[13520]: ifindex = 6  192.168.1.1
Mon Jul 13 17:44:00 2020 daemon.debug miniupnpd[13520]: level=0 type=8

按你的方法之后：
Mon Jul 13 17:46:50 2020 daemon.info miniupnpd[15317]: version 2.1.20200510 starting NAT-PMP/PCP UPnP-IGD ext if br-lan BOOTID=1594633610
Mon Jul 13 17:46:50 2020 daemon.info miniupnpd[15317]: specific IPv6 ext if
Mon Jul 13 17:46:50 2020 daemon.info miniupnpd[15317]: STUN: Performing with host=stun.ideasip.com and port=0 ...
Mon Jul 13 17:46:50 2020 daemon.debug miniupnpd[15317]: resolve_stun_host: stun.ideasip.com:3478 => 77.72.169.213:3478
Mon Jul 13 17:46:50 2020 daemon.info miniupnpd[15317]: perform_stun: local ports 50018 35361 33743 58709
Mon Jul 13 17:46:50 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: waiting 3 secs and 0 usecs
Mon Jul 13 17:46:50 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: received responses: 1
Mon Jul 13 17:46:50 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: waiting 3 secs and 0 usecs
Mon Jul 13 17:46:53 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: select(): no more responses
Mon Jul 13 17:46:53 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: waiting 3 secs and 0 usecs
Mon Jul 13 17:46:56 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: select(): no more responses
Mon Jul 13 17:46:56 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: waiting 3 secs and 0 usecs
Mon Jul 13 17:46:59 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: select(): no more responses
Mon Jul 13 17:46:59 2020 daemon.debug miniupnpd[15317]: parse_stun_response: Type 0x0101, Length 36, Magic Cookie 2112a442
Mon Jul 13 17:46:59 2020 daemon.debug miniupnpd[15317]: parse_stun_response: MAPPED-ADDRESS 222.79.50.204:50018
Mon Jul 13 17:46:59 2020 daemon.debug miniupnpd[15317]: parse_stun_response: SOURCE-ADDRESS 77.72.169.213:3478
Mon Jul 13 17:46:59 2020 daemon.debug miniupnpd[15317]: parse_stun_response: CHANGED-ADDRESS 77.72.169.212:3479
Mon Jul 13 17:46:59 2020 daemon.notice miniupnpd[15317]: perform_stun: 1 response out of 4 received
Mon Jul 13 17:46:59 2020 daemon.warn miniupnpd[15317]: STUN: ext interface br-lan with IP address 192.168.1.1 is now behind restrictive NAT with public IP address 222.79.50.204: Port forwarding is now impossible
Mon Jul 13 17:46:59 2020 daemon.notice miniupnpd[15317]: HTTP listening on port 5000
Mon Jul 13 17:46:59 2020 daemon.notice miniupnpd[15317]: Listening for NAT-PMP/PCP traffic on port 5351
Mon Jul 13 17:47:00 2020 daemon.info miniupnpd[15317]: HTTP REQUEST from 192.168.1.25:45447 : GET /rootDesc.xml (HTTP/1.1)
Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: accept-language HTTP header : 'en-us'
Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: Host: 192.168.1.1:5000
Mon Jul 13 17:47:00 2020 daemon.info miniupnpd[15317]: HTTP REQUEST from 192.168.1.25:45448 : GET /rootDesc.xml (HTTP/1.1)
Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: accept-language HTTP header : 'en-us'
Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: Host: 192.168.1.1:5000
Mon Jul 13 17:47:00 2020 daemon.info miniupnpd[15317]: HTTP REQUEST from 192.168.1.25:45450 : GET /rootDesc.xml (HTTP/1.1)
Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: accept-language HTTP header : 'en-us'
[... cut ...]
Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: accept-language HTTP header : 'en-us'
Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: Host: 192.168.1.1:5000
Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: level=0 type=8
Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: ifindex = 6  192.168.1.1
Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: level=0 type=8
Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: ifindex = 6  192.168.1.1
Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: level=0 type=8

caicaicai21, 2020-07-13:
> 按你的方法之前：
> Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: version 2.1.20200510 starting NAT-PMP/PCP UPnP-IGD ext if br-lan BOOTID=1594633439
> Mon Jul 13 17:43:59 2020 daemon.info [...]

[image]
按照日志，依旧不可用，我明天再翻一下源码，找找解决方法。

caicaicai21, 2020-07-14:
> 按你的方法之前：
> Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: version 2.1.20200510 starting NAT-PMP/PCP UPnP-IGD ext if br-lan BOOTID=1594633439
> Mon Jul 13 17:43:59 2020 daemon.info [...]

查了一下，你的情况是，它对stun服务器发送了4个请求，只成功接收了一个，判定你的是限制型NAT，就不给你开放功能了。
这种情况除了，改源码 miniupnpd.c 的第 2152 行，disable_port_forwarding = 1 改为 0，别无他法。
而且限制型NAT开了 upnp 也没有什么效果。

caicaicai21, 2020-07-14:
> 按你的方法之前：
> Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: version 2.1.20200510 starting NAT-PMP/PCP UPnP-IGD ext if br-lan BOOTID=1594633439
> Mon Jul 13 17:43:59 2020 daemon.info [...]

我发现一个骚操作
[image]
option external_ip '随意一个公网IP'
就可以强制开启转发功能。
不过实际效果可能是没有，只是心理上的安慰。

caicaicai21, 2020-07-14:
use_stun 这个和 external_ip 不能共存，二选一，开启 use_stun 需要同时配置 stun_host。

82kg, 2020-07-14:
> 发一下系统日志，我可以具体看一下是什么问题（upnp勾选 启用额外的日志）

Tue Jul 14 10:41:31 2020 daemon.notice miniupnpd[3654]: shutting down MiniUPnPd
Tue Jul 14 10:41:31 2020 daemon.info miniupnpd[10636]: system uptime is 413301 seconds
Tue Jul 14 10:41:31 2020 daemon.info miniupnpd[10636]: Reloading rules from lease file
Tue Jul 14 10:41:31 2020 daemon.info miniupnpd[10636]: version 2.1.20200510 starting NAT-PMP/PCP UPnP-IGD ext if eth2 BOOTID=1594694491
Tue Jul 14 10:41:31 2020 daemon.info miniupnpd[10636]: specific IPv6 ext if
Tue Jul 14 10:41:31 2020 daemon.info miniupnpd[10636]: STUN: Performing with host=stun.ideasip.com and port=0 ...
Tue Jul 14 10:41:31 2020 daemon.debug miniupnpd[10636]: resolve_stun_host: stun.ideasip.com:3478 => 77.72.169.211:3478
Tue Jul 14 10:41:31 2020 daemon.info miniupnpd[10636]: perform_stun: local ports 40444 54541 47899 43511
Tue Jul 14 10:41:31 2020 daemon.debug miniupnpd[10636]: wait_for_stun_responses: waiting 3 secs and 0 usecs
Tue Jul 14 10:41:34 2020 daemon.debug miniupnpd[10636]: wait_for_stun_responses: select(): no more responses
Tue Jul 14 10:41:34 2020 daemon.debug miniupnpd[10636]: wait_for_stun_responses: waiting 3 secs and 0 usecs
Tue Jul 14 10:41:37 2020 daemon.debug miniupnpd[10636]: wait_for_stun_responses: select(): no more responses
Tue Jul 14 10:41:37 2020 daemon.debug miniupnpd[10636]: wait_for_stun_responses: waiting 3 secs and 0 usecs
Tue Jul 14 10:41:40 2020 daemon.debug miniupnpd[10636]: wait_for_stun_responses: select(): no more responses
Tue Jul 14 10:41:40 2020 daemon.err miniupnpd[10636]: STUN: Performing STUN failed: No such device or address
Tue Jul 14 10:41:40 2020 daemon.err miniupnpd[10636]: Performing STUN failed. EXITING

toma77777, 2020-07-14:
最新版源码，经测试不用修改楼上的设置，就可以正常使用。

固件版本：OpenWrt R20.7.7 / LuCI Master (git-20.191.36863-eee6bae)

caicaicai21, 2020-07-14:
> 最新版源码，经测试不用修改楼上的设置，就可以正常使用。
> 
> 固件版本：OpenWrt R20.7.7 / LuCI Master (git-20.191.36863-eee6bae)

如果直接路由器拨号，有公网，是可以正常使用的。

toma77777, 2020-07-14:
> > 最新版源码，经测试不用修改楼上的设置，就可以正常使用。
> > 固件版本：OpenWrt R20.7.7 / LuCI Master (git-20.191.36863-eee6bae)
> 
> 如果直接路由器拨号，有公网，是可以正常使用的。

上几天编译的版本，显示如下错误哦
`Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=474, dest=[ff0e::c]:1900): sendto: Permission denied`

caicaicai21, 2020-07-14:
> Tue Jul 14 10:41:31 2020 daemon.notice miniupnpd[3654]: shutting down MiniUPnPd
> Tue Jul 14 10:41:31 2020 daemon.info miniupnpd[10636]: system uptime is 413301 seconds
> Tue Jul 14 10:41:31 2020 [...]

它找不到你 IPV6 的 if，先禁用 miniupnpd 的 ipv6
配置加上
option ipv6_disable '1'

caicaicai21, 2020-07-14:
> > > 最新版源码，经测试不用修改楼上的设置，就可以正常使用。
> > > 固件版本：OpenWrt R20.7.7 / LuCI Master (git-20.191.36863-eee6bae)
> > 
> > 
> > 如果直接路由器拨号，有公网，是可以正常使用的。
> 
> 上几天编译的版本，显示如下错误哦
> `Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=474, dest=[ff0e::c]:1900): sendto: Permission denied`

这个偶尔出现不是什么大问题，它只是发送数据包错误而已。

alepennyq, 2020-07-14:
> > 最新版源码，经测试不用修改楼上的设置，就可以正常使用。
> > 固件版本：OpenWrt R20.7.7 / LuCI Master (git-20.191.36863-eee6bae)
> 
> 如果直接路由器拨号，有公网，是可以正常使用的。

我就是软路由拨号，有公网ip，怎么不可以啊

caicaicai21, 2020-07-14:
> > > 最新版源码，经测试不用修改楼上的设置，就可以正常使用。
> > > 固件版本：OpenWrt R20.7.7 / LuCI Master (git-20.191.36863-eee6bae)
> > 
> > 
> > 如果直接路由器拨号，有公网，是可以正常使用的。
> 
> 我就是软路由拨号，有公网ip，怎么不可以啊

所以新版miniupnpd很玄学，可以尝试以上方法，看看有没有效。

caicaicai21, 2020-07-14:
> > > > 最新版源码，经测试不用修改楼上的设置，就可以正常使用。
> > > > 固件版本：OpenWrt R20.7.7 / LuCI Master (git-20.191.36863-eee6bae)
> > > 
> > > 
> > > 如果直接路由器拨号，有公网，是可以正常使用的。
> > 
> > 
> > 我就是软路由拨号，有公网ip，怎么不可以啊
> 
> 所以新版的miniupnpd很玄学，可以尝试以上方法，看看有没有效。

caicaicai21, 2020-07-14:
> 最新版源码，经测试不用修改楼上的设置，「活动的UPnP重定向」正常，系统日志也没再有关于UPnP的错误。
> 
> 固件版本：OpenWrt R20.7.7 / LuCI Master (git-20.191.36863-eee6bae)

你是同步到哪天的？我是同步到 Jul 8, 2020。

alepennyq, 2020-07-14:
> > 按你的方法之前：
> > Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: version 2.1.20200510 starting NAT-PMP/PCP UPnP-IGD ext if br-lan BOOTID=1594633439
> > Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: specific IPv6 ext if
> > Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: Reserved / private IP address 192.168.1.1 on ext interface br-lan: Port forwarding is impossible
> > Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: You are probably behind NAT, enable option ext_perform_stun=yes to detect public IP address
> > Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: Or use ext_ip= / -o option to declare public IP address
> > Mon Jul 13 17:43:59 2020 daemon.notice miniupnpd[13520]: HTTP listening on port 5000
> > Mon Jul 13 17:43:59 2020 daemon.notice miniupnpd[13520]: Listening for NAT-PMP/PCP traffic on port 5351
> > Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: HTTP REQUEST from 192.168.1.25:45399 : GET /rootDesc.xml (HTTP/1.1)
> > Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: accept-language HTTP header : 'en-us'
> > Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: Host: 192.168.1.1:5000
> > Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: HTTP REQUEST from 192.168.1.25:45400 : GET /rootDesc.xml (HTTP/1.1)
> > Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: accept-language HTTP header : 'en-us'
> > Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: Host: 192.168.1.1:5000
> > Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: HTTP REQUEST from 192.168.1.25:45401 : GET /rootDesc.xml (HTTP/1.1)
> > Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: accept-language HTTP header : 'en-us'
> > Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: Host: 192.168.1.1:5000
> > Mon Jul 13 17:43:59 2020 daemon.info miniupnpd[13520]: HTTP REQUEST from 192.168.1.25:45402 : GET /rootDesc.xml (HTTP/1.1)
> > Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: accept-language HTTP header : 'en-us'
> > Mon Jul 13 17:43:59 2020 daemon.debug miniupnpd[13520]: Host: 192.168.1.1:5000
> > Mon Jul 13 17:44:00 2020 daemon.debug miniupnpd[13520]: level=0 type=8
> > Mon Jul 13 17:44:00 2020 daemon.debug miniupnpd[13520]: ifindex = 6 192.168.1.1
> > Mon Jul 13 17:44:00 2020 daemon.debug miniupnpd[13520]: level=0 type=8
> > 按你的方法之后：
> > Mon Jul 13 17:46:50 2020 daemon.info miniupnpd[15317]: version 2.1.20200510 starting NAT-PMP/PCP UPnP-IGD ext if br-lan BOOTID=1594633610
> > Mon Jul 13 17:46:50 2020 daemon.info miniupnpd[15317]: specific IPv6 ext if
> > Mon Jul 13 17:46:50 2020 daemon.info miniupnpd[15317]: STUN: Performing with host=stun.ideasip.com and port=0 ...
> > Mon Jul 13 17:46:50 2020 daemon.debug miniupnpd[15317]: resolve_stun_host: stun.ideasip.com:3478 => 77.72.169.213:3478
> > Mon Jul 13 17:46:50 2020 daemon.info miniupnpd[15317]: perform_stun: local ports 50018 35361 33743 58709
> > Mon Jul 13 17:46:50 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: waiting 3 secs and 0 usecs
> > Mon Jul 13 17:46:50 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: received responses: 1
> > Mon Jul 13 17:46:50 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: waiting 3 secs and 0 usecs
> > Mon Jul 13 17:46:53 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: select(): no more responses
> > Mon Jul 13 17:46:53 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: waiting 3 secs and 0 usecs
> > Mon Jul 13 17:46:56 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: select(): no more responses
> > Mon Jul 13 17:46:56 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: waiting 3 secs and 0 usecs
> > Mon Jul 13 17:46:59 2020 daemon.debug miniupnpd[15317]: wait_for_stun_responses: select(): no more responses
> > Mon Jul 13 17:46:59 2020 daemon.debug miniupnpd[15317]: parse_stun_response: Type 0x0101, Length 36, Magic Cookie 2112a442
> > Mon Jul 13 17:46:59 2020 daemon.debug miniupnpd[15317]: parse_stun_response: MAPPED-ADDRESS 222.79.50.204:50018
> > Mon Jul 13 17:46:59 2020 daemon.debug miniupnpd[15317]: parse_stun_response: SOURCE-ADDRESS 77.72.169.213:3478
> > Mon Jul 13 17:46:59 2020 daemon.debug miniupnpd[15317]: parse_stun_response: CHANGED-ADDRESS 77.72.169.212:3479
> > Mon Jul 13 17:46:59 2020 daemon.notice miniupnpd[15317]: perform_stun: 1 response out of 4 received
> > Mon Jul 13 17:46:59 2020 daemon.warn miniupnpd[15317]: STUN: ext interface br-lan with IP address 192.168.1.1 is now behind restrictive NAT with public IP address 222.79.50.204: Port forwarding is now impossible
> > Mon Jul 13 17:46:59 2020 daemon.notice miniupnpd[15317]: HTTP listening on port 5000
> > Mon Jul 13 17:46:59 2020 daemon.notice miniupnpd[15317]: Listening for NAT-PMP/PCP traffic on port 5351
> > Mon Jul 13 17:47:00 2020 daemon.info miniupnpd[15317]: HTTP REQUEST from 192.168.1.25:45447 : GET /rootDesc.xml (HTTP/1.1)
> > Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: accept-language HTTP header : 'en-us'
> > Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: Host: 192.168.1.1:5000
> > Mon Jul 13 17:47:00 2020 daemon.info miniupnpd[15317]: HTTP REQUEST from 192.168.1.25:45448 : GET /rootDesc.xml (HTTP/1.1)
> > Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: accept-language HTTP header : 'en-us'
> > Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: Host: 192.168.1.1:5000
[... cut ...]
> > Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: ifindex = 6 192.168.1.1
> > Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: level=0 type=8
> > Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: ifindex = 6 192.168.1.1
> > Mon Jul 13 17:47:00 2020 daemon.debug miniupnpd[15317]: level=0 type=8
> 
> 查了一下，你的情况是，它对stun服务器发送了4个请求，只成功接收了一个，判定你的是限制型NAT，就不给你开放功能了。
> 这种情况除了改源码 miniupnpd.c 的第 2152 行，disable_port_forwarding = 1 改为 0，别无他法(而且只改一个地方好像也不行)。
> 而且限制型NAT开了 upnp 也没有什么效果。
> 
> 我的原本就有公网IP，只不过是光猫拨号，添加这两个选项就有效。1:1非限制型NAT。

我用用电脑测NAT类型，是NAT1啊
[image]

toma77777, 2020-07-14:
> > > > 最新版源码，经测试不用修改楼上的设置，就可以正常使用。
> > > > 固件版本：OpenWrt R20.7.7 / LuCI Master (git-20.191.36863-eee6bae)
> > > 
> > > 
> > > 如果直接路由器拨号，有公网，是可以正常使用的。
> > 
> > 
> > 上几天编译的版本，显示如下错误哦
> > `Mon Jul 13 10:39:09 2020 daemon.err miniupnpd[7543]: try_sendto(sock=8, len=474, dest=[ff0e::c]:1900): sendto: Permission denied`
> 
> 这个偶尔出现不是什么大问题，它只是发送数据包错误而已。

这个错误是开机后就一直出现，每秒几条日志，哈哈。不过现在正常了就不管了。谢谢你

toma77777, 2020-07-14:
> > 最新版源码，经测试不用修改楼上的设置，「活动的UPnP重定向」正常，系统日志也没再有关于UPnP的错误。
> > 固件版本：OpenWrt R20.7.7 / LuCI Master (git-20.191.36863-eee6bae)
> 
> 你是同步到哪天的？我是同步到 Jul 8, 2020。

我是昨天（2020.7.13）重新拉取代码编译的。

caicaicai21, 2020-07-14:
> 我用用电脑测NAT类型，是NAT1啊
> [image]

那你试试把 stun_host 的值换成你软件上那个 stun 服务器地址。

caicaicai21, 2020-07-14:
> 这个错误是开机后就一直出现，每秒几条日志，哈哈。不过现在正常了就不管了。谢谢你

估计是跟IPV6有关系

helianling, 2020-07-14:
移动IPV4 内网， IPV6公网， Upnp完美运行。

1. 关闭IPV6， upnp.init 文件 ipv6_disable修改为1。

[image]

2. 修改dl文件夹下miniupnpd-2.1.20200510.tar.gz源代码，把内网IP检测方法永远return 0。

[image]

修改完重新编译，移动大内网就可以完美运行了，理论上有公网IP地址的是更没有问题的。

[image]

[image]

82kg, 2020-07-15:
> 它找不到你 IPV6 的 if，先禁用 miniupnpd 的 ipv6
> 配置加上
> option ipv6_disable '1'

 SERVER: OpenWRT/SNAPSHOT UPnP/1.1 MiniUPnPd/2.1.20200510
 LOCATION: http://192.168.2.1:5000/rootDesc.xml
 OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
 01-NLS: 1594777280
 BOOTID.UPNP.ORG: 1594777280
 CONFIGID.UPNP.ORG: 1337
Wed Jul 15 09:42:34 2020 daemon.debug miniupnpd[23286]: SendSSDPResponse(): 0 bytes to 192.168.2.2:55902 ST: HTTP/1.1 200 OK
 CACHE-CONTROL: max-age=120
 ST: urn:schemas-upnp-org:service:DeviceProtection:1
 USN: uuid:cc330da5-7a10-4f09-aea0-457e3ee0e604::urn:schemas-upnp-org:service:DeviceProtection:1
 EXT:
 SERVER: OpenWRT/SNAPSHOT UPnP/1.1 MiniUPnPd/2.1.20200510
 LOCATION: http://192.168.2.1:5000/rootDesc.xml
 OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
 01-NLS: 1594777280
 BOOTID.UPNP.ORG: 1594777280
 CONFIGID.UPNP.ORG: 1337
Wed Jul 15 09:42:34 2020 daemon.debug miniupnpd[23286]: SendSSDPResponse(): 0 bytes to 192.168.2.2:55902 ST: HTTP/1.1 200 OK
 CACHE-CONTROL: max-age=120
 ST: urn:schemas-upnp-org:service:WANIPv6FirewallControl:1
 USN: uuid:cc330da5-7a10-4f09-aea0-457e3ee0e606::urn:schemas-upnp-org:service:WANIPv6FirewallControl:1
 EXT:
 SERVER: OpenWRT/SNAPSHOT UPnP/1.1 MiniUPnPd/2.1.20200510
 LOCATION: http://192.168.2.1:5000/rootDesc.xml
 OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
 01-NLS: 1594777280
 BOOTID.UPNP.ORG: 1594777280
 CONFIGID.UPNP.ORG: 1337
Wed Jul 15 09:42:34 2020 daemon.debug miniupnpd[23286]: SendSSDPResponse(): 0 bytes to 192.168.2.2:55902 ST: HTTP/1.1 200 OK
 CACHE-CONTROL: max-age=120
 ST: urn:schemas-upnp-org:service:WANCommonInterfaceConfig:1
 USN: uuid:cc330da5-7a10-4f09-aea0-457e3ee0e605::urn:schemas-upnp-org:service:WANCommonInterfaceConfig:1
 EXT:
 SERVER: OpenWRT/SNAPSHOT UPnP/1.1 MiniUPnPd/2.1.20200510
 LOCATION: http://192.168.2.1:5000/rootDesc.xml
 OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
 01-NLS: 1594777280
 BOOTID.UPNP.ORG: 1594777280
 CONFIGID.UPNP.ORG: 1337
Wed Jul 15 09:42:34 2020 daemon.debug miniupnpd[23286]: SendSSDPResponse(): 0 bytes to 192.168.2.2:55902 ST: HTTP/1.1 200 OK
 CACHE-CONTROL: max-age=120
 ST: urn:schemas-upnp-org:service:WANPPPConnection:1
 USN: uuid:cc330da5-7a10-4f09-aea0-457e3ee0e606::urn:schemas-upnp-org:service:WANPPPConnection:1
 EXT:
 SERVER: OpenWRT/SNAPSHOT UPnP/1.1 MiniUPnPd/2.1.20200510
 LOCATION: http://192.168.2.1:5000/rootDesc.xml
 OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
 01-NLS: 1594777280
 BOOTID.UPNP.ORG: 1594777280
 CONFIGID.UPNP.ORG: 1337
Wed Jul 15 09:42:34 2020 daemon.debug miniupnpd[23286]: SendSSDPResponse(): 0 bytes to 192.168.2.2:55902 ST: HTTP/1.1 200 OK
 CACHE-CONTROL: max-age=120
 ST: urn:schemas-upnp-org:service:Layer3Forwarding:1
 USN: uuid:cc330da5-7a10-4f09-aea0-457e3ee0e604::urn:schemas-upnp-org:service:Layer3Forwarding:1
 EXT:
 SERVER: OpenWRT/SNAPSHOT UPnP/1.1 MiniUPnPd/2.1.20200510
 LOCATION: http://192.168.2.1:5000/rootDesc.xml
 OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
 01-NLS: 1594777280
 BOOTID.UPNP.ORG: 1594777280
 CONFIGID.UPNP.ORG: 1337
Wed Jul 15 09:42:34 2020 daemon.debug miniupnpd[23286]: SendSSDPResponse(): 0 bytes to 192.168.2.2:55902 ST: HTTP/1.1 200 OK
 CACHE-CONTROL: max-age=120
 ST: uuid:cc330da5-7a10-4f09-aea0-457e3ee0e604
 USN: uuid:cc330da5-7a10-4f09-aea0-457e3ee0e604
 EXT:
 SERVER: OpenWRT/SNAPSHOT UPnP/1.1 MiniUPnPd/2.1.20200510
 LOCATION: http://192.168.2.1:5000/rootDesc.xml
 OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
 01-NLS: 1594777280
 BOOTID.UPNP.ORG: 1594777280
 CONFIGID.UPNP.ORG: 1337
Wed Jul 15 09:42:34 2020 daemon.debug miniupnpd[23286]: SendSSDPResponse(): 0 bytes to 192.168.2.2:55902 ST: HTTP/1.1 200 OK
 CACHE-CONTROL: max-age=120
 ST: uuid:cc330da5-7a10-4f09-aea0-457e3ee0e605
 USN: uuid:cc330da5-7a10-4f09-aea0-457e3ee0e605
 EXT:
 SERVER: OpenWRT/SNAPSHOT UPnP/1.1 MiniUPnPd/2.1.20200510
 LOCATION: http://192.168.2.1:5000/rootDesc.xml
 OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
 01-NLS: 1594777280
 BOOTID.UPNP.ORG: 1594777280
 CONFIGID.UPNP.ORG: 1337
Wed Jul 15 09:42:34 2020 daemon.debug miniupnpd[23286]: SendSSDPResponse(): 0 bytes to 192.168.2.2:55902 ST: HTTP/1.1 200 OK
 CACHE-CONTROL: max-age=120
 ST: uuid:cc330da5-7a10-4f09-aea0-457e3ee0e606
 USN: uuid:cc330da5-7a10-4f09-aea0-457e3ee0e606
 EXT:
 SERVER: OpenWRT/SNAPSHOT UPnP/1.1 MiniUPnPd/2.1.20200510
 LOCATION: http://192.168.2.1:5000/rootDesc.xml
 OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
 01-NLS: 1594777280
 BOOTID.UPNP.ORG: 1594777280
 CONFIGID.UPNP.ORG: 1337
Wed Jul 15 09:42:34 2020 daemon.info miniupnpd[23286]: HTTP REQUEST from 192.168.2.2:48648 : GET /rootDesc.xml (HTTP/1.1)
Wed Jul 15 09:42:34 2020 daemon.debug miniupnpd[23286]: Host: 192.168.2.1:5000
Wed Jul 15 09:42:34 2020 daemon.info miniupnpd[23286]: HTTP REQUEST from 192.168.2.2:48650 : GET /rootDesc.xml (HTTP/1.1)
Wed Jul 15 09:42:34 2020 daemon.debug miniupnpd[23286]: Host: 192.168.2.1:5000
Wed Jul 15 09:42:34 2020 daemon.info miniupnpd[23286]: HTTP REQUEST from 192.168.2.2:48652 : POST /ctl/IPConn (HTTP/1.1)
Wed Jul 15 09:42:34 2020 daemon.debug miniupnpd[23286]: Host: 192.168.2.1:5000
Wed Jul 15 09:42:34 2020 daemon.info miniupnpd[23286]: SOAPAction: urn:schemas-upnp-org:service:WANIPConnection:1#GetGenericPortMappingEntry
Wed Jul 15 09:42:34 2020 daemon.info miniupnpd[23286]: GetGenericPortMappingEntry: index=0
Wed Jul 15 09:42:34 2020 daemon.info miniupnpd[23286]: Returning UPnPError 713: SpecifiedArrayIndexInvalid
Wed Jul 15 09:42:34 2020 daemon.info miniupnpd[23286]: HTTP REQUEST from 192.168.2.2:48654 : POST /ctl/IPConn (HTTP/1.1)
Wed Jul 15 09:42:34 2020 daemon.debug miniupnpd[23286]: Host: 192.168.2.1:5000
Wed Jul 15 09:42:34 2020 daemon.info miniupnpd[23286]: SOAPAction: urn:schemas-upnp-org:service:WANIPConnection:1#GetGenericPortMappingEntry
Wed Jul 15 09:42:34 2020 daemon.info miniupnpd[23286]: GetGenericPortMappingEntry: index=0
Wed Jul 15 09:42:34 2020 daemon.info miniupnpd[23286]: Returning UPnPError 713: SpecifiedArrayIndexInvalid

这样算成功了吗。我用群辉测的，群辉提示upnp通过测试，但是OP里 啥都没有
[image]

82kg, 2020-07-15:
> 它找不到你 IPV6 的 if，先禁用 miniupnpd 的 ipv6
> 配置加上
> option ipv6_disable '1'

[image]
[image]
[image]

好像还没用。。。

caicaicai21, 2020-07-15:
> [image]
> [image]
> [image]
>
> 好像还没用。。。

可以根据上面的提示，修改源码，重新编译。

caicaicai21, 2020-07-15:
> 移动IPV4内网,IPV6公网, Upnp完美运行需要关闭IPV6和修改源代码。
> 电信或联通有公网IPV4的,只需要关闭IPV6就可以完美运行了。
> 
> 1. 关闭IPV6， miniupnpd.init 文件 ipv6_disable修改为1。
> 
> [image]
> 
> 1. 修改dl文件夹下miniupnpd-2.1.20200510.tar.gz源代码，把内网IP检测方法永远return 0。
> 
> [image]
> 
> 1. 修改完重新编译，移动大内网就可以完美运行了，理论上有公网IP地址的是更没有问题的。
> 
> [image]
> 
> [image]
> 
> 1. 如果不修改源代码，只修改配置文件option external_ip '随意一个公网IP'，SOAP请求还是会失败，HTTP请求会成功，对于手机迅雷，都是SOAP请求，所以在UPNP列表不出现手机迅雷的请求，对于电脑版迅雷是有SOAP和HTTP两种请求的，HTTP请求会成功，列表中只会显示两条MiniTP SDK记录，所以必须通过修改源代码的方式才能完美修复SOAP请求。
> 
> [image]

可以整个patch。

dolphinpaopao, 2020-07-15:
> 移动IPV4内网,IPV6公网, Upnp完美运行需要关闭IPV6和修改源代码。
> 电信或联通有公网IPV4的,只需要关闭IPV6就可以完美运行了。
>
> 1. 关闭IPV6， miniupnpd.init 文件 ipv6_disable修改为1。
>
> [image]
>
> 1. 修改dl文件夹下miniupnpd-2.1.20200510.tar.gz源代码，把内网IP检测方法永远return 0。 [...]

在外网测试这个端口是开放的吗？现在最新源码编译，不用修改，电信网络测试upnp正常。

xhqpp, 2020-07-15:
> 移动IPV4内网,IPV6公网, Upnp完美运行需要关闭IPV6和修改源代码。
> 电信或联通有公网IPV4的,只需要关闭IPV6就可以完美运行了。
>
> 1. 关闭IPV6， miniupnpd.init 文件 ipv6_disable修改为1。
>
> [image]
>
> 1. 修改dl文件夹下miniupnpd-2.1.20200510.tar.gz源代码，把内网IP检测方法永远return 0。 [...]

内网开upnp有作用吗

chiyu521, 2020-07-15:
> > 移动IPV4内网，IPV6公网，Upnp完美运行需要关闭IPV6和修改源代码。
> > 电信或联通有公网IPV4的，只需要关闭IPV6就可以完美运行了。
> > 
> > 1. 关闭IPV6，miniupnpd.init文件ipv6_disable修改为1。
> > 
> > [image]
> > 
> > 1. 修改dl文件夹下miniupnpd-2.1.20200510.tar.gz源代码，把内网IP检测方法永远返回0。
> > 
> > [image]
> > 
> > 1. 修改完重新编译，移动大内网就可以完美运行了，理论上有公网IP地址的是更没有问题的。
> > 
> > [image]
> > [image]
> > 
> > 1. 如果不修改源代码，只修改了配置文件option external_ip'随意一个公网IP'，SOAP请求还是会失败，HTTP请求会成功，对于手机迅雷，都是SOAP请求，所以在UPNP列表不出现手机迅雷的请求，对于电脑版迅雷是有SOAP和HTTP两个请求的，HTTP请求会成功，列表中只会显示两个MiniTP SDK记录，因此必须通过修改源代码的方式才能完美地修复SOAP请求。
> > 
> > [image]
> 
> 内网开upnp有作用吗

我按照您的方法 玩街霸依然是无法和朋友联机 换回老版本的固件就没问题  头疼

zjy2200656, 2020-07-15:
不折腾了，老老实实换回旧版2017的miniupnpd，完美工作

alepennyq, 2020-07-16:
> 不折腾了，老老实实换回旧版2017的miniupnpd，完美工作

2017版的ipk麻烦发我邮箱下 alex_qiu@126.com 谢谢

82kg, 2020-07-16:
> 不折腾了，老老实实换回旧版2017的miniupnpd，完美工作

sun82kg@qq.com 谢谢

82kg, 2020-07-16:
> 不折腾了，老老实实换回旧版2017的miniupnpd，完美工作

是在5.4内核下编译的2017版吗

zjy2200656, 2020-07-16:
> > 不折腾了，老老实实换回旧版2017的miniupnpd，完美工作
> 
> 是在5.4内核下编译的2017版吗

是的

82kg, 2020-07-16:
> > > 不折腾了，老老实实换回旧版2017的miniupnpd，完美工作
> > 
> > 
> > 是在5.4内核下编译的2017版吗
> 
> 是的

谢谢，你是arm的cpu？ 我是x86的。。。。

zjy2200656, 2020-07-16:
> > > > 不折腾了，老老实实换回旧版2017的miniupnpd，完美工作
> > > 
> > > 
> > > 是在5.4内核下编译的2017版吗
> > 
> > 
> > 是的
> 
> 谢谢，你是arm的cpu？ 我是x86的。。。。

是啊 armv7

msoayu56, 2020-07-16:
> > > > 不折腾了，老老实实换回旧版2017的miniupnpd，完美工作
> > > 
> > > 
> > > 是在5.4内核下编译的2017版吗
> > 
> > 
> > 是的
> 
> 谢谢，你是arm的cpu？ 我是x86的。。。。

我有个lede-test的repo，你可以试试，软件包都是老的。

DanielChen008, 2020-07-17:
> > > > > 不折腾了，老老实实换回旧版2017的miniupnpd，完美工作
> > > > 
> > > > 
> > > > 是在5.4内核下编译的2017版吗
> > > 
> > > 
> > > 是的
> > 
> > 
> > 谢谢，你是arm的cpu？ 我是x86的。。。。
> 
> 是啊 armv7

2017 版本的在哪找？我也依旧不能打开UPNP，方便的话告之一下，晚上回去编译一个 x86 的替换掉好了~

[38 more replies]

closed 2020-07-29 by msoayu56