coolsnowwolf/lede #5021: 关于最新采用官方的 Package 之后的一些所谓 “Bug” 的说明

coolsnowwolf, 2020-07-07:
长期以来，这个源代码一直被很多人诟病的地方，就是 Package 不够“新”， 没有天天随着官方 Op 更新

所以这次索性直接切换到官方的的 trunk package ，满足大家天天“新”的愿望

所以以下功能并不是 Bug，而且官方的功能，包含且不限于以下问题：

1. DDNS 脚本不接受 （*） 的泛域名
2. 不支持 NTFS-3G 自动挂载
3. x64 的 Grub 在某些固件上必须手动键盘回车才接着启动
4. fork-daap 之类某次 patch 不能通过编译，或者编译后不能打开
5. Transmission 的 luci 不自动关联主程序
6. gstreamer 版本不匹配导致 UPNP render 不能用
7. MINUPNPD 官方版本对国内常用软件兼容欠佳

既然大家都喜欢“新” 和 “官方”， 我也没必要去维护一个所谓的“好用” 稳定版，白浪费精力不说还被人怨念。

PS: luci 要不也切换为 官方的 js 版本，这样比较新？

zerofo, 2020-07-07:
难怪... 从上周开始 觉得有哪里不对。瞎捣鼓了一下才编译过，用着也没那么顺畅。   不过我无所谓，用着还是比 koolshare的自在。

hcym, 2020-07-07:
爱的代价😂

dolphinpaopao, 2020-07-07:
支持更新，不过我还用着旧版本。内核快速迭代更新+应用层的package稳定=兼顾硬件支持安全修复和持续使用。
路由作为网关，丢在角落忘记它存在默默服务才是本职工作的最佳体现。
作为开源项目要折腾可以去官方openwrt折腾，自己多学编程知识自己修改各种随意折腾，开源项目不是来折腾作者的。

weiyutf, 2020-07-07:
😂不折腾了，稳定才是最重要的

9l, 2020-07-07:
别啊大佬，别理那些无理取闹的人了，换回来吧

fhbyljj, 2020-07-07:
不是很懂，这意味着什么？是不是像小米稳定版和开发板之间的区别？官方的不太稳定？大雕的更加稳定

fhbyljj, 2020-07-07:
其实就是当初大雕固件的比明月稳我才一直用大雕的，估计都有两年了吧，大雕你坚持自己就好

fhbyljj, 2020-07-07:
他们有意见，想追新，他们自己用官方源码编译就行了，反正我一直用着你源码编译的固件

ivybridgei, 2020-07-07:
大雕还是换回来吧，好多包都是老版的稳定，比如upnp，之前的2017年版和国内大多数软件都兼容，游戏机也兼容，，2019以后的版本就不行了，没必要跟着官方追新，哪个稳定用哪个。

ivybridgei, 2020-07-07:
用大雕源码就是图个稳定，想天天折腾的直接拉官方源码编译不就好了

georip, 2020-07-07:
没有大雕特色的。稳中求进才是大雕。SB们就喜欢追新。

hyy-666, 2020-07-07:
事到如今，说啥也没用了，大雕加油吧。

coolsnowwolf, 2020-07-07:
其实只想表达，又快又新很难同时做到的，毕竟我自己才一个人，也不是职业只维护这个项目，只能取舍

evangelion015, 2020-07-07:
求稳，大雕坚持自己的主张

Ozzy-hzy, 2020-07-07:
理解万岁

DanielChen008, 2020-07-07:
很多人都是伸手索取，不知道开源的意义，如楼上所说开源不是让你们来折腾作者.
issues 或者类似我们公司内使用的ticket 是给使用者反馈BUG，不是提需求，提新需求在那种外包公司是要加钱的好么，请回到首页扫支付宝捐赠先 :)

coolsnowwolf, 2020-07-07:
MINUPNPD 早期版本兼容国内更多常用应用，但在新的内核已经不能编译通过了，需要打补丁

hyy-666, 2020-07-07:
> 其实只想表达，又快又新很难同时做到的，毕竟我自己才一个人，也不是职业只维护这个项目，只能取舍

大雕，有机会回退的话，请您务必保留SFE，SFE的效果貌似比之前的好很多,性能暂不详，谢谢你维护了一份这么好的固件。

hyy-666, 2020-07-07:
> 很多人都是伸手索取，不知道开源的意义，如楼上所说开源不是让你们来折腾作者.
> issues 或者类似我们公司内使用的ticket 是给使用者反馈BUG，不是提需求，提新需求在那种外包公司是要加钱的好么，请回到首页扫支付宝捐赠先 :)

非常有道理，很多人就是白嫖还嘴臭。

BOBINIUNIU, 2020-07-07:
有的人就是捐赠的时候没看到人，提起要求来一大堆。新版不能提高性能且不说，反而各种问题，还是退回到4.19比较稳定。
之前看到有人说5.4内核比4.19内核的cpu占用率和温度下降20%，反正我是没体验出来，CPU温度和4.19一样，甚至还高点。
爱追新的去隔壁刷闭源固件吧。

Felix2yu, 2020-07-07:
这些新的功能有考虑推给上游吗，如果曾经尝试过那上有维护者是什么理由拒绝的？

Aes64X, 2020-07-07:
提需求？请亮出你的捐赠数额截图，看够不够这个需求。

coolsnowwolf, 2020-07-07:
> 这些新的功能有考虑推给上游吗，如果曾经尝试过那上有维护者是什么理由拒绝的？

例如默认禁止返回 IPv6 记录，老外就觉得一脸问号，他们想象不到为啥会有这样的墙和场景

Aes64X, 2020-07-07:
> 这些新的功能有考虑推给上游吗，如果曾经尝试过那上有维护者是什么理由拒绝的？

上游维护的思维很奇特。你可以试试把大雕里有的设备支持推过去审核看看，那家伙。。。这也不让改那也不行的，满篇的问你为什么改这为什么改那。退个通用补丁还行，这么多“特色”的东西解释都要解释上半个月吧。

eallion, 2020-07-07:
路由器这种东西，稳定才是最重要的。

BOBINIUNIU, 2020-07-07:
> 的设备支持

哈哈，说得没错，这些有中国特色的功能他们是无法理解的。作为路由器系统，能适应当下的网络环境才是最重要的。

dingpengyu, 2020-07-07:
> 其实只想表达，又快又新很难同时做到的，毕竟我自己才一个人，也不是职业只维护这个项目，只能取舍
坚持自己即可。

kkss2020, 2020-07-07:
同感,开源不是用来折腾作者的,坚持自己把合适的做好才是正道,想最新可以去隔壁去官方好了.

iceman0911, 2020-07-07:
老路由，隔一段时间自己编译升级一下，这次升完无限重启，重刷还是用问题，后来只能刷回官方，然后编译了openwrt那个稳定版刷上去定定心心用吧。

littleBanshee, 2020-07-07:
router,  nothing but stability

xiehuayi75, 2020-07-07:
大雕怎能让他人左右

xiehuayi75, 2020-07-07:
自己的项目自己做主，岂能让他人左右。，安心做自己就好

wushenghn, 2020-07-07:
刚编了修改root密码后web登不上了，

wushenghn, 2020-07-07:
web不能修改密码，命令修改web登不上

hiliuyj, 2020-07-07:
我是一个做技术的, 现在越来越认识到，协调与管理也是一个很麻烦的工作。

aizhouyi, 2020-07-08:
我觉得大雕做自己就好啊，大雕只是分享自己稳定用就好了。白嫖还提需求？issue是问题意思，issue是用来反馈bug的，不是用来提需求的。做自己的项目，别人爱用不用。给惯的。

songbangyan, 2020-07-08:
绝大部分程序员应该都理解，新即意味着还有大部分未发现的bug。
所以对于需要最新的人，你可以直接屏蔽。
还是稳定最重要！！

fs20sc, 2020-07-08:
好了。终于切回来了，心里一颗石头终于落地……
不得不说，官方package这bug出得太及时了 :)

1974294589, 2020-07-08:
> 
> 
> 好了。终于切回来了，心里一颗石头终于落地……
> 不得不说，官方package这bug出得太及时了 :)

其实自己切回来也是可以的

closed 2020-07-08 by coolsnowwolf

elonj, 2020-07-09:
> > 好了。终于切回来了，心里一颗石头终于落地……
> > 不得不说，官方package这bug出得太及时了 :)
> 
> 其实自己切回来也是可以的

请问怎么切回4.19？可以说一下做法吗？谢谢

CodeChenL, 2020-07-23:
> > > 好了。终于切回来了，心里一颗石头终于落地……
> > > 不得不说，官方package这bug出得太及时了 :)
> > 
> > 
> > 其实自己切回来也是可以的
> 
> 请问怎么切回4.19？可以说一下做法吗？谢谢

我也想知道,有大佬解答吗

CORE2021, 2020-07-23:
target/linux/*/Makefile

elonj, 2020-07-23:
> > > > 好了。终于切回来了，心里一颗石头终于落地……
> > > > 不得不说，官方package这bug出得太及时了 :)
> > > 
> > > 
> > > 其实自己切回来也是可以的
> > 
> > 
> > 请问怎么切回4.19？可以说一下做法吗？谢谢
> 
> 我也想知道,有大佬解答吗

试试coolsnowwolf/openwrt

pupboss, 2020-09-14:
这些坑都是亲自踩一下才知道疼，我以前也是这个想法，觉得官方维护的人多就是稳定，自己折腾过发现真的不行