coolsnowwolf/lede #5022: 编译出来的.img.gz格式固件解压失败

minfengyk, 2020-07-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [* ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：全新编译出来的openwrt-x86-64-generic-squashfs-combined.img.gz格式固件，可以直接刷写固件的方式升级。但用dd方式刷机，准备解压成img格式时失败。解压软件提示：解压失败；错误码：使用p7zip。换了几种方式解压都提示错误。4.19内核没这个问题。

### (2) 路由器型号和固件版本

主机型号 | Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz : 1 Core 2 Thread
固件版本 | OpenWrt R20.7.1 / LuCI Master (git-20.187.38662-0cb730a)
内核版本 | 5.4.50

### (3) 详细日志

A：没有日志。

bigfoxtail, 2020-07-07:
编译时是会生成img格式的文件的，直接用那个

coolsnowwolf, 2020-07-07:
目前不会生成 gz 了

minfengyk, 2020-07-07:
> 目前不会生成 gz 了
我是为了省空间，编译时选了生成gz，看来只好取消这个选项，后期自己压缩。

closed 2020-07-07 by coolsnowwolf

wangweiweiwei, 2021-02-19:
> 编译时是会生成img格式的文件的，直接用那个

https://github.com/wangweiweiwei/lede/actions/runs/578209657

wangweiweiwei, 2021-02-19:
> 编译时是会生成img格式的文件的，直接用那个

https://github.com/wangweiweiwei/lede/actions/runs/578209657
你好为什么我这个没有生成img文件?

EnnawYang, 2021-02-19:
> > 编译时是会生成img格式的文件的，直接用那个
> 
> https://github.com/wangweiweiwei/lede/actions/runs/578209657
> 你好为什么我这个没有生成img文件?

[image]
没有吗？

wangweiweiwei, 2021-02-19:
> > > 编译时是会生成img格式的文件的，直接用那个
> > 
> > 
> > https://github.com/wangweiweiwei/lede/actions/runs/578209657
> > 你好为什么我这个没有生成img文件?
> 
> [image]
> 没有吗？

打扰了，没想到那个压缩包里面是

wangweiweiwei, 2021-02-19:
> > > 编译时是会生成img格式的文件的，直接用那个
> > 
> > 
> > https://github.com/wangweiweiwei/lede/actions/runs/578209657
> > 你好为什么我这个没有生成img文件?
> 
> [image]
> 没有吗？

谢谢，没想到那个压缩包里面是，下载比较慢还没有下载！