coolsnowwolf/lede #5023: 5.4内核无法挂载NTFS移动硬盘

evangelion015, 2020-07-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：挂载NTFS移动硬盘显示不存在
添加的软件包：
kmod-ata-core,kmod-scsi-generic,kmod-fs-cifs,kmod-fs-nfs,kmod-fs-common
kmod-usb-ohci,kmod-usb-storage,kmod-usb-storage-extras,kmod-usb-uas,kmod-usb2&3
ntfs-3g
[image]

### (2) 路由器型号和固件版本

A：树莓派4B，

OpenWrt R20.7.1 / LuCI Master (git-20.187.38662-0cb730a)
内核版本 | 5.4.50

### (3) 详细日志

A：
Mon Jul  6 16:17:59 2020 kern.info kernel: [  278.856148] usb 2-2: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
Mon Jul  6 16:17:59 2020 kern.info kernel: [  278.881404] usb 2-2: New USB device found, idVendor=152d, idProduct=0578, bcdDevice=82.01
Mon Jul  6 16:17:59 2020 kern.info kernel: [  278.889607] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mon Jul  6 16:17:59 2020 kern.info kernel: [  278.896762] usb 2-2: Product: UNITEK USB3.0 TO SATA
Mon Jul  6 16:17:59 2020 kern.info kernel: [  278.901655] usb 2-2: Manufacturer: UNITEK
Mon Jul  6 16:17:59 2020 kern.info kernel: [  278.905677] usb 2-2: SerialNumber: 0123456789ABCDEF
Mon Jul  6 16:17:59 2020 kern.info kernel: [  278.918340] scsi host0: uas
Mon Jul  6 16:17:59 2020 kern.notice kernel: [  278.922123] scsi 0:0:0:0: Direct-Access     UNITEK   USB3.0 TO SATA   8201 PQ: 0 ANSI: 6
Mon Jul  6 16:17:59 2020 kern.notice kernel: [  278.931585] sd 0:0:0:0: Attached scsi generic sg0 type 0
Mon Jul  6 16:17:59 2020 kern.notice kernel: [  278.937832] sd 0:0:0:0: [sda] 976773168 512-byte logical blocks: (500 GB/466 GiB)
Mon Jul  6 16:17:59 2020 kern.notice kernel: [  278.945325] sd 0:0:0:0: [sda] 4096-byte physical blocks
Mon Jul  6 16:17:59 2020 kern.notice kernel: [  278.950832] sd 0:0:0:0: [sda] Write Protect is off
Mon Jul  6 16:17:59 2020 kern.debug kernel: [  278.955630] sd 0:0:0:0: [sda] Mode Sense: 53 00 00 08
Mon Jul  6 16:17:59 2020 kern.notice kernel: [  278.956236] sd 0:0:0:0: [sda] Disabling FUA
Mon Jul  6 16:17:59 2020 kern.notice kernel: [  278.960423] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
Mon Jul  6 16:17:59 2020 kern.warn kernel: [  278.970141] sd 0:0:0:0: [sda] Optimal transfer size 33553920 bytes not a multiple of physical block size (4096 bytes)
Mon Jul  6 16:17:59 2020 kern.info kernel: [  278.998359]  sda: sda1 sda2
Mon Jul  6 16:17:59 2020 kern.notice kernel: [  279.004237] sd 0:0:0:0: [sda] Attached SCSI disk
Mon Jul  6 16:18:02 2020 kern.err kernel: [  282.272657] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:18:08 2020 kern.err kernel: [  288.296308] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:18:14 2020 kern.err kernel: [  294.316605] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:18:20 2020 kern.err kernel: [  300.336700] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:18:26 2020 kern.err kernel: [  306.361184] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:18:32 2020 kern.err kernel: [  312.381673] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:18:38 2020 kern.err kernel: [  318.408584] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:18:44 2020 kern.err kernel: [  324.428774] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:18:50 2020 kern.err kernel: [  330.453424] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:18:56 2020 kern.err kernel: [  336.478011] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:19:02 2020 kern.err kernel: [  342.498685] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:19:08 2020 kern.err kernel: [  348.520699] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:19:14 2020 kern.err kernel: [  354.543213] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:19:20 2020 kern.err kernel: [  360.568089] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
Mon Jul  6 16:19:26 2020 kern.err kernel: [  366.592704] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52

EnnawYang, 2020-07-07:
https://github.com/coolsnowwolf/lede/issues/5016#issuecomment-654607843

closed 2020-07-07 by evangelion015

evangelion015, 2020-07-07:
> [#5016 (comment)](https://github.com/coolsnowwolf/lede/issues/5016#issuecomment-654607843)

加了antfs-mount，还是无法挂载

f3404923, 2021-06-30:
我的R4S 用L大固件也是一样无法挂载，但是用友善官方固件正常识别