coolsnowwolf/lede #5031: 编译最新版报错，请大神修复一下。

jlzsss, 2020-07-08:
报错内容如下：
 make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/libffi'
2020-07-08T01:10:37.0938879Z time: package/feeds/packages/libffi/compile#0.17#0.06#0.23
2020-07-08T01:10:37.1079329Z make[3]: Entering directory '/workdir/openwrt/feeds/packages/lang/python/python-pip-conf'
2020-07-08T01:10:37.2425027Z make[3]: Nothing to be done for 'compile'.
2020-07-08T01:10:37.2432282Z make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/python/python-pip-conf'
2020-07-08T01:10:37.2433108Z time: package/feeds/packages/python-pip-conf/compile#0.11#0.02#0.14
2020-07-08T01:10:37.2568854Z make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/expat'
2020-07-08T01:10:37.4298999Z CFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/workdir/openwrt/staging_dir/host/lib -L/workdir/openwrt/staging_dir/hostpkg/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/host/lib" make -j1 -C /workdir/openwrt/build_dir/hostpkg/expat-2.2.9  
2020-07-08T01:10:37.4334152Z make[4]: Entering directory '/workdir/openwrt/build_dir/hostpkg/expat-2.2.9'
2020-07-08T01:10:37.4436016Z make[5]: Entering directory '/workdir/openwrt/build_dir/hostpkg/expat-2.2.9'
2020-07-08T01:10:37.4449234Z make[6]: Entering directory '/workdir/openwrt/build_dir/hostpkg/expat-2.2.9'
2020-07-08T01:10:37.4487130Z make[6]: Leaving directory '/workdir/openwrt/build_dir/hostpkg/expat-2.2.9'
2020-07-08T01:10:37.4501647Z make[6]: Entering directory '/workdir/openwrt/build_dir/hostpkg/expat-2.2.9'
2020-07-08T01:10:37.4530412Z [  4%] Linking C shared library libexpat.so
2020-07-08T01:10:37.4732084Z Using catalogs: /etc/sgml/catalog
2020-07-08T01:10:37.4733976Z Using stylesheet: /usr/share/docbook-utils/docbook-utils.dsl#print
2020-07-08T01:10:37.4735050Z Working on: /workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml
2020-07-08T01:10:37.4837494Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:17:0:E: DTD did not contain element declaration for document type name
2020-07-08T01:10:37.4840537Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:19:9:E: element "REFENTRY" undefined
2020-07-08T01:10:37.4841530Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:20:15:E: element "REFENTRYINFO" undefined
2020-07-08T01:10:37.4842268Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:21:12:E: element "ADDRESS" undefined
2020-07-08T01:10:37.4842928Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:22:15:E: element "EMAIL" undefined
2020-07-08T01:10:37.4843606Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:24:11:E: element "AUTHOR" undefined
2020-07-08T01:10:37.4844286Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:25:19:E: element "FIRSTNAME" undefined
2020-07-08T01:10:37.4844983Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:26:17:E: element "SURNAME" undefined
2020-07-08T01:10:37.4845667Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:28:14:E: element "COPYRIGHT" undefined
2020-07-08T01:10:37.4846369Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:29:11:E: element "YEAR" undefined
2020-07-08T01:10:37.4847056Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:30:13:E: element "HOLDER" undefined
2020-07-08T01:10:37.4847764Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:32:12:E: element "DATE" undefined
2020-07-08T01:10:37.4848441Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:34:10:E: element "REFMETA" undefined
2020-07-08T01:10:37.4849127Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:35:17:E: element "REFENTRYTITLE" undefined
2020-07-08T01:10:37.4850100Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:37:15:E: element "MANVOLNUM" undefined
2020-07-08T01:10:37.4850947Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:39:13:E: element "REFNAMEDIV" undefined
2020-07-08T01:10:37.4851607Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:40:12:E: element "REFNAME" undefined
2020-07-08T01:10:37.4852341Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:42:15:E: element "REFPURPOSE" undefined
2020-07-08T01:10:37.4853037Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:44:17:E: element "REFSYNOPSISDIV" undefined
2020-07-08T01:10:37.4853700Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:45:16:E: element "CMDSYNOPSIS" undefined
2020-07-08T01:10:37.4854352Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:46:14:E: element "COMMAND" undefined
2020-07-08T01:10:37.4855004Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:47:7:E: element "ARG" undefined
2020-07-08T01:10:37.4856094Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:47:15:E: element "OPTION" undefined
2020-07-08T01:10:37.4856679Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:48:7:E: element "ARG" undefined
2020-07-08T01:10:37.4857495Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:48:15:E: element "OPTION" undefined
2020-07-08T01:10:37.4858614Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:49:7:E: element "ARG" undefined
2020-07-08T01:10:37.4859517Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:49:15:E: element "OPTION" undefined
2020-07-08T01:10:37.4860326Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:50:7:E: element "ARG" undefined
2020-07-08T01:10:37.4860924Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:50:15:E: element "OPTION" undefined
2020-07-08T01:10:37.4861309Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:52:7:E: element "ARG" undefined
2020-07-08T01:10:37.4861997Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:52:15:E: element "OPTION" undefined
2020-07-08T01:10:37.4862643Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:52:31:E: element "REPLACEABLE" undefined
2020-07-08T01:10:37.4863119Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:53:7:E: element "ARG" undefined
2020-07-08T01:10:37.4863486Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:53:15:E: element "OPTION" undefined
2020-07-08T01:10:37.4863859Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:55:7:E: element "ARG" undefined
2020-07-08T01:10:37.4864215Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:55:15:E: element "OPTION" undefined
2020-07-08T01:10:37.4864577Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:55:31:E: element "REPLACEABLE" undefined
2020-07-08T01:10:37.4864944Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:56:7:E: element "ARG" undefined
2020-07-08T01:10:37.4865297Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:56:15:E: element "OPTION" undefined
2020-07-08T01:10:37.4865649Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:57:7:E: element "ARG" undefined
2020-07-08T01:10:37.4865997Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:57:15:E: element "OPTION" undefined
2020-07-08T01:10:37.4866339Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:59:7:E: element "ARG" undefined
2020-07-08T01:10:37.4866694Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:59:15:E: element "OPTION" undefined
2020-07-08T01:10:37.4867036Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:60:7:E: element "ARG" undefined
2020-07-08T01:10:37.4867385Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:60:15:E: element "OPTION" undefined
2020-07-08T01:10:37.4867957Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:61:14:E: element "ARG" undefined
2020-07-08T01:10:37.4868310Z onsgmls:/workdir/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:61:22:E: element "OPTION" undefined
[... cut ...]
2020-07-08T01:10:37.5188782Z make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2020-07-08T01:10:37.5189701Z package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
2020-07-08T01:10:37.5190194Z make: *** [world] Error 2
2020-07-08T01:10:37.5190501Z make[1]: Leaving directory '/workdir/openwrt'
2020-07-08T01:10:37.5191212Z /workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
2020-07-08T01:10:37.5212918Z ##[error]Process completed with exit code 2.

bigfoxtail, 2020-07-08:
make clean 重新编译试试

jlzsss, 2020-07-08:
我用的是actions-openwrt编译的固件，应该是最新的代码。

wwqgtxx, 2020-07-08:
我这边也是一样的错误
``` log
2020-07-08T00:29:41.7602183Z make[2]: Entering directory '/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/feeds/packages/libs/expat'
2020-07-08T00:29:41.9134232Z CFLAGS="-O2 -I/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/staging_dir/host/include -I/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/staging_dir/hostpkg/include -I/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/staging_dir/host/include -I/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/staging_dir/hostpkg/include -I/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/staging_dir/host/lib -L/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/staging_dir/hostpkg/lib -L/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/staging_dir/target-x86_64_musl/host/lib" make -j1 -C /home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9  
2020-07-08T00:29:41.9183185Z make[3]: Entering directory '/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9'
2020-07-08T00:29:41.9284249Z make[4]: Entering directory '/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9'
2020-07-08T00:29:41.9311434Z make[5]: Entering directory '/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9'
2020-07-08T00:29:41.9356254Z make[5]: Leaving directory '/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9'
2020-07-08T00:29:41.9370935Z make[5]: Entering directory '/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9'
2020-07-08T00:29:41.9398177Z [  4%] Linking C shared library libexpat.so
2020-07-08T00:29:41.9592381Z Using catalogs: /etc/sgml/catalog
2020-07-08T00:29:41.9593081Z Using stylesheet: /usr/share/docbook-utils/docbook-utils.dsl#print
2020-07-08T00:29:41.9593456Z Working on: /home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml
2020-07-08T00:29:41.9880064Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:17:0:E: DTD did not contain element declaration for document type name
2020-07-08T00:29:41.9883879Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:19:9:E: element "REFENTRY" undefined
2020-07-08T00:29:41.9884829Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:20:15:E: element "REFENTRYINFO" undefined
2020-07-08T00:29:41.9885450Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:21:12:E: element "ADDRESS" undefined
2020-07-08T00:29:41.9886053Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:22:15:E: element "EMAIL" undefined
2020-07-08T00:29:41.9886653Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:24:11:E: element "AUTHOR" undefined
2020-07-08T00:29:41.9887260Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:25:19:E: element "FIRSTNAME" undefined
2020-07-08T00:29:41.9887856Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:26:17:E: element "SURNAME" undefined
2020-07-08T00:29:41.9888469Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:28:14:E: element "COPYRIGHT" undefined
2020-07-08T00:29:41.9889060Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:29:11:E: element "YEAR" undefined
2020-07-08T00:29:41.9889652Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:30:13:E: element "HOLDER" undefined
2020-07-08T00:29:41.9890247Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:32:12:E: element "DATE" undefined
2020-07-08T00:29:41.9890838Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:34:10:E: element "REFMETA" undefined
2020-07-08T00:29:41.9891430Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:35:17:E: element "REFENTRYTITLE" undefined
2020-07-08T00:29:41.9892151Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:37:15:E: element "MANVOLNUM" undefined
2020-07-08T00:29:41.9892865Z onsgmls:/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:39:13:E: element "REFNAMEDIV" undefined
[... cut ...]
2020-07-08T00:29:42.0194282Z time: package/feeds/packages/expat/host-compile#0.19#0.04#0.26
2020-07-08T00:29:42.0194601Z package/Makefile:111: recipe for target 'package/feeds/packages/expat/host/compile' failed
2020-07-08T00:29:42.0194936Z make[1]: Leaving directory '/home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt'
2020-07-08T00:29:42.0195311Z /home/runner/work/Action-OpenWrt-Plus/Action-OpenWrt-Plus/openwrt/include/toplevel.mk:222: recipe for target 'package/compile' failed
2020-07-08T00:29:42.0259433Z ##[error]Process completed with exit code 2.
```

Memories0ff, 2020-07-08:
**同样的错误**
> [  4%] Linking C shared library libexpat.so
Using catalogs: /etc/sgml/catalog
Using stylesheet: /usr/share/docbook-utils/docbook-utils.dsl#print
Working on: /home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:17:0:E: DTD did not contain element declaration for document type name
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:19:9:E: element "REFENTRY" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:20:15:E: element "REFENTRYINFO" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:21:12:E: element "ADDRESS" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:22:15:E: element "EMAIL" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:24:11:E: element "AUTHOR" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:25:19:E: element "FIRSTNAME" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:26:17:E: element "SURNAME" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:28:14:E: element "COPYRIGHT" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:29:11:E: element "YEAR" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:30:13:E: element "HOLDER" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:32:12:E: element "DATE" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:34:10:E: element "REFMETA" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:35:17:E: element "REFENTRYTITLE" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:37:15:E: element "MANVOLNUM" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:39:13:E: element "REFNAMEDIV" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:40:12:E: element "REFNAME" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:42:15:E: element "REFPURPOSE" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:44:17:E: element "REFSYNOPSISDIV" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:45:16:E: element "CMDSYNOPSIS" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:46:14:E: element "COMMAND" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:47:7:E: element "ARG" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:47:15:E: element "OPTION" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:48:7:E: element "ARG" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:48:15:E: element "OPTION" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:49:7:E: element "ARG" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:49:15:E: element "OPTION" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:50:7:E: element "ARG" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:50:15:E: element "OPTION" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:52:7:E: element "ARG" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:52:15:E: element "OPTION" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:52:31:E: element "REPLACEABLE" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:53:7:E: element "ARG" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:53:15:E: element "OPTION" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:55:7:E: element "ARG" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:55:15:E: element "OPTION" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:55:31:E: element "REPLACEABLE" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:56:7:E: element "ARG" undefined
onsgmls:/home/openwrtcompilation/source_code/lede/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:56:15:E: element "OPTION" undefined
[... cut ...]
make[2]: Leaving directory '/home/openwrtcompilation/source_code/lede'
package/Makefile:107: recipe for target '/home/openwrtcompilation/source_code/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/openwrtcompilation/source_code/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrtcompilation/source_code/lede'
/home/openwrtcompilation/source_code/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

Pantheon-Paradoxes, 2020-07-08:
哈，都同样错误

ptsa, 2020-07-08:
我也是一样错误

coolsnowwolf, 2020-07-08:
这个是官方的 package 包，我无权限修复，请向官方提交

coolsnowwolf, 2020-07-08:
[image]
来自官方的这个提交的错误

wwqgtxx, 2020-07-08:
[image]
主要是官方早上才合并进feeds仓库，所以凌晨编译的都没事

bigfoxtail, 2020-07-08:
没有修bug的技术非得要玩新版当小白鼠，傻眼了吧

kisaragychihaya, 2020-07-08:
让你们天天新，舒服了不

hcym, 2020-07-08:
没看懂，中午编译的正常

Myoko, 2020-07-08:
所以说 4.19才是最稳的！

closed 2020-07-08 by coolsnowwolf

wushenghn, 2020-07-08:
官方的坑刚更新等同步

anjue39, 2022-09-14:
类似的错误发生在这几天2022年

# download package环节出现以下错误

Run cd openwrt  
。。。  
Makefile:140: \*\*\* missing separator. Stop.  
make\[5\]: \*\*\* \[Makefile:30: image-prereq\] Error 2  
make\[4\]: \*\*\* \[Makefile:11: prereq\] Error 2  
time: target/linux/prereq#0.08#0.04#0.12  
ERROR: target/linux failed to build.  
make\[3\]: \*\*\* \[target/Makefile:30: target/linux/prereq\] Error 1  
make\[2\]: \*\*\* \[target/Makefile:[22](https://github.com/anjue39/K3_LEDE/actions/runs/3056149234/jobs/4930004320#step:10:23): /workdir/openwrt/staging\_dir/target-arm\_cortex-a9\_musl\_eabi/stamp/.target\_prereq\] Error 2  
make\[1\]: \*\*\* \[/workdir/openwrt/include/toplevel.mk:208: prereq\] Error 2  
make: \*\*\* \[/workdir/openwrt/include/toplevel.mk:2[28](https://github.com/anjue39/K3_LEDE/actions/runs/3056149234/jobs/4930004320#step:10:29): tools/flock/compile\] Error 2  
Error: Process completed with exit code 2.

# organaze files环节出现以下错误

Run cd openwrt/bin/targets/_/_  
cd openwrt/bin/targets/_/_  
rm -rf packages  
echo "FIRMWARE=$PWD" >> $GITHUB\_ENV  
echo "::set-output name=status::success"  
shell: /usr/bin/bash -e {0}  
env:  
REPO\_URL: [https://github.com/coolsnowwolf/lede](https://github.com/coolsnowwolf/lede)  
REPO\_BRANCH: master  
FEEDS\_CONF: feeds.conf.default  
CONFIG\_FILE: ssrp+argon.config  
DIY\_P1\_SH: diy-part1.sh  
DIY\_P[2](https://github.com/anjue39/K3_LEDE/actions/runs/3056149234/jobs/4930004320#step:14:2)\_SH: diy-part2.sh  
UPLOAD\_BIN\_DIR: false  
UPLOAD\_FIRMWARE: true  
UPLOAD\_COWTRANSFER: false  
UPLOAD\_WETRANSFER: false  
UPLOAD\_RELEASE: false  
TZ: Asia/Shanghai  
/home/runner/work/\_temp/4[3](https://github.com/anjue39/K3_LEDE/actions/runs/3056149234/jobs/4930004320#step:14:3)0[4](https://github.com/anjue39/K3_LEDE/actions/runs/3056149234/jobs/4930004320#step:14:4)3cbc-f[5](https://github.com/anjue39/K3_LEDE/actions/runs/3056149234/jobs/4930004320#step:14:5)55-425[6](https://github.com/anjue39/K3_LEDE/actions/runs/3056149234/jobs/4930004320#step:14:6)\-8411-6[7](https://github.com/anjue39/K3_LEDE/actions/runs/3056149234/jobs/4930004320#step:14:7)6[9](https://github.com/anjue39/K3_LEDE/actions/runs/3056149234/jobs/4930004320#step:14:9)f72c5[10](https://github.com/anjue39/K3_LEDE/actions/runs/3056149234/jobs/4930004320#step:14:10)0.sh: line 1: cd: openwrt/bin/targets/_/_: No such file or directory  
Error: Process completed with exit code 1.