coolsnowwolf/lede #5037: 最新编译k2p，无错误显示，输出文件大小和文件名错误。

ltactor, 2020-07-08:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译输出openwrt-ramips-mt7621-phicomm_k2p-initramfs-kernel.bin   15751kb 和其他几个仅几KB的文件，
没有openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade.bin  和 openwrt-ramips-mt7621-phicomm_k2p-initramfs-kernel.bin 文件。
使用daoker 编译、  github actions编译、 新debain编译，三种环境完全按官方教程按步骤进行，同样结果。
前一段时间编译正常。
仅比默认配置多选择helloworld 、WireGuard
日志没有明显错误。

### (2) 路由器型号和固件版本

A：k2p  最新固件

### (3) 详细日志

A：Generating index for package ./libcap-ng_0.7.10-1_mipsel_24kc.ipk
Generating index for package ./libev_4.31-1_mipsel_24kc.ipk
Generating index for package ./libnet-1.2.x_1.2-rc3-4_mipsel_24kc.ipk
Generating index for package ./libpcre_8.44-1_mipsel_24kc.ipk
Generating index for package ./libsodium_1.0.18-1_mipsel_24kc.ipk
Generating index for package ./miniupnpd_2.1.20200510-3_mipsel_24kc.ipk
Generating index for package ./nlbwmon_2020-04-11.1-34a18876-2_mipsel_24kc.ipk
Generating index for package ./shadowsocks-libev-config_3.3.4-4_mipsel_24kc.ipk
Generating index for package ./shadowsocks-libev-ss-local_3.3.4-4_mipsel_24kc.ipk
Generating index for package ./shadowsocks-libev-ss-redir_3.3.4-4_mipsel_24kc.ipk
Generating index for package ./wget_1.20.3-4_mipsel_24kc.ipk
Signing package index...
make[2]: Leaving directory '/home/song/k2p'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/home/song/k2p'
make[2]: Nothing to be done for 'json_overview_image_info'.
make[2]: Leaving directory '/home/song/k2p'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/song/k2p'
make[2]: Leaving directory '/home/song/k2p'
make[1]: Leaving directory '/home/song/k2p'

请教哪里出了问题？

EnnawYang, 2020-07-08:
那就是固件太大了呗

ltactor, 2020-07-08:
> 那就是固件太大了呗

以前编译同样配置，生成固件只有12.4MB

coolsnowwolf, 2020-07-08:
新的 5.4 无线驱动本身就 8.4MB 了

coolsnowwolf, 2020-07-08:
以前是 4.8MB，大了 3.6MB

ltactor, 2020-07-08:
> 以前是 4.8MB，大了 3.6MB

感谢回复，那只能取消掉不用的选项了。

ltactor, 2020-07-08:
解决了
就是选了helloworld+WireGuard 超出了固件大小限制。
取消了不产用的屏蔽广告+ddns和wireguard，可能还有几个不常用的，可以编译成功。 但备份文件不能通用。

closed 2020-07-08 by ltactor

lfm09, 2020-08-15:
这个问题很容易难倒新手，建议固件超大未生成时，要加个出错提示。

另外，我还碰到过 squashfs-sysupgrade.bin 比 initramfs-kernel.bin 更小的情况，不知什么原因？

xlf1234, 2020-08-15:
固件太大的 错误提示一直都有