coolsnowwolf/lede #5040: x86_64编译失败，已make clean过

li-xunhuan, 2020-07-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：X86_64

### (3) 详细日志

```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for automount:
 *      antfs-mount
 * opkg_install_cmd: Cannot install package automount.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-cifsd:
 *      ksmbd-server
 * opkg_install_cmd: Cannot install package luci-app-cifsd.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-rclone:
 *      fuse-utils
 * opkg_install_cmd: Cannot install package luci-app-rclone.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-proto-bonding:
 *      proto-bonding
 * opkg_install_cmd: Cannot install package luci-proto-bonding.
package/Makefile:65: recipe for target 'package/install' failed
```

xiaoguoyi, 2020-07-08:
把bin文件夹删了再编译

minlang112, 2020-07-08:
./scripts/feeds clean

然后重新走一遍编译流程，并且如xiaoguoyi所述尝试删除Bin内已生成的文件

wushenghn, 2020-07-08:
重新clone就好了

closed 2020-07-08 by coolsnowwolf

kktt007, 2020-07-09:
我也是出错了

```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for automount:
 * 	antfs-mount
 * opkg_install_cmd: Cannot install package automount.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for autosamba:
 * 	wsdd2
 * opkg_install_cmd: Cannot install package autosamba.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-proto-bonding:
 * 	proto-bonding
 * opkg_install_cmd: Cannot install package luci-proto-bonding.
```

li-xunhuan, 2020-07-11:
> 我也是出错了
>
> ```
> Collected errors:
>  * satisfy_dependencies_for: Cannot satisfy the following dependencies for automount:
>  * 	antfs-mount
>  * opkg_install_cmd: Cannot install package automount.
>  * [...]

我`make distclean`之后再编译，可以了！

hu-min, 2021-06-05:
> make distclean

晕死，从新开始，还是没通过，好坑。