coolsnowwolf/lede #5042: 新3编译失败

EnnawYang, 2020-07-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：新3编译失败，应该与刚刚合并的PR有关

### (2) 路由器型号和固件版本

A：Newifi-D2

### (3) 详细日志

A：

```
rm -f /home/xw/openwrt/lede-autobuild/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_buffalo_wsr-600dhp.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/xw/openwrt/lede-autobuild/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.50/arch/mips/boot/dts -I/home/xw/openwrt/lede-autobuild/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.50/arch/mips/boot/dts/include -I/home/xw/openwrt/lede-autobuild/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.50/include/ -undef -D__DTS__  -o /home/xw/openwrt/lede-autobuild/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_dlink_dir-860l-b1.dtb.tmp ../dts/mt7621_dlink_dir-860l-b1.dts
/home/xw/openwrt/lede-autobuild/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.50/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/xw/openwrt/lede-autobuild/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_dlink_dir-860l-b1.dtb /home/xw/openwrt/lede-autobuild/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_dlink_dir-860l-b1.dtb.tmp
rm -f /home/xw/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_dlink_dir-860l-b1.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/xw/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.50/arch/mips/boot/dts -I/home/xw/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.50/arch/mips/boot/dts/include -I/home/xw/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.50/include/ -undef -D__DTS__  -o /home/xw/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_dlink_dir_878_a1.dtb.tmp ../dts/mt7621_dlink_dir_878_a1.dts
mipsel-openwrt-linux-musl-cpp: error: ../dts/mt7621_dlink_dir_878_a1.dts: No such file or directory
mipsel-openwrt-linux-musl-cpp: warning: '-x assembler-with-cpp' after last input file has no effect
mipsel-openwrt-linux-musl-cpp: fatal error: no input files
compilation terminated.
make[5]: *** [Makefile:201: /home/xw/openwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_dlink_dir_878_a1.dtb] Error 1
make[5]: Leaving directory '/home/xw/openwrt/lede/target/linux/ramips/image'
make[4]: *** [Makefile:25: install] Error 2
make[4]: Leaving directory '/home/xw/openwrt/lede/target/linux/ramips'
make[3]: *** [Makefile:13: install] Error 2
make[3]: Leaving directory '/home/xw/openwrt/lede/target/linux'
time: target/linux/install#31.63#9.62#40.19
make[2]: *** [target/Makefile:25: target/linux/install] Error 2
make[2]: Leaving directory '/home/xw/openwrt/lede'
make[1]: *** [target/Makefile:19: /home/xw/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/xw/openwrt/lede'
make: *** [/home/xw/openwrt/lede/include/toplevel.mk:224：world] 错误 2

```

BoBppy, 2020-07-08:
k2p这里差不多的，初步怀疑那几个pr改的dtsi有点问题

coolsnowwolf, 2020-07-08:
就是那几个吧

BoBppy, 2020-07-08:
要revert么

closed 2020-07-08 by coolsnowwolf