coolsnowwolf/lede #5044: R2S在固件中刷机不生效

ramondelee, 2020-07-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：R2S是TF卡，在固件中刷机不生效

### (2) 路由器型号和固件版本

A：型号：R2S，版本：内核版本 | 5.4.48
OpenWrt R20.7.1 / LuCI Master (git-20.187.38662-0cb730a)

### (3) 详细日志

A：

DanielChen008, 2020-07-09:
L大没有提供 R2S 的硬件支持吧，我记得之前说过友善是自己的SDK，你这种问题需要自行解决，这不是 LEDE 的BUG反馈。

wushenghn, 2020-07-10:
现在的码里有r2s,我自己编译的刷进去都跑起来了

DanielChen008, 2020-07-10:
> 现在的码里有r2s,我自己编译的刷进去都跑起来了

噢？make menuconfig 里已经加入了 R2S 的处理器选项？我晚上回家看看！

ramondelee, 2020-07-10:
> 现在的码里有r2s,我自己编译的刷进去都跑起来了

可以在固件中刷机成功吗？

wushenghn, 2020-07-10:
> > 现在的码里有r2s,我自己编译的刷进去都跑起来了
> 
> 可以在固件中刷机成功吗？

可以我现在就在用

ramondelee, 2020-07-10:
> > > 现在的码里有r2s,我自己编译的刷进去都跑起来了
> > 
> > 
> > 可以在固件中刷机成功吗？
> 
> 可以我现在就在用

你是哪个版本编译的，我的是前天的

wushenghn, 2020-07-10:
> > > > 现在的码里有r2s,我自己编译的刷进去都跑起来了
> > > 
> > > 
> > > 可以在固件中刷机成功吗？
> > 
> > 
> > 可以我现在就在用
> 
> 你是哪个版本编译的，我的是前天的，直接在固件刷机无效

前天晚上编的

ramondelee, 2020-07-10:
> > > > > 现在的码里有r2s,我自己编译的刷进去都跑起来了
> > > > 
> > > > 
> > > > 可以在固件中刷机成功吗？
> > > 
> > > 
> > > 可以我现在就在用
> > 
> > 
> > 你是哪个版本编译的，我的是前天的，直接在固件刷机无效
> 
> 前天晚上编的

在固件中直接刷.gz包吗

wushenghn, 2020-07-10:
> > > > > > 现在的码里有r2s,我自己编译的刷进去都跑起来了
> > > > > 
> > > > > 
> > > > > 可以在固件中刷机成功吗？
> > > > 
> > > > 
> > > > 可以我现在就在用
> > > 
> > > 
> > > 你是哪个版本编译的，我的是前天的，直接在固件刷机无效
> > 
> > 
> > 前天晚上编的
> 
> 在固件中直接刷.gz包吗

你搞笑的么解压刷img

ramondelee, 2020-07-10:
> > > > > > > 现在的码里有r2s,我自己编译的刷进去都跑起来了
> > > > > > 
> > > > > > 
> > > > > > 可以在固件中刷机成功吗？
> > > > > 
> > > > > 
> > > > > 可以我现在就在用
> > > > 
> > > > 
> > > > 你是哪个版本编译的，我的是前天的，直接在固件刷机无效
> > > 
> > > 
> > > 前天晚上编的
> > 
> > 
> > 在固件中直接刷.gz包吗
> 
> 你搞笑的么解压刷img

现在编译一版试试

ramondelee, 2020-07-12:
试了可以刷成功，不保存配置好像密码还是要重置才会清空配置

ramondelee, 2020-07-13:
测试多次，确认，刷机后必须重置才会生效，不然会自动保存原配置（刷时去掉保存配置复选），另外重置后必须断电重启网口才能正常识别

closed 2021-02-04 by coolsnowwolf