coolsnowwolf/lede #5047: make menuconfig 显示无法找到ncurses.h

swhoro, 2020-07-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：make menuconfig 后会显示无法找到ncurses.h
已经确定安装所有依赖
ubuntu版本为20.04

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
[image]

minlang112, 2020-07-09:
试一下  sudo apt-get install libncurses5-dev

swhoro, 2020-07-09:
> 试一下 sudo apt-get install libncurses5-dev

[image]
已经安装

minlang112, 2020-07-09:
> > 试一下 sudo apt-get install libncurses5-dev
> 
> [image]
> 已经安装

那看起来是pull的库文件有问题，删掉重新git pull吧

swhoro, 2020-07-09:
> > > 试一下 sudo apt-get install libncurses5-dev
> > 
> > 
> > [image]
> > 已经安装
> 
> 那看起来是pull的库文件有问题，删掉重新git pull吧

是整个lede文件夹删掉再clone吗？
还是只删掉某个文件夹呢？
不知道库文件是哪个

minlang112, 2020-07-10:
> > > > 试一下 sudo apt-get install libncurses5-dev
> > > 
> > > 
> > > [image]
> > > 已经安装
> > 
> > 
> > 那看起来是pull的库文件有问题，删掉重新git pull吧
> 
> 是整个lede文件夹删掉再clone吗？
> 还是只删掉某个文件夹呢？
> 不知道库文件是哪个

整个删除，然后重新git clone吧，保证网络通畅

closed 2020-07-10 by coolsnowwolf