coolsnowwolf/lede #5051: zstd-1.4.5.tar.zst: cannot execute binary file: Exec format error

openwrt-1983, 2020-07-09:
make[3]: Entering directory '/home/mima1234/openwrt/lede/feeds/packages/utils/zstd'
touch /home/mima1234/openwrt/lede/build_dir/target-mipsel_24kc_musl/zstd-1.4.5/.prepared_75483c4667fb02d11dfd8b39eae2a0ba_6664517399ebbbc92a37c5bb081b5c53_check
. /home/mima1234/openwrt/lede/include/shell.sh;  /home/mima1234/openwrt/lede/dl/zstd-1.4.5.tar.zst | tar -C /home/mima1234/openwrt/lede/build_dir/target-mipsel_24kc_musl/zstd-1.4.5/.. -xf - 
bash: /home/mima1234/openwrt/lede/dl/zstd-1.4.5.tar.zst: cannot execute binary file: Exec format error
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
Makefile:101: recipe for target '/home/mima1234/openwrt/lede/build_dir/target-mipsel_24kc_musl/zstd-1.4.5/.prepared_75483c4667fb02d11dfd8b39eae2a0ba_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/mima1234/openwrt/lede/build_dir/target-mipsel_24kc_musl/zstd-1.4.5/.prepared_75483c4667fb02d11dfd8b39eae2a0ba_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/home/mima1234/openwrt/lede/feeds/packages/utils/zstd'
time: package/feeds/packages/zstd/compile#0.04#0.09#0.25
package/Makefile:107: recipe for target 'package/feeds/packages/zstd/compile' failed
make[2]: *** [package/feeds/packages/zstd/compile] Error 2
make[2]: Leaving directory '/home/mima1234/openwrt/lede'
package/Makefile:103: recipe for target '/home/mima1234/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/mima1234/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mima1234/openwrt/lede'
/home/mima1234/openwrt/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

openwrt-1983, 2020-07-09:
这个问题我折腾了好几次源码都是这样

closed 2020-07-09 by coolsnowwolf

AmadeusGhost, 2020-07-09:
你网络问题

happyboy19, 2020-07-23:
遇到了同样的问题，你解决了吗？

shuzhifenxi, 2020-08-01:
如果不是unpack.mk不完整，就是系统没有装zstd，装一下就好了，这个神坑我也是搞了半天才明白。