coolsnowwolf/lede #5054: 默认的配置编译出来没有ipv6

xh0823, 2020-07-09:
默认的编译出来了wan6口没有ipv6地址要选择哪个呢？

minlang112, 2020-07-09:
请编译时候勾选 ipv6 helper，默认是不带的

xh0823, 2020-07-09:
> 请编译时候重置ipv6 helper，替代是不带的

是哪个选项呢？55r里面怎么找不到在哪里选择

minlang112, 2020-07-09:
> > 请编译时候重置ipv6 helper，替代是不带的
> 
> 是哪个选项呢？55r里面怎么找不到在哪里选择

Extra packages —> ipv6helper

xh0823, 2020-07-09:
ssr怎么搞出来呢 谢谢你说下

   

  
  
Extra packages —&gt; ipv6helper

keivenliao, 2020-07-10:
[image]
[image]
[image]
[image]
[image]

编译完毕后，以上几个地方设置一下就应该可以了。

hcwang, 2020-07-10:
使用以上设置后，SSR Plus是否可以正常使用？多谢！

minlang112, 2020-07-10:
> 使用以上设置后，SSR Plus是否可以正常使用？多谢！

SSR plus和这个并无任何关系，在git clone之后编辑feeds.conf.default，去掉helloworld那行的#，保存，然后再继续进行feeds的更新配置，然后make menuconfig，在luci的application里就能找到你想要的东西了。

hcwang, 2020-07-10:
SSR Plus和ipv6有冲突呀，在群里面搜索一下。

minlang112, 2020-07-10:
> SSR Plus和ipv6有冲突呀，在群里面搜索一下。

没有冲突，可以同时使用。如果开启了IPV6的AAAA解析（默认选项是关闭），部分墙外的双栈网站会优先返回IPV6地址，导致第一次访问的时候返回被污染的结果，多刷新几次就正常了；另部分屏蔽中国IP的双栈网站会因为该问题导致拒绝访问，将此种站点加入强制走代理的列表即可。

hcwang, 2020-07-18:
我使用了以上设置，IPV6工作一切正常。另外，使用Softether VPN搭建了一个VPN Server，使用的是TAP类型，将其与LAN网卡进行关联。但通过Softether VPN远程拔上来的设备只能获得LAN的IPV4地址，而无法获得IPV6地址，是有什么地方设置不对吗？

NightZy, 2020-07-19:
> 我使用了以上设置，IPV6工作一切正常。另外，使用Softether VPN搭建了一个VPN Server，使用的是TAP类型，将其与LAN网卡进行关联。但通过Softether VPN远程拔上来的设备只能获得LAN的IPV4地址，而无法获得IPV6地址，是有什么地方设置不对吗？

我也遇到同样的问题，并且这个问题在官方原版openwrt中也有

NightZy, 2020-08-02:
> > 我使用了以上设置，IPV6工作一切正常。另外，使用Softether VPN搭建了一个VPN Server，使用的是TAP类型，将其与LAN网卡进行关联。但通过Softether VPN远程拔上来的设备只能获得LAN的IPV4地址，而无法获得IPV6地址，是有什么地方设置不对吗？
> 
> 我也遇到同样的问题，并且这个问题在官方原版openwrt中也有，可能是 Softether VPN 配置的问题，求指点

我遇到的问题与你略有不同，我通过 softether vpn 可以获得 ipv6 地址，但不能获得 ipv6 默认网关，所以 ipv6 不能正常使用，我找到了解决方案。
[https://github.com/SoftEtherVPN/SoftEtherVPN/issues/1065#issuecomment-570795731](url)

hcwang, 2020-08-03:
> > > 我使用了以上设置，IPV6工作一切正常。另外，使用Softether VPN搭建了一个VPN Server，使用的是TAP类型，将其与LAN网卡进行关联。但通过Softether VPN远程拔上来的设备只能获得LAN的IPV4地址，而无法获得IPV6地址，是有什么地方设置不对吗？
> > 
> > 
> > 我也遇到同样的问题，并且这个问题在官方原版openwrt中也有，可能是 Softether VPN 配置的问题，求指点
> 
> 我遇到的问题与你略有不同，我通过 softether vpn 可以获得 ipv6 地址，但不能获得 ipv6 默认网关，所以 ipv6 不能正常使用，我找到了解决方案。
> [SoftEtherVPN/SoftEtherVPN#1065 (comment)](https://github.com/SoftEtherVPN/SoftEtherVPN/issues/1065#issuecomment-570795731)

非常感谢，可否把你的TAP设置的详细步骤发个教程，多谢了！

NightZy, 2020-08-03:
> 非常感谢，可否把你的TAP设置的详细步骤发个教程，多谢了！

除了上面说的解决方案，我并没有进行什么特殊的配置，以下就是步骤了。

1. tap设备的配置
[image]
2. Openwrt：桥接tap到LAN上
[image]
3. 获取不到ipv6默认网关的解决方法
[image]

hcwang, 2020-08-04:
> 除了上面说的解决方案，我并没有进行什么特殊的配置，以下就是步骤了。
>
> 1. tap设备的配置
>    [image]
> 2. Openwrt：桥接tap到LAN上
>    [image]
> 3. 获取不到ipv6默认网关的解决方法
>    [image]

好像你们都没有vlan以及无线给桥接，以下是我的设置。
[image]

closed 2021-02-04 by coolsnowwolf

yicj, 2022-08-10:
正在研究呢，我试试呢。